zephyrproject-rtos/zephyr #36226: Add support for interrupt and work queue affinity

vladimir-podbrezsky, 2021-06-14:
**Is your enhancement proposal related to a problem? Please describe.**
We need to run ISR, task and work queue on the same CPU for reducing cache misses. This may improve network performance through the use of threads, work queues and interrupts affinity.
For Aarch64, containing GIC controller, the set of GICD_IROUTER<n> registers may be used. 

**Describe the solution you'd like**
Add new interfaces allowing IRQ and "work queue" to run on the specific CPU.

andyross, 2021-06-14:
A CPU affinity API for threads (e.g. your task and work queue are just threads) already exists, see `k_thread_cpu_mask_*()`.

The kernel so far has been agnostic about interrupt targetting, and different systems do it differently (often **very** differently, as with x86 MSI/MSI-X where the device driver needs to be involved).  My personal opinion is that in an RTOS we don't want to be supporting a prescriptive API unless we're sure we have one that works for everyone.

In practice, this is always going to be a low-level optimization decision on the part of an application, which means that's going to need to happen long after the platform is chosen, which means that it's unlikely to benefit from being written to a portable API.  Maybe the right choice is to let us grow a few IRQ direction APIs at the SoC or driver layer (see some work currently on MSI for an example, and it's likely I'll be submitting another for the intel_adsp soon) and see what we can do to unify them?

nashif, 2021-06-23:
@vladimir-podbrezsky do you have a proposal for implementing this? If you have code you can always submit as a PR.

nashif, 2026-11-02:
> Add support for interrupt and work queue affinity

A work queue is a thread, so the existing k_thread_cpu_mask_*() and k_thread_cpu_pin() calls on the thread returned by k_work_queue_thread_get() do exactly that. 

> interrupt

IRQ affinity now exists per interrupt controller added in 795177248a5

closed 2026-11-02 by nashif