zephyrproject-rtos/zephyr #36237: fs_open returns 0 on existing file with FS_O_CREATE | FS_O_WRITE

ddkn, 2021-06-14:
Hello,

I might be a little confused when using fatfs and calling `fs_open` with options `FS_O_CREATE | FS_O_WRITE`, it generates a status of `0`, regardless if the file exists or not? As the documentation claims to report a `-ERRNO` depending on the state, technically it should report a `-EEXISTS` if I am trying to create a file that already exists?.  Should it not mimic what the [elm FatFs](http://elm-chan.org/fsw/ff/doc/open.html) driver does with `FR_EXIST` from `f_open`.

I am unsure if this is expected because how littlefs functions.

**Expected behavior**
Expecting a different status when trying to open an existing file versus a non-existent file.

**Impact**
This causes some difficulty in determining if this is the newest file, unless I write a wrapper on exploring the entry from listing directories.

de-nordic, 2021-06-14:
`FS_O_CREATE` means that file will be created when it does not exist, it passes `FA_OPEN_ALWAYS` flag to FAT driver, which, in case of Elm FAT driver, means to open existing or create a new file.

ddkn, 2021-06-15:
Hi @de-nordic ,

Thanks for getting back to me! So there isn't a way using the fatfs driver to determine if a file exists? As I do not want to overwrite a file, meaning I need to look at the entries from a directory list, as in the `sample/subsys/fs/fatfs` to see if a file exists?

Or can I use the Elm FAT driver directly without going through the Zephyr framework?

Thanks!

de-nordic, 2021-06-15:
@ddkn You may use `fs_stat(...)` that will fail when entry does not exist, if you are not interested in entry type this is sufficient test.

de-nordic, 2021-06-15:
The use of Elm FAT driver is a development risk, because it is not warranted that this is/will be backend used by FAT FS support layer within Zephyr; the VFS layer within Zephyr is supposed to provide uniform access to file systems, regardless of implementation details, and hide the details from applications.
This may also require some, rather not encouraged, access to internal structures of Zephyr VFS.

galak, 2021-06-15:
please re-open if this is an actual issue.

closed 2021-06-15 by galak

ddkn, 2021-06-15:
No thanks, I appreciate the feedback. I was confused about the operation, but wasn't sure what the alternative was.

Cheers,