zephyrproject-rtos/zephyr #36238: net_if.c: possible mutex deadlock

dayjaby, 2021-06-14:
**Describe the bug**
Configuring the shell and the IEEE 802.15.4 device "DW1000" in my prj.conf , I stumbled upon a forever waiting mutex in `iface_router_find_default` when asking in the shell for `$ net iface`.

Sadly, in net_if.c any of the logging tools seem to be not working. I used gdb to trace the error back to the `k_mutex_lock` line. Using a timeout of `K_MSEC(1000)` instead of `K_FOREVER` helped with this issue.

**To Reproduce**
Steps to reproduce the behavior:
```
west build -p auto -b nucleo_f446re zephyr-app && west flash
screen /dev/ttyUSB0 115200 8N1
net iface
```

**Expected behavior**
That `iface_router_find_default` returns in a reasonable amount of time without having to rely on timeouts for the mutexes.

**Logs and console output**
```
[00:00:00.012,000] <dbg> net_pkt.net_pkt_write: (main): pkt 0x2000677c data 0x8019fc0 length 16
[00:00:00.012,000] <dbg> net_pkt.net_pkt_write: (main): pkt 0x2000677c data 0x200072c0 length 40
[00:00:00.012,000] <dbg> net_pkt.net_pkt_skip: (main): pkt 0x2000677c skip 40
[00:00:00.018,000] <dbg> net_pkt.net_pkt_skip: (main): pkt 0x2000677c skip 40
[00:00:00.027,000] <inf> dw1000: IRQ work handler: RX TO
[00:00:00.028,000] <inf> dw1000: IRQ work handler: TXFRS
[00:00:00.028,000] <dbg> net_pkt.net_pkt_unref_debug: (main): TX [5] pkt 0x2000677c ref 0 frags 0x20006940 (ieee802154_send():303)
[00:00:00.028,000] <dbg> net_pkt.net_pkt_unref_debug: (main): TDATA (?) [0] frag 0x20006940 ref 0 frags (nil) (ieee802154_send():303)
[00:00:00.028,000] <dbg> net_pkt.net_pkt_frag_unref_debug: (main): TDATA (?) [0] frag 0x20006940 ref 0 (net_pkt_unref_debug():586)
--- 3 messages dropped ---
[00:00:01.023,000] <dbg> net_pkt.net_pkt_alloc_buffer_debug: (sysworkq): Data allocation maximum size 64 (requested 20)
[00:00:01.023,000] <dbg> net_pkt.pkt_alloc_buffer: (sysworkq): TDATA (?) [0] frag 0x20006940 ref 1 (net_ipv6_send_rs():1985)
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x200072c0 length 40
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 40
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x200072e8 length 4
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 4
[00:00:01.023,000] <dbg> net_pkt.net_pkt_memset: (sysworkq): pkt 0x2000677c byte 0 amount 4
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x2000570c length 2
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x20000204 length 8
[00:00:01.023,000] <dbg> net_pkt.net_pkt_memset: (sysworkq): pkt 0x2000677c byte 0 amount 6
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x200072c0 length 40
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 40
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 8
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 32
[00:00:01.023,000] <dbg> net_pkt.net_pkt_write: (sysworkq): pkt 0x2000677c data 0x200072e8 length 4
[00:00:01.023,000] <dbg> net_pkt.net_pkt_skip: (sysworkq): pkt 0x2000677c skip 4
uart:~$ net iface

Interface 0x20000ba4 (IEEE 802.15.4) [1]
========================================
Link addr : E2:2A:0F:00:7D:F5:A9:3B
MTU       : 125
Flags     : AUTO_START,IPv6
IPv6 unicast addresses (max 2):
        fe80::e02a:f00:7df5:a93b autoconf preferred infinite
        2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 3):
        ff02::1
        ff02::1:fff5:a93b
        ff02::1:ff00:1
IPv6 prefixes (max 2):
        <none>
```
**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Zephyr SDK commit: 3d39f72a88b3100ac83e7c6c285ae567aca640d0

jukkar, 2021-06-14:
> Sadly, in net_if.c any of the logging tools seem to be not working.

I did not understand what do you mean by this, could you elaborate this more?

I do not have IEEE 802.15.4 networks and that hw available, have you tried to replicate the issue with some other board / bearer?

dayjaby, 2021-06-14:
> I did not understand what do you mean by this, could you elaborate this more?

@jukkar Neither `printk` nor `NET_DBG` or similar commands were outputted to the console/shell if put one line before this. My plan would have been to printk all locks/unlocks in this file to see if this mutex is still locked.

I got another STM board to try it. And I just tried thebasic SPI communications with an Arty A7/VexRiscV. I will report my results when I know more.

jukkar, 2021-06-15:
> Neither `printk` nor `NET_DBG` or similar commands were outputted to the console/shell if put one line before this. My plan would have been to printk all locks/unlocks in this file to see if this mutex is still locked.

That is indeed weird, I have had no problems to enable debugging in net_if.c after setting `CONFIG_NET_IF_LOG_LEVEL_DBG=y`

jukkar, 2021-06-15:
> I got another STM board to try it.

Thanks for the help. A lot of testing is done with Ethernet, so if you are able to replicate the issue with Ethernet, it helps to speed up any fixes.

closed 2021-08-29 by github-actions[bot]