zephyrproject-rtos/zephyr #36242: Zephyr Upstream + sdk-nrf BLE NUS SHELL LOG/CBPRINTF build problem.

cederom, 2021-06-14:
**Describe the bug**
Hello world :-) I am using Zephyr upstream v2.6.0 and BLE NUS SHELL from SDK-NORDIC v1.5.1. After updates to Zephyr and SDK_NORDIC I get log related build error blocker that did not show up before (it did build and work fine in that configuration bare metal zephyr and sdk-nordic). Problem seems to be related with logging and its triggered from nus code. Reverting to older releases of sdk-nrf and zephyr does not help (is anything cached and preserved between `west update`?). I am running out of ideas. Any hints would be much appreciated :-)

**To Reproduce**
1. Use bare metal Zephyr (v2.5.99, v2.6.0-rc{1,2,3}) and sdk-nordic master with no problem.
2. Update Zephyr to v2.6.0.
3. Update sdk-nordic master with `west update`.
4. Observe a secure boot commit from sdk-nordic depends on commit on sdk-zephyr fork (file does not exist in Zephyr upstream).
5. Revert sdk-nordic to v1.5.1.
6. Observe missing file problem gone but new problem with NUS build error.

**Expected behavior**
Code builds fine as before.

**Impact**
Blocker of firmware build.

**Logs and console output**
```
/xxx/zephyr/include/sys/cbprintf_internal.h:183:19: error: invalid use of undefined type 'struct bt_conn'
  183 |   *_Generic((arg) + 0, \
      |                   ^
/xxx/zephyr/include/sys/cbprintf_internal.h:270:3: note: in expansion of macro 'Z_CBPRINTF_STORE_ARG'
  270 |   Z_CBPRINTF_STORE_ARG(&_buf[_idx], _arg); \
      |   ^~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/sys/cbprintf_internal.h:283:2: note: in expansion of macro 'Z_CBPRINTF_PACK_ARG2'
  283 |  Z_CBPRINTF_PACK_ARG2(_pbuf, _pkg_len, _pkg_offset, _pmax, arg)
      |  ^~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/sys/util_loops.h:1075:2: note: in expansion of macro 'Z_CBPRINTF_PACK_ARG'
 1075 |  fixed_arg0(x)
      |  ^~~~~~~~~~
/xxx/zephyr/include/sys/util_loops.h:38:2: note: in expansion of macro 'Z_FOR_EACH_EXEC'
   38 |  z_call(2, x, fixed_arg0, fixed_arg1)
      |  ^~~~~~
/xxx/zephyr/include/sys/util_loops.h:23:53: note: in expansion of macro 'Z_FOR_LOOP_3'
   23 |     _57, _58, _59, _60, _61, _62, _63, _64, N, ...) N
      |                                                     ^
/xxx/zephyr/include/sys/util_loops.h:1078:2: note: in expansion of macro 'Z_FOR_EACH_ENGINE'
 1078 |  Z_FOR_EACH_ENGINE(Z_FOR_EACH_EXEC, sep, F, _, __VA_ARGS__)
      |  ^~~~~~~~~~~~~~~~~
/xxx/zephyr/include/sys/util_macro.h:371:2: note: in expansion of macro 'Z_FOR_EACH'
  371 |  Z_FOR_EACH(F, sep, REVERSE_ARGS(__VA_ARGS__))
      |  ^~~~~~~~~~
/xxx/zephyr/include/sys/cbprintf_internal.h:357:2: note: in expansion of macro 'FOR_EACH'
  357 |  FOR_EACH(Z_CBPRINTF_PACK_ARG, (;), __VA_ARGS__);\
      |  ^~~~~~~~
/xxx/zephyr/include/sys/cbprintf_internal.h:376:2: note: in expansion of macro 'Z_CBPRINTF_STATIC_PACKAGE_GENERIC'
  376 |  Z_CBPRINTF_STATIC_PACKAGE_GENERIC(packaged, inlen, outlen, \
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/sys/cbprintf.h:133:2: note: in expansion of macro 'Z_CBPRINTF_STATIC_PACKAGE'
  133 |  Z_CBPRINTF_STATIC_PACKAGE(packaged, inlen, outlen, \
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/logging/log_msg2.h:247:3: note: in expansion of macro 'CBPRINTF_STATIC_PACKAGE'
  247 |   CBPRINTF_STATIC_PACKAGE(_msg->data, _plen, \
      |   ^~~~~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/logging/log_msg2.h:411:3: note: in expansion of macro 'Z_LOG_MSG2_STACK_CREATE'
  411 |   Z_LOG_MSG2_STACK_CREATE(_domain_id, _source, _level, _data, \
      |   ^~~~~~~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/logging/log_msg2.h:421:2: note: in expansion of macro 'Z_LOG_MSG2_CREATE2'
  421 |  Z_LOG_MSG2_CREATE2(_try_0cpy, _mode, UTIL_CAT(Z_LOG_FUNC_PREFIX_, _level), \
      |  ^~~~~~~~~~~~~~~~~~
/xxx/zephyr/include/logging/log_core.h:315:3: note: in expansion of macro 'Z_LOG_MSG2_CREATE'
  315 |   Z_LOG_MSG2_CREATE(UTIL_NOT(IS_ENABLED(CONFIG_USERSPACE)), _mode, \
      |   ^~~~~~~~~~~~~~~~~
/xxx/zephyr/include/logging/log_core.h:331:2: note: in expansion of macro 'Z_LOG2'
  331 |  Z_LOG2(_level, __log_current_const_data, __log_current_dynamic_data, __VA_ARGS__)
      |  ^~~~~~
/xxx/zephyr/include/logging/log.h:71:25: note: in expansion of macro 'Z_LOG'
   71 | #define LOG_DBG(...)    Z_LOG(LOG_LEVEL_DBG, __VA_ARGS__)
      |                         ^~~~~
/xxx/nrf/subsys/bluetooth/services/nus.c:39:2: note: in expansion of macro 'LOG_DBG'
   39 |  LOG_DBG("Data send, conn %p", conn);
      |  ^~~~~~~
[93/243] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/bluetooth/hci/h4.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /xxx/build
```

**Environment (please complete the following information):**
 - FreeBSD.
 - Zephyr upstream + SDK-NRF + GNU ARM GCC

**Additional context**
It turns out that SDK-ZEPHYR is tied to a nrfconnect / SDK-ZEPHYR local fork which seems to lag and diverge from Zephyr upstream. I would prefer to work on Zephyr Upstream directly and only use Nordic stuff as modules. It worked before that way. In perfect world probably the BLE NUS SHELL code could be part of Zephyr Upstream or vendor specific modules could be Zephyr Upstream compatible.. but that may not going to happen [1] [2].
[1] https://devzone.nordicsemi.com/f/nordic-q-a/76244/nrfconnect-sdk-nrf-modules-trusted-firmware-m-kconfig-and-kconfig-tfm-nordic-sdk-not-compatible-with-zephyr-upstream
[2] https://github.com/zephyrproject-rtos/zephyr/issues/35075

cederom, 2021-06-15:
Allright, so the v1.5.1 sdk-nrf does not build with upstream Zephyr. Later master can build BLE NUS SHELL with upstream Zephyr, until TFM is introduced that depends on sdk-zephyr fork. Will try to find the NUS build fix and TFM commit that prevents build with upstream Zephyr.

de-nordic, 2021-06-15:
The NRF fork of Zephyr (sdk-zephyr) is updated quite regularly from upstream Zephyr and will probably soon be updated from up to version 2.6.0. Unfortunately NRF sdk is not warranted to work with upstream Zephyr due to NRF specific features added or fixes that may not be yet available in upstream Zephyr.
All we can probably do is ask you for patience, and to wait for the next update to the sdk-zephyr.

carlescufi, 2021-08-16:
Closing this since it's nRF Connect SDK related. Thanks for the input!

closed 2021-08-16 by carlescufi