zephyrproject-rtos/zephyr #36247: samples: usb: testusb: Problems with using with cdc-acm

wngur300, 2021-06-15:
**Describe the bug**
I want to use 2 classes like **samples/subsys/usb/hid-cd**c (subsys/usb/class : **cdc_acm.c, loopback.c**)
In samples/usb/cdc-acm source, set CONFIG_USB_DEVICE_LOOPBACK=y option to prj.conf,
and tried loopback testing as shown in the document README.md in samples: usb: testusb, but it does not work.
(I tested the development board that I have as the host.)

**Environment (please complete the following information)**
OS: Linux ubuntu 18.04
Toolchain: zephyr-sdk-0.11.4
Commit SHA or Version used: 7a3b253

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_wb55rg ..
3. west flash

**Expected behavior**
I hope the port used as cdc-acm and the port used as loopback will behave normally.

**Logs and console output**  

Add LOOPBACK Device Settings in cdc_acm Sample :
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr CDC ACM sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y

CONFIG_USB_DEVICE_LOOPBACK=y
```
**loopback test (Not working properly):**
[image]

**When i tested the testusb sample [samples/ subsy/ usb/ testusb] (Normal operation):**
[image]

**Additional context**
I tested loopback by setting CONFIG_USB_DEVICE_LOOPBACK option in cdc-acm sample because I wanted to know if two usb classes were available like hid-cdc.
Is it impossible to receive data  from the host and use the /dev/ttyACM port?

jfischer-no, 2021-06-15:
> In samples/usb/cdc-acm source, set CONFIG_USB_DEVICE_LOOPBACK=y option to prj.conf,
> and tried loopback testing as shown in the document README.md in samples: usb: testusb, but it does not work.

It is not intended to work with testusb in this configuration.

closed 2021-06-15 by jfischer-no

wngur300, 2021-06-15:
> > In samples/usb/cdc-acm source, set CONFIG_USB_DEVICE_LOOPBACK=y option to prj.conf,
> > and tried loopback testing as shown in the document README.md in samples: usb: testusb, but it does not work.
> 
> It is not intended to work with testusb in this configuration.

> > In samples/usb/cdc-acm source, set CONFIG_USB_DEVICE_LOOPBACK=y option to prj.conf,
> > and tried loopback testing as shown in the document README.md in samples: usb: testusb, but it does not work.
> 
> It is not intended to work with testusb in this configuration.

Both cdc_acm and hid are available, but are both cdc_acm and loopback usb classes unavailable?

jfischer-no, 2021-06-15:
> Both cdc_acm and hid are available, but are both cdc_acm and loopback usb classes unavailable?

What do you mean by unavailable? Again it is not intended to work with testusb in this configuration, or what do you expect the testusb tool to do?

wngur300, 2021-06-24:
> What do you mean by unavailable? 

A USB driver created by Windows for communication allows data to be sent and received from boards containing sources of zephyr/subsys/usb/class/loopback.c.  

**loopback.c:**
[image]

**send test:**
[image]

There is an example of using two types: usb-hid and cdc_acm in zephyr/samples/subsys/usb/hid-cdc.
So I thought I could use both cdc_acm and loopback.c.

In Windows OS, data can be sent and received via loopback.c source, and I want to use COM PORT by adding cdc_acm.c source.
However, COM PORT is not made.
Two things cannot be used together: cdc_cam.c and loopback.c.

I'm guessing that the USB driver made for communication is not allowed to create another COM PORT, but I asked you to hear your opinion about this situation.