zephyrproject-rtos/zephyr #36250: tests/subsys/cpp/cxx - doesn't compile on native_posix when CONFIG_EXCEPTIONS=y

avigreen1978, 2021-06-15:
**Describe the bug**
When compiling c++ code for `native_posix` board with `CONFIG_EXCEPTIONS=y`, the following error is produced:
```
c++: error: /crtbegin.o: No such file or directory
c++: error: /crtend.o: No such file or directory
ninja: build stopped: subcommand failed.
```
The source of this problem is that no cmake file sets `LIBGCC_DIR` value

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/subsys/cpp/cxx
2. Add `CONFIG_EXCEPTIONS=y`, `CONFIG_LIB_CPLUSPLUS=y`
3. west build -b native_posix
4. See error

**Expected behavior**
Some cmake file before should set the `LIBGCC_DIR` value

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04

stephanosio, 2021-06-15:
cc @tejlmand

stephanosio, 2021-06-15:
It looks like `${LIBGCC_DIR}` is not set/valid for the `native_posix` target.
https://github.com/zephyrproject-rtos/zephyr/blob/08dfe74c15a8e9d5afa3aa3cb0713e5d894018fa/cmake/linker/ld/target.cmake#L16-L24

PR #36262 (2021-06-16): Fix native_posix build when C++ exception handling is enabled
This series fixes the broken `native_posix*` build when C++ exception handling is enabled (`CONFIG_EXCEPTIONS=y`).

The actual fix is limited to the [last commit](https://github.com/zephyrproject-rtos/zephyr/commit/b428a3ce718ea3fa036d203e85671275065ad42a), but I've found a few other things that require fixing while investigating this issue, so I'm including them here as well.

Fixes #36250.
closed 2021-06-16 by aescolar