zephyrproject-rtos/zephyr #36251: RFC: API Change: Bluetooth callback register

LingaoM, 2021-06-15:
### Problem description

The current cb register is based on the dynamic connection of the linked list,
and there is no need to use the dynamic at present.

### Proposed change

Change the cb register to the static mode of zsection, and save part of the RAM.

## Detailed RFC

This change will introduce `Z_ITERABLE_SECTION_ROM` to replace `sys_slist`.

### Proposed change (Detailed)

1. Change `bt_conn_cb_register` to `BT_CONN_CB_DEF`. see https://github.com/zephyrproject-rtos/zephyr/pull/36119
2. Change `bt_le_per_adv_sync_cb_register` to `BT_LE_PER_ADV_SYNC_CB_DEF`.
3. Change `bt_le_scan_cb_register` to `BT_LE_SCAN_CB_DEF`.
4. Change `bt_gatt_cb_register` to `BT_GATT_CB_DEF`.

### Concerns and Unresolved Questions

Is there a possibility of unregister?

## Alternatives

Consider whether to coexist with old style, and adding macros is the old way,
which is disabled by default.

joerchan, 2021-06-15:
I would say adding a static definition in addition would be the best approach. There might be reasons for applications wanting to have dynamic allocation, in addition to that not being a breaking change.

This would be similar to GATT services, which have `BT_GATT_SERVICE_DEFINE` and `bt_gatt_service_register`

Thalley, 2022-02-14:
@LingaoM is this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/36119?

LingaoM, 2022-02-14:
Fixed

closed 2022-02-14 by LingaoM