zephyrproject-rtos/zephyr #36254: Zephyr shell subsystem not work with ARC hardware boards.

IRISZZW, 2021-06-15:
**Describe the bug**
Zephyr shell subsystem not work with ARC hardware boards.

**To Reproduce**
Steps to reproduce the behavior with samples/hello_world :
1. enable shell for hello_world example (add CONFIG_SHELL=y in prj.conf file)
2. west build -b iotdk -d build samples/hello_world
3. west flash
4. See error

**Logs and console output**
When build and run this for `iotdk` and `hsdk`, there is no output and prompt. the CPU is loop in idle thread.

When build and run for `nsim_em` or `nsim_hs`, it works fine:

```
west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner arc-nsim
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.6.0-307-ge09cae9585a9  ***
Hello World! nsim

uart:~$ help
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or its subcommands.
You can try to call commands with <-h> or <--help> parameter for more information.

Shell supports following meta-keys:
  Ctrl + (a key from: abcdefklnpuw)
  Alt  + (a key from: bf)
Please refer to shell documentation for more details.

Available commands:
  clear    :Clear screen.
  device   :Device commands
  help     :Prints the help message.
  history  :Command history.
  kernel   :Kernel commands
  resize   :Console gets terminal screen size or assumes default in case the
            readout fails. It must be executed after each terminal width change
            to ensure correct text display.
  shell    :Useful, not Unix-like shell commands.
uart:~$

```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA: be956927446f84a74f4e5568b576d7b21a47ef08

ruuddw, 2021-06-15:
Don't forget to set the CONFIG_CONSOLE_GETLINE=y (or GETCHAR) as well, to ensure via CONFIG_CONSOLE_HANDLER=y callbacks for handling interrupts are enabled.

abrodkin, 2021-09-07:
It turned-out to be not a generic ARC issue, but rather 2 separate problem with 2 boards:
1. HSDK, see https://github.com/zephyrproject-rtos/zephyr/issues/37862
2. IoTDK, see https://github.com/zephyrproject-rtos/zephyr/pull/37707

closed 2021-09-07 by abrodkin