zephyrproject-rtos/zephyr #36256: SPI4 & 3 MISO not working on nRF5340

koffes, 2021-06-15:
We have used SPI4 on the nRF5340 for a year+ with no issues. With a recent Zephyr upgrade, the CLK, CS, and MOSI works fine, but the data coming in on MISO is not registered in Zephyr.

After a lot of bisecting, the lines which introduced the issue comes from nrfx:
https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/hal/nrf_gpio.h#L580

This used to be: 
```
 reg->PIN_CNF[pin_number] = ((uint32_t)dir << GPIO_PIN_CNF_DIR_Pos)
                               | ((uint32_t)input << GPIO_PIN_CNF_INPUT_Pos)
                               | ((uint32_t)pull << GPIO_PIN_CNF_PULL_Pos)
                               | ((uint32_t)drive << GPIO_PIN_CNF_DRIVE_Pos)
                               | ((uint32_t)sense << GPIO_PIN_CNF_SENSE_Pos);
}
```

And this works fine.

Now however, SPI3 MISO does not work at all. I can get SPI4 MISO to work with the following lines added:

```
nrf_gpio_pin_mcu_select(8, NRF_GPIO_PIN_MCUSEL_PERIPHERAL); 
nrf_gpio_pin_mcu_select(9, NRF_GPIO_PIN_MCUSEL_PERIPHERAL);
nrf_gpio_pin_mcu_select(10, NRF_GPIO_PIN_MCUSEL_PERIPHERAL);
```
Then MISO works for pin reset, but not debug reset it seems.

I'd appreciate if you could have a look  @anangl. I do realize that we likely need to reproduce this on a 5340DK, as we are now on custom HW.

carlescufi, 2021-06-15:
@koffes is this a duplicate of #35945?

anangl, 2021-06-15:
> @koffes is this a duplicate of #35945?

@carlescufi I believe it is not. In #35945, there is no output (on SCK and MOSI), and the problem here is the lack of input data (from MISO).

koffes, 2021-06-15:
@carlescufi they may be related but not a duplicate.

koffes, 2021-06-17:
@anangl  I have tried to replicate on both latest main (master) : d6949b75a72c7deda5da4a99208903f1feac36ee
as well as the commit where we first discovered the issue: 6b2cc7e57ea82c507409e4a1f4ff3613558bf5bd

I have used drivers/spi/spi_loopback and added the following:
[spi4_diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/6670119/spi4_diff.txt)
[nrf5340dk_nrf5340_cpuapp.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/6670123/nrf5340dk_nrf5340_cpuapp.conf.txt)

I added a jumper loopback between P1.13 and P1.14 and tests passed on both hashes.

koffes, 2021-06-18:
We are able to reproduce. Seems like a nrfx issue.

koffes, 2021-06-25:
So, the issue is that https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf5340dk_nrf5340/nrf5340_cpunet_reset.c#L44-L45
Gives pin 10 to the NET core if NET is enabled. This used to make no difference, as this was overwritten. Though, with this change in nrfx: https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/hal/nrf_gpio.h#L580-L592
this information is retained and MISO does not register any data.

closed 2021-09-08 by github-actions[bot]