zephyrproject-rtos/zephyr #36263: up_squared: kernel.memory_protection.mem_map.x86_64 failed.

chen-png, 2021-06-15:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/mem_protect/mem_map/
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build zephyr-v2.6.0-248-gc9aa260f0cfc
Running test suite test_mem_map

START - test_z_phys_map_rw
E: Page fault at address 0x2ffe04d6 (error code 0x3)
E: Access violation: supervisor thread not allowed to write
E: PTE: 0x2ffe0000 -> 0x000000000010d000: A XD 
E: RAX: 0x000000002ffe04d6 RBX: 0x000000000000138b RCX: 0x000000002ffe04d6 RDX: 0x000000000000138a
E: RSI: 0x000000000010ac7e RDI: 0x000000000010a821 RBP: 0x000000000059b3c0 RSP: 0x000000000059b3a0
E:  R8: 0x000000000010a410  R9: 0x000000000010ac65 R10: 0x0000000000112cc0 R11: 0x0000000000000001
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000059b3a0 RFLAGS: 0x0000000000010246 CS: 0x0018 CR3: 0x00000000007b0000
E: RIP: 0x000000000010092b
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10fec0 (test_z_phys_map_rw)
Caught system error -- reason 0
PASS - test_z_phys_map_rw in 0.80 seconds

......
mapped a page: 0x2ffd7000 - 0x2ffd8000
trying to access 0x2ffd8008
E: Page fault at address 0x2ffd8008 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000000001 RBX: 0x000000002ffd7000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x000000000000000a RDI: 0x0000000000598cc0 RBP: 0x000000000059b3c0 RSP: 0x000000000059b3b0
E:  R8: 0x0000000000000001  R9: 0x0000000000000000 R10: 0x00000000ffffffff R11: 0x0000000000000032
E: R12: 0x00000000001108e0 R13: 0x0000000000000000 R14: 0x00000000001108e0 R15: 0x0000000000000000
E: RSP: 0x000000000059b3b0 RFLAGS: 0x0000000000010247 CS: 0x0018 CR3: 0x00000000007b0000
E: RIP: 0x00000000001006e0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10fec0 (test_k_mem_map_guard_after)
Caught system error -- reason 0
PASS - test_k_mem_map_guard_after in 0.89 seconds

**START - test_k_mem_map_exhaustion**
Free memory: 2139410432
E: insufficient virtual address space (requested 12288)
Mapped 64865 pages
Free memory now: 1873723392
E: Page fault at address 0xffffffffffffffff (error code 0x0)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to read
E: PML4E: not present
E: RAX: 0x0000000000000001 RBX: 0x000000000000fd61 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x0000000000001fb0 RDI: 0x0000000000598d30 RBP: 0x000000000059b3c0 RSP: 0x000000000059b390
E:  R8: 0x0000000000000001  R9: 0x0000000000001fb0 R10: 0x0000000000118b80 R11: 0x0000000000000000
E: R12: 0x000000007f84d000 R13: 0xffffffffffffffff R14: 0x00000000000000c4 R15: 0xffffffffffffffff
E: RSP: 0x000000000059b390 RFLAGS: 0x0000000000010286 CS: 0x0018 CR3: 0x00000000007b0000
E: RIP: 0x0000000000100374
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10fec0 (test_k_mem_map_exhaustion)
**Caught system error -- reason 0
Unexpected fault during test**

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: Zephyr-sdk-0.12.4
 - Commit ID: c9aa260f0cfc

chen-png, 2021-08-30:
debug and found that it was ok to use k_mem_map to map all available virtual memory, but it failed at using k_mem_unmap to free those mapped pages before, because when **running k_mem_unmap, the addr of 0x99d000 points to an invalid 0xffffffffffffffff**.

>current addr: 0x991000 --> last addr: 0x994000
current addr: 0x994000 --> last addr: 0x997000
current addr: 0x997000 --> last addr: 0x99a000
current addr: 0x99a000 --> last addr: 0x99d000
**current addr: 0x99d000 --> last addr: 0xffffffffffffffff**
E: Page fault at address 0xffffffffffffffff (error code 0x0)

in the code below, we used current mapped addr points to last mapped addr, it likes a link list which will be used to find address for free pages.
>*((uintptr_t *)addr) = POINTER_TO_UINT(last_mapped)

**so the addr of 0x99d000 should point to 0x9a0000, rather than 0xffffffffffffffff**, I am trying to find out why it occurred on up_squared,
and @dcpleung  do you have any suggestions about the possible problems, then I can dig into it, thanks.

chen-png, 2021-09-06:
tested on qemu_x86 with same memory size configuration on up_squared, but it was passed, it could support map more pages.
SRAM_SIZE: 2G    KERNEL_VM_SIZE: 1G

>START - test_k_mem_map_exhaustion
Free memory: 2141577216
E: insufficient virtual address space (requested 12288)
Mapped 86895 pages
Free memory now: 1785655296
Unmapped 86895 pages
Free memory now: 2141577216
 PASS - test_k_mem_map_exhaustion in 4.462 seconds
===================================================================
Test suite test_mem_map succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

@dcpleung is this related to SRAM hardware on up_squared? because I think the code logic is correct.

dcpleung, 2021-09-07:
I can't see anything wrong by skimming through the code. Will need to debug to see what is going on.

chen-png, 2021-09-10:
@dcpleung I used code below to print each mapped page info: virtual address --> mapped physical address and stored value in this address
 >int level = 0;
        pentry_t val = 0;
for (addr = last_mapped; addr <= start_addr; addr += 0x3000) {
 z_x86_pentry_get(&level, &val, z_x86_kernel_ptables, (void *)addr);
printk(" virtual addr: %p --> physical addr: %llx, value: %p\n", addr, val & 0x000FFFFFFFFFF000, *((uintptr_t *)addr));
}

and I found that virtual address from  0x2ffd4000 to  0x1711000, they are mapped successfully and the stored value is correct,
but ****starting from virtual address 0x170e000, the stored value is incorrect, it's 0xffffffffffffffff, I guess this value means that we don't have access to this physical address, so any write operation is invalid and reads will return all '1'.**

because according to doc of up_squared, it describes as below:
**This is the range of addresses presented to the memory controller after all security
checks and reclaiming has been done. The memory controller is aware of the total
amount of populated DRAM and addresses above this range are considered invalid –
writes will be dropped and reads will return all ‘1’s.**

I tried to set SRAM_SIZE as 0x10000000(256M), then this test was passed. but I don't think this is the root cause, I can't find any describe about that there is a gap in ram of up_squared, do you have any idea on this? thanks.

 >virtual addr: 0x16f6000 --> physical addr: 0x10008000, value: 0xffffffffffffffff
 virtual addr: 0x16f9000 --> physical addr: 0x10007000, value: 0xffffffffffffffff
 virtual addr: 0x16fc000 --> physical addr: 0x10006000, value: 0xffffffffffffffff
 virtual addr: 0x16ff000 --> physical addr: 0x10005000, value: 0xffffffffffffffff
 virtual addr: 0x1702000 --> physical addr: 0x10004000, value: 0xffffffffffffffff
 virtual addr: 0x1705000 --> physical addr: 0x10003000, value: 0xffffffffffffffff
 virtual addr: 0x1708000 --> physical addr: 0x10002000, value: 0xffffffffffffffff
 virtual addr: 0x170b000 --> physical addr: 0x10001000, value: 0xffffffffffffffff
 **virtual addr: 0x170e000 --> physical addr: 0x10000000, value: 0xffffffffffffffff**
 virtual addr: 0x1711000 --> physical addr: 0xffff000, value: 0x1714000
 virtual addr: 0x1714000 --> physical addr: 0xfffe000, value: 0x1717000
 virtual addr: 0x1717000 --> physical addr: 0xfffd000, value: 0x171a000
.....
virtual addr: 0x2ffce000 --> physical addr: 0x7c0000, value: 0x2ffd1000
 virtual addr: 0x2ffd1000 --> physical addr: 0x7bf000, value: 0x2ffd4000
 virtual addr: 0x2ffd4000 --> physical addr: 0x7be000, value: (nil)^M

dcpleung, 2021-09-10:
Trying booting Linux and look at dmesg. There should be a e820 map which tells you which regions are usable.

chen-png, 2021-09-14:
@dcpleung I boot ubuntu on up_squared and got the e820 map info in log as below, I found that the ram address from **0x10000000-0x12150fff is reserved**, it should means that this address range couldn't be used, but in the **apollo_lake.dtsi, the ram is a continuous range from 0x0 to DT_DRAM_SIZE(0x80000000).**
 
>installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
**installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved**
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000077b0cfff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000077b0d000-0x0000000079c2afff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000079c2b000-0x0000000079c43fff] ACPI data
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000079c44000-0x0000000079ca3fff] ACPI NVS
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000079ca4000-0x000000007a040fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a041000-0x000000007a08cfff] type 20
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a08d000-0x000000007a3fafff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a3fb000-0x000000007a3fbfff] ACPI NVS
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a3fc000-0x000000007a425fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a426000-0x000000007a964fff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a965000-0x000000007a966fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007a967000-0x000000007affffff] usable
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
installer/syslog:Apr  8 15:47:48 ubuntu kernel: [    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable

dcpleung, 2021-09-14:
We don't have the mechanism to deal with memory holes at this point. Limiting memory to below 0x100000000 is the only way for now to pass the test.

PR #38547 (2021-09-16): tests: mem_map: limit memory below 0x10000000 on up_squared.
there is a memory hole from address 0x10000000-0x12150fff
in the ram on up_squared, we don't have access to read/write
this range, so limit the memory range below 0x10000000.

fix: #36263
closed 2021-09-16 by nashif