zephyrproject-rtos/zephyr #36270: TF-M: introduce uniformity in Non-Secure target names

ioannisg, 2021-06-15:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, the names we choose for the non-secure version of TrustZone-enabled platforms have different suffix versions:
All `_ns`, `ns`, `_nonsecure` are used.

**Describe the solution you'd like**
Ideally, we should pick one option to use in Zephyr and force this as a convention.

We should probably do this for the LTS2.

ioannisg, 2021-06-15:
@microbuilder @galak does this sound reasonable?

Laczen, 2021-06-16:
As proposed on slack: Could we change the way s/ns are defined: as an example for the cpuapp in nrf5340:

a. We start by defining cpuapp (and "board" for processing by Kconfig.defconfig): In the dts of this board `sram0_s` is defined as the whole `sram0` region (excluding the `ipc_shm` region),
`flash0_s` is defined as the  whole `flash0_s` region,
chosen is: `zephyr, sram = sram0_s`, `zephyr, flash = flash0_s`.

b. from the cpuapp we derive cpuapp_s by:
deleting node `sram0_s` and splitting it up in `sram0_s` and `sram0_ns`
deleting node `flash0_s` and splitting it up in `flash0_s` and `flash0_ns`
chosen is: `zephyr, sram = sram0_s`, `zephyr, flash = flash0_s`

c. from cpuapp_s cpuapp_ns is derived by setting chosen: `zephyr, sram = sram0_ns`, `zephyr, flash = flash0_ns`

To integrate mcuboot (or a different bootloader) a similar method is used: cpuapp_mcuboot deletes node flash0_s and redefines it as mcuboot, slot0, slot1, scratch. And similar for cpuapp_s_mcuboot, cpuapp_ns_mcuboot.

microbuilder, 2021-06-24:
My proposal would be to standardise on `_ns`, since it's brief but clear, and the underscore helps differentiate it from the underlying MCU, core or board name.

galak, 2021-06-28:
`_ns` sounds good to me as well.

PR #36780 (2021-07-10): boards: arm: Standardise naming of NS boards
This PR renames all non-secure board targets to make use of a standard `_ns` suffix.

Boards Concerned:

- `bl5340_dvk` (`bl5340_dvk_cpuappns` -> `bl5340_dvk_cpuapp_ns`)
- `lpcxpresso55s69` (`lpcxpresso55s69_ns` already good)
- `mps2_an521` (`mps2_an521_nonsecure` -> `mps2_an521_ns`)
- `musca_b1` (`v2m_musca_b1_nonsecure` -> `v2m_musca_b1_ns`)
- `musca_s1` (`v2m_musca_s1_nonsecure` -> `v2m_musca_s1_ns`)
- `nrf5340dk_nrf5340` (`nrf5340dk_nrf5340_cpuappns` -> `nrf5340dk_nrf5340_cpuapp_ns`)
- `nrf9160dk_nrf9160` (`nrf9160dk_nrf9160ns` -> `nrf9160dk_nrf [... cut]
closed 2021-07-10 by nashif