zephyrproject-rtos/zephyr #36282: Overwrite mode for RTT logging

kurddt, 2021-06-16:
Currently even if the SEGGER RTT "stack" allows it it is not possible to select the overwrite mode for RTT logging. In this mode instead of skipping logging when the buffer is full the oldest data is replaced.

I would like to be able to select this mode using a new Kconfig option (`LOG_BACKEND_RTT_MODE_OVERWRITE`) for logger behavior choice https://docs.zephyrproject.org/latest/reference/kconfig/choice_391.html#choice-391 

No alternative is possible to obtain the same behavior

carlescufi, 2021-06-16:
@kurddt would you be able to send a Pull Request for this?

kurddt, 2021-06-16:
I'm not that familiar with the logging subsystem nor the SEGGER internals. If I find some time to do so before somebody else do, I will. But I have a couple of device driver I want to contribute first

PR #37795 (2021-09-28): logging: rtt: Add overwrite mode
In this mode the oldest data is overwritten
when the buffer is full

Closes #36282
commit a55a5b74 (2021-09-28): logging: rtt: Add overwrite mode

In this mode the oldest data is overwritten
when the buffer is full

Closes #36282
closed 2021-09-28 by carlescufi