zephyrproject-rtos/zephyr #36289: eswifi gets a deadlock on b_l4s5i_iot01a target

oopsmonk, 2021-06-16:
**Describe the bug**

[samples/net/wifi](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.6.0/samples/net/wifi) dosen't work on `b_l4s5i_iot01a` target. 
the program hangs after wifi related command, like `wifi scan` and `wifi connect`.

**To Reproduce**
Steps to reproduce the behavior:
1. cp -r samples/net/wifi ~/my-wifi-test && cd ~/my-wifi-test
2. west build -b b_l4s5i_iot01a
3. west flash
4. shell is hung after `wifi scan` command
```
[00:00:00.000,000] <dbg> wifi_eswifi.eswifi_init: 
[00:00:00.000,000] <dbg> wifi_eswifi.eswifi_spi_init: success
[00:00:00.000,000] <dbg> net_core.net_init: (main): Priority 90
[00:00:00.000,000] <dbg> net_core.l3_init: (main): Network L3 init done
[00:00:00.000,000] <dbg> net_if.net_if_init: (main): 
[00:00:00.000,000] <dbg> net_if.init_iface: (main): On iface 0x2000058c
[00:00:00.000,000] <dbg> wifi_eswifi.eswifi_iface_init: 
[00:00:00.510,000] <dbg> wifi_eswifi.eswifi_spi_request: cmd=(nil) (0 byte), rsp=0x20000965 (1600 byte)
[00:00:01.000,000] <dbg> wifi_eswifi.eswifi_spi_read_msg: 
uart:~$ wifi scan

```
5. using `west debug`, the code calls `k_mutex_lock(&eswifi->mutex, K_FOREVER)` at `eswifi_lock` waiting for an unlock `in eswifi_mgmt_scan` function.
```c
if (atomic_get(&eswifi->mutex_owner) != (atomic_t)(uintptr_t)_current) {
		k_mutex_lock(&eswifi->mutex, K_FOREVER);
		atomic_set(&eswifi->mutex_owner, (atomic_t)(uintptr_t)_current);
		eswifi->mutex_depth = 1;
} else {
```

**Expected behavior**

The output should be the same as the example document.

```
shell> wifi scan
Scan requested
shell>
Num  | SSID                             (len) | Chan | RSSI | Sec
1    | kapoueh!                         8     | 1    | -93  | WPA/WPA2
2    | mooooooh                         8     | 6    | -89  | WPA/WPA2
3    | Ap-foo blob..                    13    | 11   | -73  | WPA/WPA2
4    | gksu                             4     | 1    | -26  | WPA/WPA2
----------
Scan request done
```

**Impact**
The WiFi connection doesn't work.

**Logs and console output**
see the step 4 in reproduce section.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: zephyr-sdk-0.12.3
 - Zephyr version: [v2.6.0](https://github.com/zephyrproject-rtos/zephyr/tree/v2.6.0) and main(d7368995048990744c6583f28a6f31078d225e94)

**Additional context**

A deadlock happens when calling `wifi scan` from the shell because of `eswifi_iface_init` doesn't release the mutex. 
`eswifi_iface_init` locks the mutex waiting for initialization steps but `eswifi_reset` never returned.
`eswifi_spi_request` which is called by `eswifi_reset` keeps waiting for `eswifi_spi_cmddata_ready`.
```
Run till exit from #0  eswifi_spi_request (eswifi=0x200007f0 <eswifi0>, cmd=<optimized out>, clen=<optimized out>, rsp=0x20000965 <eswifi0+373> "", rlen=1600) 
at zephyr/drivers/wifi/eswifi/eswifi_bus_spi.c:158
```

erwango, 2021-06-17:
@loicpoulain Would you have time to have a look ?

erwango, 2021-06-17:
@nandojve Would you have time to have a look ?

loicpoulain, 2021-06-17:
> A deadlock happens when calling wifi scan from the shell because 

Doesn't it timeout after a certain time?

Look like the 'data ready' pin is probably not correctly configured on that board, driver is waiting for that pin to be high.

oopsmonk, 2021-06-17:
> > A deadlock happens when calling wifi scan from the shell because
> 
> Doesn't it timeout after a certain time?
> 
> Look like the 'data ready' pin is probably not correctly configured on that board, driver is waiting for that pin to be high.

how long is the timeout? I waited for at least 2 minutes. 
where I can change it?

oopsmonk, 2021-06-18:
> It is normally about 1min: eswifi_spi_wait_cmddata_ready
> But because we do a lot of short ksleep I would not be surprised it take much more time.
> 
> Anyway, even if you fix the timeout that will not solve the root cause, that the driver-data pin value is not correct, but that would be good to identify which call to eswifi_spi_wait_cmddata_ready is failing.

the timeout didn't fail but the code stuck in the while loop which it's no timeout.
https://github.com/zephyrproject-rtos/zephyr/blob/79a6c07536bc14583198f8e3555df6134d8822cf/drivers/wifi/eswifi/eswifi_bus_spi.c#L166-L170

the device tree looks right, not sure if I miss something? 
https://github.com/zephyrproject-rtos/zephyr/blob/79a6c07536bc14583198f8e3555df6134d8822cf/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts#L146-L155

loicpoulain, 2021-06-18:
> <&gpioe 0 GPIO_ACTIVE_HIGH>;

I think this is wrong, chip-select is usually active-low, try to set GPIO_ACTIVE_LOW instead.

oopsmonk, 2021-06-18:
> > <&gpioe 0 GPIO_ACTIVE_HIGH>;
> 
> I think this is wrong, chip-select is usually active-low, try to set GPIO_ACTIVE_LOW instead.

setting data-gpios to GPIO_ACTIVE_LOW, I got `DATA ready timeout` from eswifi_spi_wait_cmddata_ready.
```
[00:01:06.510,000] <err> wifi_eswifi: DATA ready timeout
[00:01:06.510,000] <err> wifi_eswifi: Unable to reset device
[00:01:06.510,000] <dbg> net_if.net_if_post_init: (main): 
[00:01:06.510,000] <dbg> net_if.net_if_up: (main): iface 0x2000058c
```

according to [GPIO API](https://docs.zephyrproject.org/latest/reference/peripherals/gpio.html#c.GPIO_ACTIVE_HIGH) is should set to `GPIO_ACTIVE_HIGH`, but it depends on SPI driver design, I think.

oopsmonk, 2021-06-18:
oh, I got it. I changed chip-select pin to `GPIO_ACTIVE_LOW`, it seems to work. let me do more tests.

https://github.com/zephyrproject-rtos/zephyr/blob/79a6c07536bc14583198f8e3555df6134d8822cf/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts#L134-L135

loicpoulain, 2021-06-21:
@oopsmonk, good, could you submit the fix if you confirm it works. CS is active low for eswifi chips.

erwango, 2021-06-21:
Thanks @loicpoulain for the great support!

PR #36422 (2021-06-21): boards: arm: b_l4s5i_iot01a: SPI chip-select is active low.
According to RM0432, the chip-select of SPI is active low.

Fixes #36289
commit a5fce202 (2021-06-21): boards: arm: b_l4s5i_iot01a: SPI chip-select is active low.

According to RM0432, the chip-select of SPI is active low.

Fixes #36289
closed 2021-06-21 by MaureenHelm