zephyrproject-rtos/zephyr #36294: TF-M: enhance the TF-M/Zephyr integration for Zephyr v2.7.0 release

ioannisg, 2021-06-16:
Enhance the TF-M integration with Zephyr, for the LTS2 release

- [ ] Upmerge TF-M to 1.4 when ready
- [ ] Improve TF-M documentation regarding the integration with Zephyr
       - Build and flash instructions, in all the boards
       - List of boards supported
       - Additional doc improvements
- [ ] Consolidate NS target naming
- [ ] FW update sample
- [ ] Complete Kconfig for TF-M configuration, reflecting latest added Tf-M CMake options
- [ ] Update MCUboot for TF-M, potentially adding a local path

microbuilder, 2021-06-16:
Thanks for putting this together.

I made a bunch of notes for myself for the TF-M presentation for ZDS 2021, and can recycle those to give people a starting point to understand TF-M: https://gist.github.com/microbuilder/53ac490db265e86e5840a9f752b3dc48

ioannisg, 2021-07-28:
just FYI @cfriedt , this is planned for 2.7.0

closed 2021-11-04 by microbuilder