zephyrproject-rtos/zephyr #36297: Move BSS section to the end of image

vladimir-podbrezsky, 2021-06-16:
**Is your enhancement proposal related to a problem? Please describe.**
A binary Zephyr image has a huge size because BSS section is inserted to this image. The BSS is placed between .text and .data sections. In case the BSS section is the last one in the image the size of binary file will be reduced because BSS section is not loadable.

yashi, 2021-06-17:
Care to add more detail? board, zephyr version, how you built?

vladimir-podbrezsky, 2021-06-20:
We use platform based on Cortex ARM Aarch64 and it's same as "bcm958402m2_a72".  The version of Zephyr is 2.6.0. 
Actually,  location of BSS section is defined in file "zephyr/include/arch/arm64/scripts/linker.ld".

yashi, 2021-06-21:
#24619, #21747

henrikbrixandersen, 2022-11-17:
@vladimir-podbrezsky, @yashi Can this be closed now that https://github.com/zephyrproject-rtos/zephyr/pull/21747 is merged?

yashi, 2022-12-12:
For `CONFIG_USERSPACE=n`, yes. For `CONFIG_USERSPACE=y`, please refere to #24619.  But since this one is a duplicate of them, we can close it.  @vladimir-podbrezsky, please reopen it if you need.  Thanks.

closed 2022-12-12 by yashi