zephyrproject-rtos/zephyr #36301: soc: cypress: Port Zephyr to Cypress CYW43907

giladchen, 2021-06-16:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
The Cypress CYW43907 SoC does not have support currently in Zephyr.

**Describe why you are asking for this support?**
I am developing code for the Cypress CYW943907 EV board. We would like that support will be added for it in Zephyr.

Is adding support to this SoC in progress? If not i would like my attempt at this task.

galak, 2021-06-16:
@giladchen not aware of anyone work on this, please feel free to go ahead.

closed 2021-06-16 by galak

k6dsp, 2022-11-29:
Hm. this marked as completed and even end up in release note for 2.7.0. Can someone point me to the implementation?