zephyrproject-rtos/zephyr #36324: add project groups to upsteam west manifest

temap, 2021-06-16:
## Introduction

Add project groups to upstream west manifest.

### Problem description

Currently, I have ~30 CI twister jobs that check out the Zephyr project. I don't need to check out and test the vendor's HAL and other libraries. I need only a Zephyr core for testing. Current checkout takes around 2.2GB of disk space.

I want to reduce the size of manifest checkout. 

Currently, I create a downstream manifest with `hal`, `debug`, `tee` and  `bsim_hw_models blocklists` :

```sh
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos/zephyr.git
  projects:
    - name: ${zephyr_repo_name}
      remote: zephyrproject-rtos
      revision: main
      import:
        path-blocklist:
          - modules/bsim_hw_models/nrf_hw_models
          - modules/debug/*
          - modules/hal/*
          - modules/tee/*
```

Then I execute the following commands:

```sh
west init -l .
west update -o=--depth=1000 zephyr
west update -o=--depth=1 -n
west update -o=--depth=1000 zephyr
```
As a result, I have to update the Zephyr repo twice to restore git history for the [twister JUnit report](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib.py#L2670).

### Proposed change

Add project hal, tee, debug, fs, crypto, audio, bsim_hw_models  and tools groups 
In this case, I could checkout required modules for testing with a shallow clone option:

```sh
west init -m https://github.com/temap/zephyr.git --mr rfc-west-groups-manifest
west -v update --group-filter=-hal,-tee,-bsim_hw_models,-debug -o=--depth=1 -n
```

With this change the source size could be reduced to ~900MB:
```sh
4.0K    ./.west
564M    ./zephyr
267M    ./modules
7.5M    ./bootloader
19M     ./tools
857M    .
```
## Detailed RFC

https://github.com/temap/zephyr/commit/ae2a4fef8dddc4f886c5838cbfd737434c9e5d6d

nashif, 2021-06-16:
@carlescufi  @mbolivar-nordic  FYI

mbolivar-nordic, 2021-06-16:
I agree this is a valid use case for west manifest project groups. The diff in the detailed RFC also has the advantage that results of a default `west init` + `west update` remain unchanged.

I think it's up to the TSC to decide what groups to apply to individual projects.

galak, 2021-06-16:
@mbolivar-nordic how are groups defined?

mbolivar-nordic, 2021-06-16:
https://docs.zephyrproject.org/latest/guides/west/manifest.html#project-groups-and-active-projects

temap, 2021-06-17:
I suggest breaking manifest into logic groups or by the vendor. It's okay to check out everything for development purpose once but not a great idea for CI.
I don't want to check out 1GB of stm32 hal when I don't use it. I am looking for something modular.

As an alternative, add a name-filter option in the west to disable checkout specific projects by name.

beriberikix, 2021-07-10:
Big fan of this improvement and the approach. We're coming from the angle of a new user getting started. Downloading everything adds meaningful time and tends to confuse our new users.

Instead of waiting for the TSC to propose groupings, maybe we can vote on the proposed list in the RFC?

Once this is upstream I would suggest including some mention of it in https://github.com/zephyrproject-rtos/example-application

carlescufi, 2021-07-14:
@temap after discussion in the TSC there seem to be no objections at the time for your proposal. Could you please post a Pull Request with the proposed groups so we can discuss it there?

temap, 2021-07-15:
@carlescufi done: https://github.com/zephyrproject-rtos/zephyr/pull/36981

Thanks

closed 2021-07-22 by temap