zephyrproject-rtos/zephyr #36343: Bluetooth: Mesh: Modularizing the proxy feature

LingaoM, 2021-06-17:
## Introduction

The Bluetooth proxy feature includes proxy client and proxy server. In addition to the proxy pdu message used above, pb-gatt also uses the same proxy pdu message.

Currently zephyr bluetooth mesh couples them in one file.

### Proposed change

I prefer to separate it in different files and refer to @trond-snekvik 's suggestions in https://github.com/zephyrproject-rtos/zephyr/pull/28422#discussion_r503856774.

## Detailed RFC

A file at the separation is called `gatt_services.c`, which is used to contain Mesh Provisioning Service and Mesh Proxy Service.
Another file in the separation is called `proxy_msg.c`, which is used to process Proxy pdu messages.

### Proposed change (Detailed)

```
+---------------------------------+----------------------------------+
| proxy_server.c |    pb_gatt.c   | pb_gatt_cli.c |    proxy_cli.c   |
+--------------------------------------------------------------------+
|          gatt_services.c        |           gatt_client.c          |
+--------------------------------------------------------------------+
|                            proxy_msg.c                             |
+--------------------------------------------------------------------+
```
Part of the client plan starts after the structure modularizing is completed, this is just a placeholder.

Also according to Trond's suggestion:

- Rename CONFIG_BT_MESH_PROXY to CONFIG_BT_MESH_GATT.
- Create an additional promptless entry CONFIG_BT_MESH_GATT_SERVER that selects CONFIG_BT_MESH_GATT and is selected by CONFIG_BT_MESH_GATT_PROXY or CONFIG_BT_MESH_PB_GATT.
- Replace all uses of CONFIG_BT_MESH_GATT_PROXY || CONFIG_BT_MESH_PB_GATT with this.

trond-snekvik, 2021-06-18:
Thumbs up from me, at least :slightly_smiling_face: 

CC @andrewpini

LingaoM, 2021-09-29:
@trond-snekvik 

Before starting PB-GATT Client development, there are some questions to be communicated:

1. At present, our provisioner function depends on local node provisioned, but the specification does not impose constraints. Ideally, provisioner can be enabled as long as CDB has been configured.

2. And our current `unprovisioned_beacon`,  The beacon callback cannot distinguish between PB-ADV and PB-GATT. Is it necessary to modify or add a new callback?

3. Zephyr Bluetooth host GATT client needs to turn off scanning when establishing a connection. How does this coexist with other functions of Bluetooth mesh functionality, such as relay, friend..

trond-snekvik, 2021-09-29:
@LingaoM 

1. I think this limitation might simplify future provisioning bearer implementations, and should be kept around, but if someone has a usage scenario where this needs to be possible, we could support it. I would imagine that it's a rare usage scenario, since you'll normally need to configure the models after provisioning anyway
2. Good point. If we can add a second callback that has a reasonable name, we wouldn't have to break the API. Not sure if a reasonable name exists though. We don't have any known scenarios that would require us to add a third callback like this, so I'm not very worried about scalability either.
3. True, this will be a bit annoying. I think we'll just have to accept this for now. Perhaps this can be changed in the controller in the future

LingaoM, 2021-09-29:
@jhedberg @joerchan  @Thalley @carlescufi 

I wonder if we can change the connection logic of the host first? 

According to the third description, if we do not require Host to make changes, the implementation of Bluetooth mesh is a bit complicated.

carlescufi, 2021-11-04:
@jhedberg @joerchan ping on @LingaoM 's comment above.

jhedberg, 2021-11-04:
I guess it depends on whether controllers (or rather the HCI spec) allows connections while scanning is active, or what exactly is the change request for the host? We can't just change something for the Zephyr controller and design the host around that, rather the host must follow what's allowed by the HCI spec. We do have an API do disable scanning for mesh: `bt_mesh_suspend()`, however I don't remember how that behaves when we're not provisioned.

closed 2022-10-07 by LingaoM