zephyrproject-rtos/zephyr #36344: Zephyr 2.6.0 st_ble_sensor sample is broken when compiled for nucleo_wb55rg

KevinHRea, 2021-06-17:
**Describe the bug**
The st_ble_sensor sample included in the Zephyr 2.5.0 tree compiles and runs successfully with the nucleo_wb55rg eval board, but does not work when I attempt the same from the Zephyr 2.6.0 tree. In this case the board advertises correctly can be seen with name P2PSRV1 with the LightBlue app running on a Mac, or iPhone, but times out when a connection is made.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Zephyr, Python dependencies and tool chain per Zephyr Getting Started Guide (on Ubuntu 20.04)
2. cd ~/zephyrproject/zephyr
3. git checkout v2.6.0
4. west update
3. west build -p auto -b nucleo_wb55rg samples/bluetooth/st_ble_sensor
4. west flash
5. Once device is discovered by LightBlue as P2PSRV1, click on it to connect and interrogate GATT
6. LightBlue times out and displays message "Connection Alert, Timeout interrogating the peripheral"

**Expected behavior**
LightBlue should connect successfully and display GATT characteristics. (This works with v2.5.0)

**Impact**
showstopper, we want to move our BLE project to Zephyr 2.6.0 to use the updated power management framework, and when we compile our project against Zephyr 2.6.0 we get the same behavior as we see with the st_ble_sensor sample about.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Toolchain: zephyr-sdk-0.12.4
 - Commit SHA or Version used: v2.6.0

**Additional context**
Add any other context about the problem here.

[image]
[image]

KevinHRea, 2021-06-17:
Console output:

*** Booting Zephyr OS build zephyr-v2.6.0  ***
[00:00:00.027,000] <inf> bt_hci_core: Identity: 80:E1:26:08:2D:61 (public)
[00:00:00.027,000] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0051, manufacturer 0x0030
[00:00:00.027,000] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x2151
[00:00:00.027,000] <inf> main: Bluetooth initialized
[00:00:00.030,000] <inf> main: Configuration mode: waiting connections...
[00:00:20.505,000]` <inf> main: Connected

Console output with debug logging turned on:

[00:00:00.000,000] <dbg>*** Booting Zephyr OS build zephyr-v2.6.0  ***
 bt_hci_core.bt_hci_driver_register: Registered BT IPM
--- 191 messages dropped ---
[00:00:00.039,000] <dbg> bt_hci_core.process_events: count 2
--- 247 messages dropped ---
[00:00:00.041,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 2 events
[00:00:00.041,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x0e
[00:00:00.041,000] <dbg> bt_buf.bt_buf_get_cmd_complete: sent_cmd 0x20003cf8
[00:00:00.041,000] <dbg> bt_hci_core.bt_recv: buf 0x20003cf8 len 6
[00:00:00.041,000] <dbg> bt_hci_core.hci_cmd_complete: opcode 0x2009
[00:00:00.041,000] <dbg> bt_hci_core.hci_cmd_done: opcode 0x2009 status 0x00 buf 0x20003cf8
[00:00:00.041,000] <dbg> bt_hci_core.bt_hci_cmd_send_sync: rsp 0x20003cf8 opcode 0x2009 len 1
[00:00:00.041,000] <dbg> bt_conn.bt_conn_set_state: disconnected -> connect-adv
[00:00:00.041,000] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
[00:00:00.041,000] <dbg> bt_hci_core.bt_hci_cmd_create: opcode 0x200a param_len 1
[00:00:00.041,000] <dbg> bt_hci_core.bt_hci_cmd_create: buf 0x20003cf8
[00:00:00.041,000] <dbg> bt_hci_core.bt_hci_cmd_send_sync: buf 0x20003cf8 opcode 0x200a len 4
[00:00:00.041,000] <dbg> bt_hci_core.process_events: count 2
[00:00:00.041,000] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:00.041,000] <dbg> bt_hci_core.send_cmd: calling net_buf_get
[00:00:00.042,000] <dbg> bt_hci_core.send_cmd: calling sem_take_wait
[00:00:00.042,000] <dbg> bt_hci_core.send_cmd: Sending command 0x200a (buf 0x20003cf8) to driver
[00:00:00.042,000] <dbg> bt_hci_core.bt_send: buf 0x20003cf8 len 4 type 0
[00:00:00.042,000] <dbg> hci_ipm.bt_ipm_send: CMD: buf 0x20003cf8 type 0 len 4
[00:00:00.042,000] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:00.042,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:00.042,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 2 events
[00:00:00.042,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x0e
[00:00:00.042,000] <dbg> bt_buf.bt_buf_get_cmd_complete: sent_cmd 0x20003cf8
[00:00:00.042,000] <dbg> bt_hci_core.bt_recv: buf 0x20003cf8 len 6
[00:00:00.042,000] <dbg> bt_hci_core.hci_cmd_complete: opcode 0x200a
[00:00:00.042,000] <dbg> bt_hci_core.hci_cmd_done: opcode 0x200a status 0x00 buf 0x20003cf8
[00:00:00.042,000] <dbg> bt_hci_core.bt_hci_cmd_send_sync: rsp 0x20003cf8 opcode 0x200a len 1
[00:00:00.042,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2 -> 1
[00:00:00.042,000] <inf> main: Configuration mode: waiting connections...
[00:00:00.048,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 2
[00:00:00.048,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.049,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.049,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.049,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 2
[00:00:00.049,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.050,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.050,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 5
[00:00:00.050,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 16
[00:00:00.050,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 16
[00:00:00.051,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0000 offset 0 length 16
[00:00:00.052,000] <dbg> bt_gatt: Hash: 
                                  df 98 5d a9 23 c7 96 3c  72 1b 04 a9 31 2d 81 d8 |..].#..< r...1-..
[00:00:29.968,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x3e
[00:00:29.968,000] <dbg> bt_hci_core.bt_recv: buf 0x20003c08 len 21
[00:00:29.968,000] <dbg> bt_hci_core.hci_rx_thread: buf 0x20003c08 type 1 len 21
[00:00:29.968,000] <dbg> bt_hci_core.hci_event: event 0x3e
[00:00:29.968,000] <dbg> bt_hci_core.hci_le_meta_event: subevent 0x01
[00:00:29.968,000] <dbg> bt_hci_core.le_legacy_conn_complete: status 0x00 role 1 14:C2:13:F2:78:05 (public)
[00:00:29.969,000] <dbg> bt_hci_core.bt_hci_le_enh_conn_complete: status 0x00 handle 2049 role 1 peer 14:C2:13:F2:78:05 (public) peer RPA 00:00:00:00:00:00
[00:00:29.969,000] <dbg> bt_hci_core.bt_hci_le_enh_conn_complete: local RPA 00:00:00:00:00:00
[00:00:29.969,000] <dbg> bt_conn.bt_conn_ref: handle 0 ref 1 -> 2
[00:00:29.969,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 2 -> 1
[... 27 lines cut ...]
[00:00:29.970,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:29.970,000] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000978 to poll list
[00:00:29.970,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events
[00:00:29.970,000] <dbg> bt_hci_core.hci_rx_thread: calling fifo_get_wait
[00:00:29.970,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x0f
[... cut ...]
[00:00:34.970,000] <dbg> bt_conn.send_buf: conn 0x20000978 buf 0x20003d28 len 16
[00:00:34.970,000] <dbg> bt_conn.send_frag: conn 0x20000978 buf 0x20003d28 len 16 flags 0x02
[00:00:34.970,000] <dbg> bt_hci_core.bt_send: buf 0x20003d28 len 20 type 2
[00:00:34.970,000] <dbg> hci_ipm.bt_ipm_send: ACL: buf 0x20003d28 type 2 len 20
[00:00:34.970,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:34.970,000] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000978 to poll list
[00:00:34.970,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events

KevinHRea, 2021-06-17:
When I turn on  debugging for the bluetooth subsystem, the 2.5.0 and 2.6.0 look the same until shortly after the informational log "Connected". 

Here is the entire log output from the "Connected" log message onwards for v2.6.0:
[00:00:08.206,000] <inf> main: Connected
[00:00:08.206,000] <dbg> bt_conn.bt_conn_ref: handle 2049 ref 2 -> 3
[00:00:08.206,000] <dbg> bt_gatt.bt_gatt_connected: conn 0x20000978
[00:00:08.206,000] <dbg> bt_hci_core.bt_hci_cmd_create: opcode 0x2016 param_len 2
[00:00:08.206,000] <dbg> bt_hci_core.bt_hci_cmd_create: buf 0x2000c4b8
[00:00:08.206,000] <dbg> bt_hci_core.bt_hci_cmd_send: opcode 0x2016 len 5
[00:00:08.206,000] <dbg> bt_conn.bt_conn_unref: handle 2049 ref 3 -> 2
[00:00:08.206,000] <dbg> bt_hci_core.process_events: count 2
[00:00:08.206,000] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:08.206,000] <dbg> bt_hci_core.send_cmd: calling net_buf_get
[00:00:08.206,000] <dbg> bt_hci_core.send_cmd: calling sem_take_wait
[00:00:08.206,000] <dbg> bt_hci_core.send_cmd: Sending command 0x2016 (buf 0x2000c4b8) to driver
[00:00:08.206,000] <dbg> bt_hci_core.bt_send: buf 0x2000c4b8 len 5 type 0
[00:00:08.206,000] <dbg> hci_ipm.bt_ipm_send: CMD: buf 0x2000c4b8 type 0 len 5
[00:00:08.206,000] <dbg> bt_hci_core.process_events: ev->state 1
[00:00:08.206,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:08.206,000] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000978 to poll list
[00:00:08.206,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events
[00:00:08.206,000] <dbg> bt_hci_core.hci_rx_thread: calling fifo_get_wait
[00:00:08.207,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x0f
[00:00:08.207,000] <dbg> bt_buf.bt_buf_get_cmd_complete: sent_cmd 0x2000c4b8
[00:00:08.207,000] <dbg> bt_hci_core.bt_recv: buf 0x2000c4b8 len 6
[00:00:08.207,000] <dbg> bt_hci_core.hci_cmd_status: opcode 0x2016
[00:00:08.207,000] <dbg> bt_hci_core.hci_cmd_done: opcode 0x2016 status 0x3a buf 0x2000c4b8
[00:00:13.206,000] <dbg> bt_conn.deferred_work: conn 0x20000978
[00:00:13.206,000] <dbg> bt_conn.send_conn_le_param_update: conn 0x20000978 features 0x00 params (24-40 0 42)
[00:00:13.206,000] <dbg> bt_l2cap.bt_l2cap_send_cb: conn 0x20000978 cid 5 len 12
[00:00:13.206,000] <dbg> bt_conn.bt_conn_send_cb: conn handle 2049 buf len 16 cb (nil) user_data (nil)
[00:00:13.206,000] <dbg> bt_hci_core.process_events: count 3
[00:00:13.206,000] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:13.206,000] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:13.206,000] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:13.206,000] <dbg> bt_conn.bt_conn_process_tx: conn 0x20000978
[00:00:13.206,000] <dbg> bt_conn.send_buf: conn 0x20000978 buf 0x2000c4e8 len 16
[00:00:13.206,000] <dbg> bt_conn.send_frag: conn 0x20000978 buf 0x2000c4e8 len 16 flags 0x02
[00:00:13.206,000] <dbg> bt_hci_core.bt_send: buf 0x2000c4e8 len 20 type 2
[00:00:13.206,000] <dbg> hci_ipm.bt_ipm_send: ACL: buf 0x2000c4e8 type 2 len 20
[00:00:13.206,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:13.206,000] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000978 to poll list
[00:00:13.206,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events

And here is a partial log output from 2.5.0 from the "Connected" log message onwards:

t_hci_cmd_send: opcode 0x2016 len 5
[00:00:12.648,000] <dbg> bt_conn.bt_conn_unref: handle 2049 ref 3 -> 2
[00:00:12.648,000] <dbg> bt_hci_core.process_events: count 2
[00:00:12.648,000] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:12.648,000] <dbg> bt_hci_core.send_cmd: calling net_buf_get
[00:00:12.648,000] <dbg> bt_hci_core.send_cmd: calling sem_take_wait
[00:00:12.648,000] <dbg> bt_hci_core.send_cmd: Sending command 0x2016 (buf 0x2000bb40) to driver
[00:00:12.648,000] <dbg> bt_hci_core.bt_send: buf 0x2000bb40 len 5 type 0
[00:00:12.648,000] <dbg> hci_ipm.bt_ipm_send: CMD: buf 0x2000bb40 type 0 len 5
[00:00:12.648,000] <dbg> bt_hci_core.process_events: ev->state 1
[00:00:12.648,000] <dbg> bt_conn.bt_conn_prepare_events: 
[00:00:12.648,000] <dbg> bt_conn.conn_prepare_events: Adding conn 0x20000918 to poll list
[00:00:12.648,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events
[00:00:12.648,000] <dbg> bt_hci_core.hci_rx_thread: calling fifo_get_wait
[00:00:12.648,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x0f
[00:00:12.648,000] <dbg> bt_buf.bt_buf_get_cmd_complete: sent_cmd 0x2000bb40
[00:00:12.648,000] <dbg> bt_hci_core.bt_recv: buf 0x2000bb40 len 6
[00:00:12.648,000] <dbg> bt_hci_core.hci_cmd_status: opcode 0x2016
[00:00:12.648,000] <dbg> bt_hci_core.hci_cmd_done: opcode 0x2016 status 0x3a buf 0x2000bb40
[00:00:12.685,000] <dbg> hci_ipm.bt_ipm_rx_thread: ACL: handle 2801, len 7
[00:00:12.685,000] <dbg> bt_hci_core.bt_recv: buf 0x2000ba38 len 11
[00:00:12.685,000] <dbg> bt_hci_core.hci_rx_thread: buf 0x2000ba38 type 3 len 11
[00:00:12.685,000] <dbg> bt_hci_core.hci_acl: buf 0x2000ba38
[00:00:12.685,000] <dbg> bt_hci_core.hci_acl: handle 2049 len 7 flags 2
[... 27 lines cut ...]
[00:00:12.686,000] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 3 events
[00:00:12.686,000] <dbg> bt_hci_core.hci_rx_thread: calling fifo_get_wait
[00:00:12.708,000] <dbg> hci_ipm.bt_ipm_rx_thread: EVT: hcievt->evtserial.evt.evtcode: 0x13
[00:00:12.708,000] <dbg> bt_hci_core.bt_recv: buf 0x2000bb28 len 7
[00:00:12.708,000] <dbg> bt_hci_core.hci_num_completed_packets: num_handles 1
[... cut ...]
[00:00:12.708,000] <dbg> bt_att.att_sent: conn 0x20000918 chan 0x20002f28
[00:00:12.708,000] <dbg> bt_att.bt_att_sent: chan 0x20002f20
[00:00:12.708,000] <dbg> bt_att.chan_rsp_sent: chan 0x20002f20
[00:00:12.708,000] <dbg> bt_conn.bt_conn_recv: handle 2049 len 11 flags 02
[00:00:12.708,000] <dbg> bt_conn.bt_conn_recv: First, len 11 final 7
[00:00:12.708,000] <dbg> bt_conn.bt_conn_recv: Successfully parsed 11 byte L2CAP packet
[00:00:12.708,000] <dbg> bt_l2cap.bt_l2cap_recv: Packet for CID 4 len 7
... many more log messages after this

FRASTM, 2021-06-17:
I could identify that commit 2691541ad22aeb5530c3a692e708e9958865f372 is involved in the failure.

FRASTM, 2021-06-17:
The pb comes from the drivers/bluetooth/hci/ipm_stm32wb.c
where the #if STM32_LSE_CLOCK is **NOT** defined here

FRASTM, 2021-06-17:
with a Kconfig in the drivers/bluetooth/hci/ like

```
config BT_STM32_LSE_CLOCK
	bool "lse clock"
	default y
	depends on CLOCK_STM32_LSE
	depends on BT_STM32_IPM

```
The LL_RCC_LSE_Enable() is selected, but it seems not to be enough

closed 2021-06-18 by galak