zephyrproject-rtos/zephyr #36358: Potential issue with CMAKE_OBJECT_PATH_MAX

alexsven, 2021-06-17:
Since nrf5340_cpunet_reset.c is located in the boards folder (zephyr/boards/arm/nrf5340dk_nrf5340), the path for linking it in might quickly be too long, depending on where the project is stored on the local machine.

One easy fix is to make sure to place the project as close to root as possible, but this might not be an option for all users.

tejlmand, 2021-06-17:
Just a note: `CMAKE_OBJECT_PATH_MAX` has to do with the full path, it doesn't help regarding library names that are too long.

When using `zephyr_library()` without a name, then the lib name is constructed based on it's location:
https://github.com/zephyrproject-rtos/zephyr/blob/4b54d363e8802f301cfd7f1806ca78b288181da1/cmake/extensions.cmake#L391-L409

So best solution in this case would be to slowly deprecate `zephyr_library()` and instead use `zephyr_library_named()` (or even start a transistion towards plain CMake, see https://github.com/zephyrproject-rtos/zephyr/issues/8827 )

koffes, 2021-08-23:
This issue is still valid.

closed 2022-06-25 by github-actions[bot]

nordicjm, 2022-09-02:
I asked on the cmake discussion forum and this should not be an issue, see https://discourse.cmake.org/t/object-file-names-use-absolute-path-to-sources-outside-source-tree/6383/6 and corresponding code at https://gitlab.kitware.com/cmake/cmake/-/blob/v3.24.1/Source/cmLocalGenerator.cxx#L3516-3560 which was last updated 7 months ago so should be in cmake version 3.23 onwards

mithunpandey, 2023-12-20:
We are also facing the same issue 
we are using twister to run test on out our HW
python zephyr/scripts/twister --device-testing --device-serial COM5 -p nucleo_h743zi -T app-to-build/examples
we are using GitHub, and a self hosted windows runner  
And the cmake is getting failed with 

`CMake Warning in C:/actions-runner/_work/edge-rtos-github-workflows/edge-rtos-github-workflows/zephyr/arch/arm/core/aarch32/cortex_m/CMakeLists.txt:
  The object file directory

    C:/actions-runner/_work/edge-rtos-github-workflows/edge-rtos-github-workflows/twister-out/nucleo_h743zi/drivers/SPI/Async_SPI/examples.drivers.SPI.Async_SPI/zephyr/arch/arch/arm/core/aarch32/cortex_m/CMakeFiles/arch__arm__core__aarch32__cortex_m.dir/./

  has 252 characters.  The maximum full path to an object file is 250
  characters (see CMAKE_OBJECT_PATH_MAX).  Object file

    fault.c.obj

  cannot be safely placed under this directory.  The build may not work
  correctly.`

JPHutchins, 2024-07-05:
This issue is quite prevalent when running twister from Windows.

PerMac, 2024-10-03:
We resolved the issue in twister on windows by using `--no-detailed-test-id` and `--short-build-path`