zephyrproject-rtos/zephyr #36374: Rename net_buf to something more generic and move it outside of net/

jhedberg, 2021-06-17:
Networking buffers is a separate subsystem with no dependency on networking itself. This implementation was originally created for Bluetooth and called bt_buf back then. Once work started to create a networking stack for Zephyr it made sense to reuse the buffer implementation, but Networking depending on something called bt_buf didn't feel right. Consequently, the implementation was renamed to net_buf and moved under net/ due to its namespace.

There are a couple of issues with the current situation:

 - There is no direct dependency on the networking stack, but the naming gives an impression that there might be
 - The API is used by other subsystems (a major one being Bluetooth) even though these also don't have any dependency on the networking stack

It would therefore make sense to consider renaming the subsystem and API to something more generic, and the natural choice for that would be sys_buf (based on current Zephyr naming guidelines).

tbursztyka, 2021-07-02:
sys_buf or zbuf, etc... seems like a legitimate change indeed.

nashif, 2024-05-15:
Can would revive this please? @jhedberg  @carlescufi 
What needs to be done here?

rodrigopex, 2024-05-16:
I support that. I used `net_buf` on the zbus implementation. It saved a lot of work for me.

bjarki-trackunit, 2024-05-21:
+1 to sys_buf :)

nashif, 2024-05-21:
from wg meeting:

- The net buffer, initially part of Bluetooth, was moved under networking and is now being used in other contexts.
- Proposal to rename and move it outside networking to avoid confusion.
- Agreement to promote it to a general library but keep the existing name temporarily for compatibility and adapt documentation and references, ie. remove networking from the name.
- Discussion on how to migrate users and avoid breaking downstream dependencies.
- Plan to create a wrapper to facilitate a smooth transition.

Actions:

- Initiate the first step of promoting the net buffer and updating documentation.
- Team to revisit the migration strategy post-LTS and discuss the implementation of wrappers.

hubertmis, 2024-05-22:
Good initiative!

There is also an opportunity to use net_buf in IPC for a zero-copy data plane.

nashif, 2024-09-04:
@jhedberg  do we want to move this forward?

henrikbrixandersen, 2024-09-04:
> @jhedberg do we want to move this forward?

I am actually working on a PR for this just now.

PR #78009 (2024-09-07): lib: net_buf: move the network buffer implementation to lib
Move the net_buf implementation from the networking subsystem to a library as they have no dependency on the networking subsystem.

Network buffers are used in subsystems outside of networking (e.g. Bluetooth, USB).

Fixes: #36374
commit ced80b13 (2024-09-07): lib: net_buf: move the network buffer implementation to lib

Move the net_buf implementation from the networking subsystem to a library
as they have no dependency on the networking subsystem.

Network buffers are used in subsystems outside of networking
(e.g. Bluetooth, USB).

Fixes: #36374
closed 2024-09-07 by mmahadevan108