zephyrproject-rtos/zephyr #36381: Bluetooth ASSERTION FAIL [evdone] Zephyr v2.6.0

nwsetec, 2021-06-18:
**Describe the bug**
While developing with Zephyr v2.0.6 c9fd7a6099c4fad73014aaf0ce364996afbba9e6 our app raised this Bluetooth assert:
`ASSERTION FAIL [evdone] @ .../zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:353`

I've only experienced this once but 

**To Reproduce**
Setup a project the passive Bluetooth scanning with a 60ms scanning interval and 60ms window with and the filter duplicates disable and set up the project to allow up to 4 connected BLE devices and up to 4 bonds.

[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/6675076/prj.conf.txt)

```
#define BT_LE_SCAN_PASSIVE_ALLOW_DUPLICATES BT_LE_SCAN_PARAM(BT_HCI_LE_SCAN_PASSIVE, \
					    BT_HCI_LE_SCAN_FILTER_DUP_DISABLE, \
					    BT_GAP_SCAN_FAST_INTERVAL, \
					    BT_GAP_SCAN_FAST_INTERVAL)

int scan_init(void)
{
	int rc;

	rc = bt_le_scan_start(BT_LE_SCAN_PASSIVE_ALLOW_DUPLICATES, scan_device_found);
	if (rc) {
		LOG_ERR("Scanning failed to start (err %d)\n", rc);
	}
	else {
		LOG_INF("Scanning Started\r\n");
	}

	return rc;
}
```

**Expected behavior**
No asserts happen.

**Impact**
Annoyance

**Logs and console output**
` @ .../zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:353`

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain C compiler identification is GNU 7.3.1
 - Commit SHA c9fd7a6099c4fad73014aaf0ce364996afbba9e6 

**Additional context**
Small discussion with @cvinayak on Slack

nwsetec, 2021-06-22:
We have also observed: `ASSERTION FAIL [rx_dle] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:5466`

cvinayak, 2021-06-23:
> 
> 
> We have also observed: `ASSERTION FAIL [rx_dle] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:5466`

@nwsetec Please increase the `LL_PDU_RX_CNT` count here: https://github.com/zephyrproject-rtos/zephyr/blob/c9fd7a6099c4fad73014aaf0ce364996afbba9e6/subsys/bluetooth/controller/ll_sw/ull.c#L317

I will send a PR once you can confirm, it avoids the assertion in ull_conn.c.

nwsetec, 2021-06-24:
Hi @cvinayak I have done some testing with your recommendation above with 4 mobile devices.  I have not observed any assertions in ull_conn.c and only one assertion in lll.c:353 but a new issue seems to have emerged.  After connecting and bonding four mobile devices and waiting 5 or so minutes it appears as though the connection 'slots' start to disappear.  In that one of the devices is disconnected but the device doesn't resume advertising so the fourth device can reconnect.  This keeps happening until all the connection slot are gone and no mobile devices can connect.  During this time the Zephyr device is still happily running without resets.  Previously this may have been happening but when the assert(s) occurred the device had to be reset so this removal of connections slots effect wouldn't have been observed.

cvinayak, 2021-06-24:
Hi @nwsetec, I have re-designed part of implementation to solve the assertion in ull_conn.c, has lower memory cost. Will push a PR and ping you back.

> one of the devices is disconnected but the device doesn't resume advertising so the fourth device can reconnect. 

We should follow up on this in a separate GH issue or come back to it here in this GH issue once a fix for ull_conn.c assert is approved.

Thinking aloud though, the application is not using `BT_LE_ADV_OPT_ONE_TIME` (assumption), which means the host is responsible to restart advertising on every new connection until maximum supported number of connections is established. And then when one of the connections in the maximum connected disconnects (after "waiting  5 or so minutes"), host shall start a fresh advertising . I will need a hci_core debug log to analyse this flow to check if controller returned any error code to HCI commands issues by the host.

nwsetec, 2021-06-24:
@cvinayak yes we're not using BT_LE_ADV_OPT_ONE_TIME.  I'll attempt to get a hci_core log right now.

nwsetec, 2021-06-25:
Hi Vich, we've done some testing with #36515 and have observed `ASSERTION FAIL [evdone] @WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:353` but not yet the `ull_conn.c` assert.

cvinayak, 2021-06-25:
Hi Nick, ok I will focus on the lll.c assert. Can you share the BT HCI core debug (`CONFIG_BT_DEBUG_HCI_CORE=y`) logs until the assertion? (i.e. if you can reproduce it again, you can email me, if that is convenient).

Also, describe the steps to reproduce (or the order of operations performed by the implementation and the peers).

(You may have discovered a rare race condition, I would need as much data to get to the root cause).

cvinayak, 2021-06-25:
From slack:
> Nick Ward: From my colleague's report the lll.c:353 assert happens when there is room for some more connections to be established
He had one connected and was in the process of connecting another and roughly around that time the assert happened
Also, as per GitHub request I'll create a log shortly...
vich: Ok, as the assert is around connection establishment, then I am confident the issue is related to #30245.
Nick Ward: Would you still like the log?
vich: No, not necessary. As a short term workaround you can increase the EVENT_DONE_MAX to 4

cvinayak, 2021-06-29:
@nwsetec Please confirm that with the work around of `EVENT_DONE_MAX` set to 4, you are able to mitigate the lll.c assert, right?

Both this current issue #36381 and #30245 will be kept open until the fix is upstreaned.

nwsetec, 2021-06-29:
> @nwsetec Please confirm that with the work around of `EVENT_DONE_MAX` set to 4, you are able to mitigate the lll.c assert, right?

@vich yes, confirmed, I made the modification here: https://github.com/nwsetec/zephyr/commit/a806dc1d0b5f8c799f464578387145bde7221efb
with the other commits from https://github.com/zephyrproject-rtos/zephyr/pull/36515 cherry picked in as well and I no longer see any asserts.

cvinayak, 2021-08-05:
@nwsetec I have fixed https://github.com/zephyrproject-rtos/zephyr/issues/30245 which I believe should fix this issue too.

Please remove your workaround https://github.com/nwsetec/zephyr/commit/a806dc1d0b5f8c799f464578387145bde7221efb  and check if this PR resolves the assertion you face, https://github.com/zephyrproject-rtos/zephyr/pull/37440

nwsetec, 2021-09-11:
@cvinayak while I was testing https://github.com/zephyrproject-rtos/zephyr/issues/36679 and discovering it was fixed I noticed this assert occurring.  I'm currently checked out at main SHA 30db452aec00b5e7f6f1933ef99a9a3387730ee2.

cvinayak, 2021-09-14:
Reopen this issue to investigate the issue.

cvinayak, 2021-09-26:
@nwsetec Hi, I have been analyzing the issue under simulation with clock drifts, 4 central connection to a 1 peripheral which also is scanning, performing pairing procedures. The assert does not occur under simulation.

Attached is the simulated packet trace: 
[multiple.zip](https://github.com/zephyrproject-rtos/zephyr/files/7231153/multiple.zip)

I will spend sometime next week to flash the application on target boards and see if the issue can be reproduced there.

carlescufi, 2021-09-30:
@nwsetec any help appreciated here because at this time we are completely unable to reproduce this with the v2.7-branch. Since this is a high priority bug for an LTS release, we would very grateful if you could help us move forward with it.

nwsetec, 2021-10-01:
I've started testing and this issue is still present:
`uart:~$ ASSERTION FAIL [evdone] @ C:/Users/nick.ward/PROSmart/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:365`

cvinayak, 2021-10-04:
[image]

I was able to reproduce the issue in modified bt shell app with scan interval and window of 2.5ms, and connecting two phones to it.

Following is the analysis in the order of occurrence, pipeline count corresponds to the number in the logic trace above:
- D1, Preemption of continuous scan window, pipeline count retained at 1 enqueued scan resume (after duplicates flushed from the pipeline).
- B1, New scan prepare enqueued into pipeline, increasing the enqueued count to 2.
- A1, New peripheral prepare enqueued into pipeline, increasing the enqueued count to 3.
- D2, Preemption of continuous scan window, pipeline count returns to 2 retain one scan resume event after removing duplicate scan prepare event.
- B2, New scan prepare enqueued into pipeline, increasing the enqueued count to 3.
- C1-C2, minimum ticker offset, causes the peripheral event to be late after preemption of the continuous scan.
- C2, done events generated for 2 scan prepare, 1 peripheral abort and 1 scan resume abort, against the allocated maximum of 3 done events causes the assertion.

Solution, increase `EVENT_DONE_MAX` to 4.

PR #37700 (2021-08-16): Bluetooth: Controller: Only remove duplicate resume events
When events are placed back into pipeline as resume events,
only remove duplicate resume events and not new prepare
events that may have been enqueue between the start of the
pre-empt ticker and its timeout.

Due to new prepare events that was removed, extra done
events generated cause the number of enqueued done events
to overflow and assert.

Fixes #36381.
Fixes #37597.

Controller debug pin toggle detecting a late advertising event caused by a race condition where new scan prepare is enqueued between adv prepare and its pre-empt tim [... cut]
commit 190532bc (2021-08-16): Bluetooth: Controller: Only remove duplicate resume events

When events are placed back into pipeline as resume events,
only remove duplicate resume events and not new prepare
events that may have been enqueue between the start of the
pre-empt ticker and its timeout.

Due to new prepare events that was removed, extra done
events generated cause the number of enqueued done events
to overflow and assert.

Fixes #36381.
Fixes #37597.
PR #39079 (2021-10-04): Bluetooth: Controller: Fix assertion failed [evdone]
When there are radio events with time reservations lower
than the preemption timeout of 1.5 ms, the pipeline has to
account for the maximum radio events that can be enqueued
during the preempt timeout duration. All these enqueued
events could be aborted in case of late scheduling needing
as many done event buffers.

During continuous scanning, there can be 1 active radio
event, 1 scan resume and 1 new scan prepare. If there are
peripheral prepares in addition, and due to late scheduling
all these will abort needing 4 done buffers.

Fixes #36 [... cut]
commit 072c8f2c (2021-10-04): Bluetooth: Controller: Fix assertion failed [evdone]

When there are radio events with time reservations lower
than the preemption timeout of 1.5 ms, the pipeline has to
account for the maximum radio events that can be enqueued
during the preempt timeout duration. All these enqueued
events could be aborted in case of late scheduling needing
as many done event buffers.

During continuous scanning, there can be 1 active radio
event, 1 scan resume and 1 new scan prepare. If there are
peripheral prepares in addition, and due to late scheduling
all these will abort needing 4 done buffers.

If Exten [... cut]
closed 2021-10-04 by carlescufi