zephyrproject-rtos/zephyr #36412: Blinky on ESP32: Unsupported board: led0 devicetree alias is not defined"

torntrousers, 2021-06-20:
Trying to get going with Zephy on an ESP32 and the Blinky sample fails to compile with 
```
C:/Users/Administrator.CQ-LAP-000001/zephyrproject/zephyr/samples/basic/blinky/src/main.c:24:2: error: #error "Unsupported board: led0 devicetree alias is not defined"
 #error "Unsupported board: led0 devicetree alias is not defined"
  ^~~~~
```
Googling, I think I need to add something about led0 to ```C:\Users\Administrator.CQ-LAP-000001\zephyrproject\zephyr\boards\xtensa\esp32\esp32.dts```, but its hard to work out what the syntax should be.

Can someone give me the text?

sylvioalves, 2021-06-21:
@torntrousers, it fails because, as you said, there is no default led0 definition. What pin are you using?

torntrousers, 2021-06-21:
I have a variety of boards, but lets say its a [Heltec WiFi Kit 32](https://heltec.org/project/wifi-kit-32/) which has an LED on GPIO 25 i think.

sylvioalves, 2021-06-21:
There are a few ways to make that work. Changing the .dts is a way. You can also add an overlay file within the project:
1. create a file named esp32.overlay at `samples/basic/blinky`.
2. pasted the content below (it will blink pin 2)

```c
/ {
	aliases {
		led0 = &led0;
	};

	leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&gpio0 2 0>;
			label = "LED 0";
		};
	};
};
```

torntrousers, 2021-06-21:
Woohoo, that works. Thanks @sylvioalves .

If I may, this hasn't been a particularly easy getting started process for someone new to Zephyr and ESP32 on Zephyr. I;d be happy to give some tips on how to make things a bit easier for a newbie.

sylvioalves, 2021-06-21:
@torntrousers great, let's chat in slack channel. Can you do that?

closed 2021-06-21 by sylvioalves

nefen, 2023-10-23:
I tried the same example.
It works.
BUT, if I set CONFIG_ASSERT=y, then I get:
*** Booting Zephyr OS build zephyr-v3.5.0-232-g7a27431a0c63 ***
ASSERTION FAIL [(cfg->port_pin_mask & (gpio_port_pins_t)(1UL << (pin))) != 0U] @ WEST_TOPDIR/zephyr/include/zephyr/drivers/gpio.h:1001
        Unsupported pin

Backtrace:0x40080b20:0x3ffe65b0 0x400d0985:0x3ffe65c0 0x400d049d:0x3ffe65e0 0x400812d3:0x3ffe6600 0x400d04df:0x3ffe6620

Can anyone help me?

sylvioalves, 2023-10-23:
@nefen, it says in there `Unsupported pin`. Which SoC and PIN you have been using?

nefen, 2023-10-23:
esp32-devkitc-v4 with esp32-wroom-32D.

I used various pins, but with the same result.

In the last test I used pin 25

sylvioalves, 2023-10-27:
@nefen, which Zephyr version you are using? Have you also run `west update`?

nefen, 2023-10-27:
Yes, I did

 

Φιλικά

Νίκος Εφεντάκης

Ομορφοκκλησιάς 29

15122 ΜΑΡΟΥΣΙ 

Τηλ: +30 210 2207 582

nefen, 2023-11-02:
> @nefen, which Zephyr version you are using? Have you also run `west update`?

**Can anyone please help!!!!!**

@sylvioalves I use esp32_devkitc_v4 with chip esp32_wroom_32d.

I had run complex code on this module. when built with freertos. I want to switch to Zephyr to have the same rtos for development on esp32 and nordic SOCs.

**I cannot solve the issue. It builds well (selected board is "esp32_devkitc_wroom"):**
[0/1] Re-running CMake...Loading Zephyr default modules (Zephyr base (cached)).
-- Application: C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs
-- CMake version: 3.27.7
-- Cache files will be written to: C:/Users/nefen/zephyrproject/zephyr/.cache
-- Zephyr version: 3.5.99 (C:/Users/nefen/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32_devkitc_wroom
-- Found host-tools: zephyr 0.16.3 (C:/Users/nefen/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (C:/Users/nefen/zephyr-sdk-0.16.3)
-- Found BOARD.dts: C:/Users/nefen/zephyrproject/zephyr/boards/xtensa/esp32_devkitc_wroom/esp32_devkitc_wroom.dts
-- Found devicetree overlay: esp32.overlay
-- Generated zephyr.dts: C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/dts.cmake
Parsing C:/Users/nefen/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/.config'
No change to configuration in 'C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/.config'
No change to Kconfig header in 'C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/build/zephyr/include/generated/autoconf.h'
CMake Warning at C:/Users/nefen/zephyrproject/zephyr/CMakeLists.txt:1957 (message):
  __ASSERT() statements are globally ENABLED

**Overlay file is this:**
/ {
    
	outputs {
		compatible = "gpio-leds";
		output1: output_1 {
			gpios = <&gpio0 13 GPIO_ACTIVE_HIGH>;
			label = "Enable Peripheral";
		};
    };

    aliases {
		actsensor0 = &output1;
    };
};

**The code is simple:** 

```
#define PERON0_NODE DT_ALIAS(actsensor0)			//12
struct gpio_dt_spec act_sensor0 = GPIO_DT_SPEC_GET(PERON0_NODE, gpios);
int main(void)
{
	int ret;
	if (!device_is_ready(act_sensor0.port)) {
		printk("Device is not readyn");		
		return 0;
	}

	ret = gpio_pin_configure_dt(&act_sensor0, GPIO_OUTPUT);
	if (ret < 0) {
		printk("Device cannot be configured");		

		return 0;
	}
}
```

**It fails when executes gpio_pin_configure_dt:** 
ASSERTION FAIL [(cfg->port_pin_mask & (gpio_port_pins_t)(1UL << (pin))) != 0U] @ WEST_TOPDIR/zephyr/include/zephyr/drivers/gpio.h:1001
        Unsupported pin

Backtrace:0x40080b20:0x3ffe65b0 0x400d09dd:0x3ffe65c0 0x400d04a9:0x3ffe65e0 0x400812d3:0x3ffe6600 0x400d0537:0x3ffe6620

Asfak1999, 2025-01-20:
D:/Zephyr_Projects/external/zephyr/include/zephyr/device.h:96:41: error: '__device_dts_ord_DT_N_ALIAS_led0_P_gpios_IDX_0_PH_ORD' undeclared here (not in a function)
   96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)  
      |                                         ^~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/toolchain/common.h:137:26: note: in definition of macro '_DO_CONCAT'
  137 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
D:/Zephyr_Projects/external/zephyr/include/zephyr/device.h:96:33: note: in expansion of macro '_CONCAT'
   96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)  
      |                                 ^~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/device.h:238:37: note: in expansion of macro 'DEVICE_NAME_GET'
  238 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/device.h:255:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  255 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/drivers/gpio.h:339:25: note: in expansion of macro 'DEVICE_DT_GET'
  339 |                 .port = DEVICE_DT_GET(DT_GPIO_CTLR_BY_IDX(node_id, prop, idx)),\
      |                         ^~~~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/drivers/gpio.h:375:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  375 |         GPIO_DT_SPEC_GET_BY_IDX(node_id, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
D:/Zephyr_Projects/blinky/src/main.c:21:40: note: in expansion of macro 'GPIO_DT_SPEC_GET'
   21 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                        ^~~~~~~~~~~~~~~~     
In file included from D:/Zephyr_Projects/external/zephyr/include/zephyr/arch/xtensa/arch.h:18,
                 from D:/Zephyr_Projects/external/zephyr/include/zephyr/arch/cpu.h:27,
                 from D:/Zephyr_Projects/external/zephyr/include/zephyr/kernel_includes.h:36:
D:/Zephyr_Projects/external/zephyr/include/zephyr/devicetree.h:236:32: error: 'DT_N_ALIAS_led0_P_gpios_IDX_0_VAL_pin' undeclared here (not in a function)
  236 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                                ^~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/devicetree.h:5094:9: note: in definition of macro 'DT_CAT7'
 5094 |         a1 ## a2 ## a3 ## a4 ## a5 ## a6 ## a7
      |         ^~
D:/Zephyr_Projects/external/zephyr/include/zephyr/devicetree/gpio.h:110:9: note: in expansion of macro 'DT_PHA_BY_IDX'
  110 |         DT_PHA_BY_IDX(node_id, gpio_pha, idx, pin)
      |         ^~~~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/drivers/gpio.h:340:24: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  340 |                 .pin = DT_GPIO_PIN_BY_IDX(node_id, prop, idx),                 \
      |                        ^~~~~~~~~~~~~~~~~~
D:/Zephyr_Projects/external/zephyr/include/zephyr/drivers/gpio.h:375:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  375 |         GPIO_DT_SPEC_GET_BY_IDX(node_id, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
D:/Zephyr_Projects/blinky/src/main.c:21:40: note: in expansion of macro 'GPIO_DT_SPEC_GET'
   21 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                        ^~~~~~~~~~~~~~~~     
D:/Zephyr_Projects/external/zephyr/include/zephyr/devicetree.h:236:25: note: in expansion of macro 'DT_CAT'
  236 | #define DT_ALIAS(alias) DT_CAT(DT_N_ALIAS_, alias)
      |                         ^~~~~~
D:/Zephyr_Projects/blinky/src/main.c:15:19: note: in expansion of macro 'DT_ALIAS'
   15 | #define LED0_NODE DT_ALIAS(led0)
      |                   ^~~~~~~~
D:/Zephyr_Projects/blinky/src/main.c:21:57: note: in expansion of macro 'LED0_NODE'
   21 | static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
      |                                                         ^~~~~~~~~
[95/189] Building C object zephy...hyr.dir/lib/utils/bitarray.c.obj 
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Users\User\.zephyrtools\cmake\cmake-3.22.0-windows-x86_64\bin\cmake.EXE' --build 'd:\Zephyr_Projects\blinky\build\esp32_devkitc_wrover\esp32\appcpu'

I got this error, I am using ESp32