zephyrproject-rtos/zephyr #36414: ESP32 with samples\net\wifi gives: net_if: There is no network interface to work with!

torntrousers, 2021-06-20:
Trying to run the Zephyr WiFi sample in the samples\net\wifi directory and it fails with:

```
[0;32mI (29) boot: compile time 12:56:29[0m
[0;32mI (29) boot: chip revision: 1[0m
[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (45) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00002000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 00008000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (91) boot: End of partition table[0m
[0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x04574 ( 17780) map[0m
[0;32mI (118) esp_image: segment 1: paddr=0x0001459c vaddr=0x3ffb0000 size=0x00124 (   292) load[0m
[0;32mI (120) esp_image: segment 2: paddr=0x000146c8 vaddr=0x3ffb0124 size=0x00050 (    80) load[0m
[0;32mI (129) esp_image: segment 3: paddr=0x00014720 vaddr=0x3ffb0174 size=0x00064 (   100) load[0m
[0;32mI (138) esp_image: segment 4: paddr=0x0001478c vaddr=0x3ffb01d8 size=0x008f4 (  2292) load[0m
[0;32mI (148) esp_image: segment 5: paddr=0x00015088 vaddr=0x3ffb0acc size=0x00088 (   136) load[0m
[0;32mI (156) esp_image: segment 6: paddr=0x00015118 vaddr=0x3ffb0b54 size=0x000c0 (   192) load[0m
[0;32mI (165) esp_image: segment 7: paddr=0x000151e0 vaddr=0x3ffb0c14 size=0x00100 (   256) load[0m
[0;32mI (174) esp_image: segment 8: paddr=0x000152e8 vaddr=0x3ffb0d14 size=0x00030 (    48) load[0m
[0;32mI (183) esp_image: segment 9: paddr=0x00015320 vaddr=0x3ffb0d44 size=0x00044 (    68) load[0m
[0;32mI (192) esp_image: segment 10: paddr=0x0001536c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (201) esp_image: segment 11: paddr=0x00015774 vaddr=0x40080400 size=0x0a8a4 ( 43172) load[0m
[0;32mI (229) esp_image: segment 12: paddr=0x00020020 vaddr=0x400d0020 size=0x07094 ( 28820) map[0m
[0;32mI (240) esp_image: segment 13: paddr=0x000270bc vaddr=0x4008aca4 size=0x02a7c ( 10876) load[0m
[0;32mI (252) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (252) boot: Disabling RNG early entropy source...[0m
⸮*** Booting Zephyr OS build zephyr-v2.6.0-420-ge62c3c533a37  ***

[1;32muart:~$ [m[8D[J[00:00:00.529,000] [1;31m<err> net_if: There is no network interface to work with![0m
[1;32muart:~$ [m[8D[J[1;32muart:~$ [m
```

Any tips on how to get it working?

sylvioalves, 2021-06-21:
@torntrousers Please, use `samples/boards/esp32/wifi_station` as reference.

torntrousers, 2021-06-21:
Thanks @sylvioalves . I've got that to run but it does not seem to connect. I've tried putting in the incorrect password and th elog is the same. Is there some thing I can do to debug what is happening?
```
*⸮P⸮V⸮⸮ET⸮Q⸮UU⸮T⸮⸮-⸮QR⸮⸮⸮T⸮⸮⸮RUJ⸮⸮⸮⸮Z⸮j⸮⸮T⸮J⸮R⸮J⸮⸮⸮Ԯ⸮⸮⸮U⸮⸮*ZQBU⸮⸮j⸮⸮EJ⸮R⸮Z⸮Q⸮WխQ(⸮uT⸮Z⸮QJ-]⸮Z⸮,Q⸮⸮R⸮Z⸮QڬR⸮Z⸮Q⸮⸮⸮⸮I⸮⸮⸮T⸮U⸮UZU%u⸮Ք⸮⸮(U⸮UVK%u⸮⸮⸮⸮(U⸮⸮⸮⸮⸮%u⸮ՔTQ⸮QR⸮⸮TU⸮⸮⸮T⸮i⸮⸮⸮U⸮u⸮⸮⸮⸮⸮T⸮\EQE⸮⸮jZUUEj⸮⸮P⸮E⸮QU⸮T⸮⸮⸮U⸮P⸮U⸮KRJ⸮u⸮U⸮Q⸮ZUTU⸮⸮UTE⸮⸮⸮⸮JU⸮u⸮*TJUTB⸮⸮⸮⸮V⸮E*ժBRJ⸮J⸮⸮U⸮⸮V⸮⸮U⸮[0;32mI (28) boot: ESP-IDF 22e757632 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 12:09:28[0m
[0;32mI (28) boot: chip revision: 1[0m
[0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (44) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (47) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00002000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 00008000 00001000[0m
[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (91) boot: End of partition table[0m
[0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0b57c ( 46460) map[0m
[0;32mI (129) esp_image: segment 1: paddr=0x0001b5a4 vaddr=0x3ffb0000 size=0x00100 (   256) load[0m
[0;32mI (129) esp_image: segment 2: paddr=0x0001b6ac vaddr=0x3ffb0100 size=0x000bc (   188) load[0m
[0;32mI (135) esp_image: segment 3: paddr=0x0001b770 vaddr=0x3ffb01bc size=0x00100 (   256) load[0m
[0;32mI (144) esp_image: segment 4: paddr=0x0001b878 vaddr=0x3ffb02bc size=0x01920 (  6432) load[0m
[0;32mI (156) esp_image: segment 5: paddr=0x0001d1a0 vaddr=0x3ffb1bdc size=0x00080 (   128) load[0m
[0;32mI (162) esp_image: segment 6: paddr=0x0001d228 vaddr=0x3ffb1c5c size=0x000d8 (   216) load[0m
[0;32mI (171) esp_image: segment 7: paddr=0x0001d308 vaddr=0x3ffb1d34 size=0x0004c (    76) load[0m
[0;32mI (180) esp_image: segment 8: paddr=0x0001d35c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (189) esp_image: segment 9: paddr=0x0001d764 vaddr=0x40080400 size=0x028b4 ( 10420) load[0m
[0;32mI (202) esp_image: segment 10: paddr=0x00020020 vaddr=0x400d0020 size=0x3c4cc (246988) map[0m
[0;32mI (301) esp_image: segment 11: paddr=0x0005c4f4 vaddr=0x40082cb4 size=0x1147c ( 70780) load[0m
[0;32mI (343) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (343) boot: Disabling RNG early entropy source...[0m
⸮⸮I (721) wifi:wifi driver task: 3ffb1ee8, prio:2, stack:3584, core=0
I (722) wifi:wifi firmware version: 1865b55
I (722) wifi:wifi certification version: v7.0
I (724) wifi:config NVS flash: disabled
I (727) wifi:config nano formating: disabled
I (731) wifi:Init data frame dynamic rx buffer num: 32
I (736) wifi:Init management frame dynamic rx buffer num: 32
I (741) wifi:Init management short buffer num: 32
I (746) wifi:Init dynamic tx buffer num: 32
I (750) wifi:Init static rx buffer size: 1600
I (754) wifi:Init static rx buffer num: 10
I (758) wifi:Init dynamic rx buffer num: 32
phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2
I (896) wifi:mode : softAP (8c:aa:b5:83:60:19)
I (897) wifi:Total power save buffer number: 16
I (897) wifi:Init max length of beacon: 752/752
I (901) wifi:Init max length of beacon: 752/752
*** Booting Zephyr OS build zephyr-v2.6.0-420-ge62c3c533a37  ***
I (911) wifi:mode : sta (8c:aa:b5:83:60:18)
[00:00:00.916,000] [0m<inf> esp32_wifi: WIFI_EVENT_STA_START[0m
[00:00:02.968,000] [0m<inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED[0m
[00:00:05.019,000] [0m<inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED[0m
[00:00:07.071,000] [0m<inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED[0m
[00:00:09.123,000] [0m<inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED[0m
```

sylvioalves, 2021-06-21:
@torntrousers Have you set that at the `prj.conf` file? What is the wifi security type?

torntrousers, 2021-06-21:
Yes, I found CONFIG_ESP32_WIFI_SSID and CONFIG_ESP32_WIFI_PASSWORD in zephyrproject\zephyr\samples\boards\esp32\wifi_station\prj.conf  and set those for my Wifi access point. I haven;t changed anything else , so whatever the default security type is?

torntrousers, 2021-06-21:
Any ideas @sylvioalves ?

sylvioalves, 2021-06-21:
@torntrousers I checked that with latest main branch and it worked.  Have you performed `west update` and `west espressif update` in your env? Is your wifi network configured as WPA or WPA2?

torntrousers, 2021-06-21:
Thanks @sylvioalves . Network is WPA2-Personal (according to WIndows Network Properties). The ESP32 connects to it ok with the ESP32 Arduino framework.

I had not run those west commands, I have now but after rebuilding and flashing there is no difference.
```
west update
=== updating cmsis (modules/hal/cmsis):
HEAD is now at c3bd209 DSP: fix compiler warning on inline attribute
=== updating hal_atmel (modules/hal/atmel):
HEAD is now at d17b7dd winc1500: nmspi: Don't force definition of USE_OLD_SPI_SW
=== updating hal_altera (modules/hal/altera):
HEAD is now at 23c1c1d hal_altera: Conditionally include the Altera HAL
=== updating canopennode (modules/lib/canopennode):
HEAD is now at 468d350 Merge pull request #2 from vestas-wind-systems/update_to_v1_3_master
=== updating civetweb (modules/lib/civetweb):
HEAD is now at 094aeb41 cmake: relax minimum version requirements
=== updating hal_espressif (modules/hal/espressif):
HEAD is now at 22e757632 wifi: event: move event handler to wifi driver
=== updating fatfs (modules/fs/fatfs):
HEAD is now at 1d1fcc7 fat_fs: Expose _VOLUMES to Kconfig
=== updating hal_cypress (modules/hal/cypress):
HEAD is now at 81a059f Merge pull request #1 from nandojve/pdl_scb_spi_i2c
=== updating hal_infineon (modules/hal/infineon):
HEAD is now at f1fa824 hal: infineon: initial commit of XMCLib
=== updating hal_nordic (modules/hal/nordic):
HEAD is now at 74b3b21 drivers: nrf_802154: Update the IEEE 802.15.4 component
=== updating hal_openisa (modules/hal/openisa):
HEAD is now at 40d049f openisa: Make openisa hal a zephyr_library
=== updating hal_nuvoton (modules/hal/nuvoton):
HEAD is now at b4d31f3 hal: nuvoton: add license to numicro m480 series hal
=== updating hal_microchip (modules/hal/microchip):
HEAD is now at b280eec modules: microchip: mec1501: Add Analog comparator register defines
=== updating hal_silabs (modules/hal/silabs):
HEAD is now at be39d4e modules: hal_silabs: Add Silabs Gecko HAL for EFR32FG13P SoCs
=== updating hal_st (modules/hal/st):
HEAD is now at 575de9d sensor/stmemsc: Align stmemsc i/f to v2.00
=== updating hal_stm32 (modules/hal/stm32):
HEAD is now at f8ff8d25 dts/st: use a pull-down for SPI SCK pin
=== updating hal_ti (modules/hal/ti):
HEAD is now at 3da6fae zephyr: use k_ms_to_ticks_ceil32 instead of z_ms_to_ticks
=== updating libmetal (modules/hal/libmetal):
HEAD is now at 39d049d lib: update libmetal to release v2021.04.0 + VERSION path fix
=== updating hal_quicklogic (modules/hal/quicklogic):
HEAD is now at b3a66fe Remove references to FPGA IP core drivers
=== updating lvgl (modules/lib/gui/lvgl):
HEAD is now at 31acbaa3 lv_font: Fix compiler warning when compressed fonts are disabled (#1880)
=== updating mbedtls (modules/crypto/mbedtls):
HEAD is now at 5765cb7 mbedtls: update README with laterst version information
=== updating mcuboot (bootloader/mcuboot):
HEAD is now at 7a51968 Merge: synhronize up to https://github.com/mcu-tools/mcuboot/commit/680ed07e42f
=== updating mcumgr (modules/lib/mcumgr):
HEAD is now at 5c5055f snapshot: Update mcumgr to commit e64f5060bd from upstream
=== updating net-tools (tools/net-tools):
HEAD is now at f49bd13 net-capture: Listen captured traffic and save to pcap file
=== updating hal_nxp (modules/hal/nxp):
HEAD is now at 0d11138 mxrt600: Relocate power driver to SRAM
=== updating open-amp (modules/lib/open-amp):
HEAD is now at 6010f05 lib: update open-amp lib to release v2021.04.0
=== updating loramac-node (modules/lib/loramac-node):
HEAD is now at 12019623 zephyr: radio: sx127x: override buffer read/write functions
=== updating openthread (modules/lib/openthread):
HEAD is now at f460532d4 Merge pull request #41 from kkasperczyk-no/ot_upmerge
=== updating segger (modules/debug/segger):
HEAD is now at 3a52ab2 config: Use zephyr provided options
=== updating sof (modules/audio/sof):
HEAD is now at 779f28ed4 Merge pull request #11 from nashif/kconfig_main_menu
=== updating tinycbor (modules/lib/tinycbor):
HEAD is now at 40daca9 zephyr: Remove TINYCBOR from interface libraries
=== updating tinycrypt (modules/crypto/tinycrypt):
HEAD is now at 3e9a49d cmake: Fix conditional in root CMakeLists.txt
=== updating littlefs (modules/fs/littlefs):
HEAD is now at 9e4498d lfs_util: fix log declaration when logging is disabled
=== updating mipi-sys-t (modules/debug/mipi-sys-t):
HEAD is now at 75e6715 zephyr: move platform.h to zephyr tree
=== updating nrf_hw_models (modules/bsim_hw_models/nrf_hw_models):
HEAD is now at a47e326 HW_models: NRF_RTC: Fix compare event missed when interrupt disabled
=== updating TraceRecorderSource (modules/debug/TraceRecorder):
HEAD is now at 5b5f8d7 Merge pull request #5 from nashif/update_003
=== updating hal_xtensa (modules/hal/xtensa):
HEAD is now at 2f04b61 zephyr: add SoC overlay for intel_tgl_adsp
=== updating edtt (tools/edtt):
HEAD is now at 7dd56fc Add FIXME note to handle incorrect LL/CON/MAS/BV-27-C implementation
=== updating trusted-firmware-m (modules/tee/tfm):
HEAD is now at e18b7a9b trusted-firmware-m: platform: lpcxpresso55s69: Remove pull_drivers.cmake
=== updating tfm-mcuboot (modules/tee/tfm-mcuboot):
HEAD is now at 4aa516e Bump versions for v1.7.2
=== updating nanopb (modules/lib/nanopb):
HEAD is now at d148bd2 Add zephyr module support
=== updating tensorflow (modules/lib/tensorflow):
HEAD is now at dc70a45a7cc tensorflow: add tensorflow 2.4.1 as a zephyr module
=== updating lz4 (modules/lib/lz4):
HEAD is now at 8e303c2 zephyr: make this repo a zephyr module

C:\Users\Administrator.CQ-LAP-000001\zephyrproject\zephyr>west espressif update
=== updating ESP-IDF submodules..
=== updating ESP-IDF submodules completed

C:\Users\Administrator.CQ-LAP-000001\zephyrproject\zephyr>
```

sylvioalves, 2021-06-21:
@torntrousers Is the file `prj.conf` modified (except for for ssid and pass)?

torntrousers, 2021-06-21:
@sylvioalves no. It looks like this:
```
CONFIG_WIFI=y
CONFIG_WIFI_ESP32=y
CONFIG_HEAP_MEM_POOL_SIZE=98304

CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_NET_LOG=y
CONFIG_NET_SHELL=n

CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_ECP_ALL_ENABLED=y

CONFIG_ESP32_WIFI_SSID="BT-QHATQC"
CONFIG_ESP32_WIFI_PASSWORD="xyz..."
```

torntrousers, 2021-06-22:
@sylvioalves I've tried this with my Wifi access point and also using a hotspot on my Android phone (also WPA2-Personal). Neither work with Zephyr but both work with ESP32 Arduino environment.

To workout if its some problem with my Zephyr install/build how about would it be possible for you to build the sample and send it to me so I can try you binary? My hotspot ssid/password is Balloon1/qaz123456.

Or what is some other way to debug the problem?

sylvioalves, 2021-06-22:
@torntrousers I have just tested with my phone as well and it worked with same credentials.

Find the bin file attached.

Hum, You have been working on Windows, right? I'll try in Windows as well.

```
I (729) wifi:wifi firmware version: 1865b55
I (729) wifi:wifi certification version: v7.0
I (731) wifi:config NVS flash: disabled
I (734) wifi:config nano formating: disabled
I (738) wifi:Init data frame dynamic rx buffer num: 32
I (743) wifi:Init management frame dynamic rx buffer num: 32
I (748) wifi:Init management short buffer num: 32
I (753) wifi:Init dynamic tx buffer num: 32
I (757) wifi:Init static rx buffer size: 1600
I (761) wifi:Init static rx buffer num: 10
I (765) wifi:Init dynamic rx buffer num: 32
phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2
I (894) wifi:mode : softAP (4c:11:ae:f4:98:61)
I (895) wifi:Total power save buffer number: 16
I (895) wifi:Init max length of beacon: 752/752
I (899) wifi:Init max length of beacon: 752/752
*** Booting Zephyr OS build zephyr-v2.6.0-385-g2d3195c8a2e2  ***
I (910) wifi:mode : sta (4c:11:ae:f4:98:60)
[00:00:00.915,000] <inf> esp32_wifi: WIFI_EVENT_STA_START
[00:00:02.962,000] <inf> esp32_wifi: WIFI_EVENT_STA_DISCONNECTED
I (4290) wifi:new:<11,0>, old:<1,1>, ap:<255,255>, sta:<11,0>, prof:1
I (4936) wifi:state: init -> auth (b0)
I (4942) wifi:state: auth -> assoc (0)
I (4946) wifi:state: assoc -> run (10)
I (4996) wifi:connected with Balloon1, aid = 1, channel 11, BW20, bssid = e6:d7:9e:81:74:80
I (4996) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
[zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/6693025/zephyr.zip)

I (4998) wifi:pm start, type: 1

I (5078) wifi:AP's beacon interval = 102400 us, DTIM period = 2
[00:00:05.001,000] <inf> esp32_wifi: WIFI_EVENT_STA_CONNECTED
[00:00:05.948,000] <inf> net_dhcpv4: Received: 192.168.146.247
[00:00:05.948,000] <inf> esp32_wifi_sta: Your address: 192.168.146.247
[00:00:05.948,000] <inf> esp32_wifi_sta: Lease time: 3599 seconds
[00:00:05.948,000] [0m<inf> esp32_wifi_sta: Subnet: 255.255.255.0
[00:00:05.948,000] <inf> esp32_wifi_sta: Router: 192.168.146.157
```

torntrousers, 2021-06-22:
Oh, its Zephyr isn't it - I'm on a older build than you. How do I update that?

sylvioalves, 2021-06-22:
I think it is something related to the board config itself. Let me check, I'll get back to you..

torntrousers, 2021-06-23:
@sylvioalves , I have just joined the Zephyr slack so yes I have an account now.

torntrousers, 2021-06-24:
Hi @sylvioalves , any ideas yet? Did [that binary](https://github.com/zephyrproject-rtos/zephyr/files/6696677/zephyr.zip) built on my system work for you on yours?

torntrousers, 2021-06-24:
With help from @sylvioalves on slack this has been resolved - it was the Heltec board itself and using a different ESP32 module and it all works fine. Not sure what the problem is with the Heltec board as it works fine with the Arduinio/ESP32 environemnt, but that can be for another day.

closed 2021-06-24 by torntrousers