zephyrproject-rtos/zephyr #36415: Need access to destination ip from socket recv

mspeder, 2021-06-20:
Today there is no way to retrieve destination ip address while receiving a packet thru a socket.
This is particularly useful when the interface has multiple ip addresses (for instance multicast).

A possible solution, compliant with POSIX, would be a combination of **recvmsg** and **IP_PKTINFO** socket option but recvmsg is not implemented yet in Zephyr.
Another option would be to add a dedicated function, something like recvfromto(socket, buf, len, flags, src_addr, src_addrlen, dest_addr, dest_addrlen).

jukkar, 2021-06-21:
> A possible solution, compliant with POSIX, would be a combination of **recvmsg** and **IP_PKTINFO** socket option but recvmsg is not implemented yet in Zephyr.

IMHO this would be a better solution than adding a new and non-compliant function for this. Both functions would need to dig out the destination IP address anyway so it would not make sense to implement new function for this purpose.

mspeder, 2021-06-21:
> IMHO this would be a better solution than adding a new and non-compliant function for this. Both functions would need to dig out the destination IP address anyway so it would not make sense to implement new function for this purpose.

I totally agree, however this is also a bit harder to implement. Because recvmsg is more complex with all the msghdr / iovec / control ancillary structures.
So implementing something like recvfromto would probably be easier to do, and also recvfrom could be inlined to recvfromto with NULL src_addr, maybe leading to overall smaller code footprint.

PR #65694 (2023-11-27): Add recvmsg() and pktinfo support
This PR adds support for `recvmsg()` socket call. It also adds support for `IP_PKTINFO` and `IPV6_RECVPKTINFO` socket options. Tests for both of these features are also added.

Fixes #36415