zephyrproject-rtos/zephyr #36417: tests-ci :coredump.logging_backend : zephyr-v2.6.0-286-g46029914a7ac: lpcxpresso55s28: test failed

hakehuang, 2021-06-21:
**Describe the bug**
coredump.logging_backend test is failed on zephyr-v2.6.0-286-g46029914a7ac on lpcxpresso55s28

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28 --testcase-root tests --sub-test coredump.logging_backend
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-286-g46029914a7ac  ***
Coredump: lpcxpresso55S28
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-286-g46029914a7ac

henrikbrixandersen, 2021-06-23:
@hakehuang You already reported this in #36078?

hakehuang, 2021-06-24:
@henrikbrixandersen sorry, I forget to double check the repo. let me clsoe this one.

closed 2021-06-24 by hakehuang