zephyrproject-rtos/zephyr #36419: test-ci: net.ethernet_mgmt: zephyr-v2.6.0-286-g46029914a7ac: frdm_k64f: test fails

hakehuang, 2021-06-21:
**Describe the bug**
net.ethernet_mgmt: zephyr-v2.6.0-286-g46029914a7ac: frdm_k64f: test fails

**To Reproduce**
Steps to reproduce the behavior:
cd tests/net/ethernet_mgmt
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
ethernet management

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v2.6.0-286-g46029914a7ac  ***

Running test suite ethernet_mgmt_test

===================================================================

START - test_setup

 PASS - test_setup in 0.1 seconds

===================================================================

START - test_change_mac_when_up

 PASS - test_change_mac_when_up in 0.1 seconds

===================================================================

START - test_change_mac_when_down

 PASS - test_change_mac_when_down in 0.1 seconds

===================================================================

START - test_change_auto_neg

 PASS - test_change_auto_neg in 0.1 seconds

===================================================================

START - test_change_to_same_auto_neg

 PASS - test_change_to_same_auto_neg in 0.1 seconds

===================================================================

START - test_change_link

 PASS - test_change_link in 0.1 seconds

===================================================================

START - test_change_same_link

 PASS - test_change_same_link in 0.1 seconds

===================================================================

START - test_change_unsupported_link

 PASS - test_change_unsupported_link in 0.1 seconds

===================================================================

START - test_change_duplex

 PASS - test_change_duplex in 0.1 seconds

===================================================================

START - test_change_same_duplex

 PASS - test_change_same_duplex in 0.1 seconds

===================================================================

START - test_change_qav_params

 PASS - test_change_qav_params in 0.1 seconds

===================================================================

START - test_change_qbv_params

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:915: test_change_qbv_params: (ret not equal to 0)

could not get the number of ports (-134)

 FAIL - test_change_qbv_params in 0.16 seconds

===================================================================

START - test_change_qbu_params

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:1091: test_change_qbu_params: (ret not equal to 0)

could not get the number of ports (-134)

 FAIL - test_change_qbu_params in 0.16 seconds

===================================================================

START - test_change_txtime_params

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:1292: test_change_txtime_params: (ret not equal to 0)

could not get the number of priority queues

 FAIL - test_change_txtime_params in 0.16 seconds

===================================================================

START - test_change_promisc_mode_on

 PASS - test_change_promisc_mode_on in 0.1 seconds

===================================================================

START - test_change_to_same_promisc_mode

 PASS - test_change_to_same_promisc_mode in 0.1 seconds

===================================================================

START - test_change_promisc_mode_off

 PASS - test_change_promisc_mode_off in 0.1 seconds

===================================================================

Test suite ethernet_mgmt_test failed.

===================================================================

PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.6.0-286-g46029914a7ac

hakehuang, 2021-06-23:
@dleach02 all RT series have the same problem. mimxrt1060_evk
```
*** Booting Zephyr OS build zephyr-v2.6.0-436-gfbf5b8e8deb1  ***
Running test suite ethernet_mgmt_test
===================================================================
START - test_setup
PASS - test_setup in 0.1 seconds
===================================================================
START - test_change_mac_when_up
PASS - test_change_mac_when_up in 0.1 seconds
===================================================================
START - test_change_mac_when_down
PASS - test_change_mac_when_down in 0.1 seconds
===================================================================
START - test_change_auto_neg
PASS - test_change_auto_neg in 0.1 seconds
===================================================================
START - test_change_to_same_auto_neg
PASS - test_change_to_same_auto_neg in 0.1 seconds
===================================================================
START - test_change_link
PASS - test_change_link in 0.1 seconds
===================================================================
START - test_change_same_link
PASS - test_change_same_link in 0.1 seconds
===================================================================
START - test_change_unsupported_link
PASS - test_change_unsupported_link in 0.1 seconds
===================================================================
START - test_change_duplex
PASS - test_change_duplex in 0.1 seconds
===================================================================
START - test_change_same_duplex
PASS - test_change_same_duplex in 0.1 seconds
===================================================================
START - test_change_qav_params
PASS - test_change_qav_params in 0.1 seconds
===================================================================
START - test_change_qbv_params
Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:915: test_change_qbv_params: (ret not equal to 0)
could not get the number of ports (-134)
FAIL - test_change_qbv_params in 0.16 seconds
===================================================================
START - test_change_qbu_params
Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:1091: test_change_qbu_params: (ret not equal to 0)
could not get the number of ports (-134)
FAIL - test_change_qbu_params in 0.16 seconds
===================================================================
START - test_change_txtime_params
Assertion failed at WEST_TOPDIR/zephyr/tests/net/ethernet_mgmt/src/main.c:1292: test_change_txtime_params: (ret not equal to 0)
could not get the number of priority queues
FAIL - test_change_txtime_params in 0.16 seconds
===================================================================
START - test_change_promisc_mode_on
PASS - test_change_promisc_mode_on in 0.1 seconds
===================================================================
START - test_change_to_same_promisc_mode
PASS - test_change_to_same_promisc_mode in 0.1 seconds
===================================================================
START - test_change_promisc_mode_off
PASS - test_change_promisc_mode_off in 0.1 seconds
===================================================================
Test suite ethernet_mgmt_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

dleach02, 2021-06-25:
These test cases fail because the underlying ethernet driver does not support api->get_config(). Searching the code base there are only two drivers that support this so I'm curious what targets are actually run against this particular test suite.

I'm going to submit a PR to skip the problem test cases if the underlying management function returns ENOTSUP.

PR #36761 (2021-07-08): tests: net: ethernet_mgmt: Fix tests when running in real hw
If the test is run in a board that has a network interface
enabled, then the tests might select and use wrong network interface
that is needed for the test. So this commit makes sure that
we always use the simulated network interface for the
Ethernet management tests.

Fixes #36419
commit fe3a11ba (2021-07-08): tests: net: ethernet_mgmt: Fix tests when running in real hw

If the test is run in a board that has a network interface
enabled, then the tests might select and use wrong network interface
that is needed for the test. So this commit makes sure that
we always use the simulated network interface for the
Ethernet management tests.

Fixes #36419
closed 2021-06-27 by hakehuang