zephyrproject-rtos/zephyr #36427: test: kernel.common.nano32: zephyr-v2.6.0-286-g46029914a7ac: mimxrt1060_evk: test fails

hakehuang, 2021-06-21:
**Describe the bug**

kernel.common.nano32: zephyr-v2.6.0-286-g46029914a7ac: mimxrt1060_evk: test fails

**To Reproduce**
Steps to reproduce the behavior:
cd tests/kernel/common
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**

test pass

**Impact**

kernel common nano32 support

**Logs and console output**

```
 *** Booting Zephyr OS build zephyr-v2.6.0-286-g46029914a7ac  (delayed boot 500ms) ***

    Running test suite common

    ===================================================================

    START - test_bootdelay

     PASS - test_bootdelay in 0.1 seconds

    ===================================================================

    START - test_irq_offload

    [******** Truncated '112' Lines Log ********]

     PASS - test_printk in 0.27 seconds

    ===================================================================

    START - test_timeout_order

     PASS - test_timeout_order in 0.126 seconds

    ===================================================================

    START - test_clock_uptime

     PASS - test_clock_uptime in 0.12 seconds

    ===================================================================

    START - test_clock_cycle

     PASS - test_clock_cycle in 0.3 seconds

    ===================================================================

    START - test_version

     PASS - test_version in 0.1 seconds

    ===================================================================

    START - test_multilib

     PASS - test_multilib in 0.1 seconds

    ===================================================================

    START - test_thread_context

     PASS - test_thread_context in 0.31 seconds

    ===================================================================

    START - test_errno

     PASS - test_errno in 0.1 seconds

    ===================================================================

    START - test_ms_time_duration

     PASS - test_ms_time_duration in 0.182 seconds

    ===================================================================

    START - test_bounds_check_mitigation

     PASS - test_bounds_check_mitigation in 0.1 seconds

    ===================================================================

    START - test_nop

    time k_cycle_get_32() takes 182 cycles

    arch_nop() takes 0 cycles

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/common/src/irq_offload.c:173: test_nop: (diff > 0 is false)

    arch_nop() takes 0 cpu cycles

     FAIL - test_nop in 0.19 seconds

    ===================================================================

    Test suite common failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.6.0-286-g46029914a7ac

dleach02, 2021-06-25:
@hakehuang, I'm not able to reproduce this failure.

hakehuang, 2021-06-27:
@dleach02 , let me double check it on latest code base.

dleach02, 2021-06-29:
Any update @hakehuang ?

hakehuang, 2021-06-29:
passed on zephyr-v2.6.0-560-ga6cc3f1838

closed 2021-06-29 by hakehuang