zephyrproject-rtos/zephyr #36435: RFC: API Change: Mesh

michal-narajowski, 2021-06-21:
### Problem description

The current Mesh API is missing a return value for opcode callbacks to get meaningful information from the callback. One example of such information is message length check. Some Mesh messages have fixed length while others have variable length. Current Mesh implementation only verifies that the incoming message is not shorter than the minimum length. According to Mesh Profile Specification, the implementation should verify the message length and ignore those of invalid length. Fixed-length messages can be verified by the Access layer while variable-length messages can only be checked by callback implementation. This is why we introduce two new macros `BT_MESH_LEN_EXACT(len)` and `BT_MESH_LEN_MIN(len)`. The former shall be used when defining a fixed-length message and will be verified by the Access layer. The latter is for the variable-length messages and additional checks must be done inside callback implementation. This change is backward compatible, so not using these macros is the same as using `BT_MESH_LEN_MIN(len)`.

### Proposed change

PR https://github.com/zephyrproject-rtos/zephyr/pull/34606 implements the proposed change and adjusts in-tree opcode definitions.

### Dependencies

Steps for out-of-tree users to adjust to this API change:

- Change return type of each callback function used inside `struct bt_mesh_model_op` definition.
- Add return statements inside callback implementations.
- Update `len` field inside `struct bt_mesh_model_op` and use `BT_MESH_LEN_EXACT(len)` or `BT_MESH_LEN_MIN(len)` macros. 
- Add message length verificaiton inside variable-length message callbacks.

carlescufi, 2021-07-13:
API meeting: no objections raised.

carlescufi, 2021-07-20:
Closed by #34606

closed 2021-07-20 by carlescufi