zephyrproject-rtos/zephyr #36447: net: socket: socketpair: Poll call resetting all events

romaric35, 2021-06-22:
**Describe the bug**
At the beginning of poll() call every events on all file descriptors are systematically cleared, here :
socketpair.c: zsock_poll_prepare_ctx(): k_poll_signal_reset((*pev)->obj);

**To Reproduce**
- scenario 1 (on file descriptor addition to poll):
	Create two threads (one will be called client and one server), one socketpair with one end belonging to server
	and the other end to client. On server side make a call to poll() with this fd immediatelly. On client side try to write immediately.
	Start Client thread long before server thread. Poll() on server side will never exit.
- scenario 2 (in preemptive case only):
	While handling events after a poll() the server performs a read/write operation, the write call reschedule the client. Then the client
	will immediatelly send a request to the server and wait its response with a poll(). The poll() reschedules the server that will begin a new poll loop,
	thus resetting all the events including our new client request.

**Expected behavior**
Beginning a poll() should not reset all the incoming events and
events shall be reset only when they are treated.
- scenario 1 :
	We expect poll() to not reset all the events after addition of a file descriptor to fd set in order for the server to treat client request.
- scenario 2 :
	We expect the poll() not to reset the events that would occur in the middle of two server poll() calls.

**Impact**
Loss of events on file descriptor while using poll() call on socketpair and application deadlocks when both 
client and server use poll() to synchronise communication.

**Environment (please complete the following information):**
 - OS: WSL2 on Windows
 - Toolchain Zephyr SDK : 11.4

**Additional context**
fix proposition:
 - Suppress reset signal write in prepare stage of poll function
 - Put reset signal at the end of a application read on socket
[0001-patch-kernel-socketpair.c.zip](https://github.com/zephyrproject-rtos/zephyr/files/6692431/0001-patch-kernel-socketpair.c.zip)

rlubos, 2021-06-23:
Hi @romaric35, thank you for the report.

Indeed, resetting the signal at the beginning of the `poll()` doesn't seem to be a right approach, I've reproduced the faulty behaviour with a proper unit test. I think that you fix proposal looks good, however it doesn't seem to be complete, as it covers only `POLLIN` case, some more changes are needed in order to make `POLLOUT` work as expected.

I'll try to post a PR soon with a fix proposal and unit tests covering the case.

FYI @cfriedt, an early notice, I'll likely ask you to review the fix since you're the author of the socketpair.

cfriedt, 2021-06-23:
Thanks for the patch @romaric35 - that's an interesting case that I'm sure we should have a test for :-) @rlubos if you're ok making a PR, I would be happy to review when you are ready.

rlubos, 2021-06-24:
@romaric35 @cfriedt I've submitted a PR witha a fix, please have a look: https://github.com/zephyrproject-rtos/zephyr/pull/36516

PR #36516 (2021-07-03): net: sockets: socketpair: Fix poll signalling
In case read or write were called before the actual poll() call, the poll() function was not signalled correctly about such events, which in order could lead to a deadlock if the poll() was called with infinite timeout.

Additionally, add unit tests that cover this scenario, to make sure poll() events are signalled correctly.

Fixes #36447
closed 2021-07-03 by nashif