zephyrproject-rtos/zephyr #36462: [bluetooth stack][limited_discoverable_advertising timeout] Some problems about the lim_adv_timeout

761669642, 2021-06-23:
Hi, when I test zephry ble stack on my bluetooth board with auto-pts, I meet a problem about the timeout of the limited discoverable advertising. The Bluetooth spec5.0 declare that "Devices shall remain in the limited discoverable mode no longer than
TGAP(lim_adv_timeout)." The default timeout is 180s. But when I test it, it can never stop although that the ad type flag is LimitedDiscoverable.
So, I have some problems about it.
1. Does zephry have the feature about the timeout of the limited discovering adv? something just like  macro to decide it? or call functions to set it?
2. Which layer should realize this feature? The controller stack, zephry host stack or application? 
3. The PTS case will fail if it cannot stop after lim_adv_timeout when on limited discoverable mode. So what can I do for it? Should I realize it in app layer? 
Thank you!

joerchan, 2021-08-16:
Hi. The automatic stopping of an advertiser when using the limited flag was added here: https://github.com/zephyrproject-rtos/zephyr/pull/36776
If you are on a revision before that you will have to handle this in the application.

761669642, 2021-09-02:
Thank you all so much for your efforts!

joerchan, 2021-09-02:
No problem. Closing this as the question has been answered.

closed 2021-09-02 by joerchan