zephyrproject-rtos/zephyr #36476: Add intel HAL support

kumarnac, 2021-06-23:
## Origin
https://github.com/intel/hal-intel

## Purpose
Provides  HAL support  for  intel boards and  intel SoC  specific hardware definitions to be referred by boards in zephyr 

## Mode of integration
Mode of integration - Module 
Module Name:  hal_intel

## Pull Request

Pull request (if any) with the actual implementation of the integration, be it
in the main tree or as a module (pointing to your own fork for now). Make sure
the PR is correctly labeled as "DNM"

## Description

Long description that will help reviewers discuss suitability of the
component to solve the problem at hand (there may be a better options
available.)

What is its primary functionality (e.g., SQLLite is a lightweight
database)?

What problem are you trying to solve? (e.g., a state store is
required to maintain ...)

Why is this the right component to solve it (e.g., SQLite is small,
easy to use, and has a very liberal license.)

# Dependencies

What other components does this package depend on?

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?

## Revision

Version or SHA you would like to integrate initially

## License

``BSD-3-Clause``

kumarnac, 2021-06-30:
The HAL  is required for supporting  ehl_pse_crb board  (https://github.com/kumarnac/zephyr/pull/1)  in zephyr

nashif, 2022-08-29:
closing, if we need to introduce this hal for intel, a new issue will need to be created.

closed 2022-08-29 by nashif