zephyrproject-rtos/zephyr #36495: dtc generates missing #address-cells in interrupt provider warning

ejohnso49, 2021-06-23:
**Describe the bug**
`dtc` version 1.6.1 produces a warning when building applications for ARMv7-m targets. This warning does not cause builds to fail. I recently upgraded my `dtc` so I'm not entirely sure what `dtc` version this was introduced with.

**To Reproduce**
Steps to reproduce the behavior:
1. Install `dtc` version 1.6.1
2. Run `west build -b nrf52833dk_nrf52833 zephyr/samples/hello_world` (hello_world path depends on workspace setup)

**Expected behavior**
No warnings produced when `dtc` is run.

**Impact**
This is  a small annoyance, but will be encountered frequently

**Logs and console output**
```
nrf52833dk_nrf52833.dts.pre.tmp:15.39-20.5: Warning (interrupt_provider): /soc/interrupt-controller@e000e100: Missing #address-cells in interrupt provider
  also defined at nrf52833dk_nrf52833.dts.pre.tmp:354.7-356.3
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: arm-gcc v 9.3.1
 - Commit SHA or Version used: v2.6.0

**Additional context**
Add the following line seems to remove the warning but I am not knowledgeable enough on the DeviceTree to know if this
fix is valid or how to apply it to other architectures:
```diff
diff --git a/dts/arm/armv7-m.dtsi b/dts/arm/armv7-m.dtsi
index cd244f4594..9f2dc66734 100644
--- a/dts/arm/armv7-m.dtsi
+++ b/dts/arm/armv7-m.dtsi
@@ -11,6 +11,7 @@
                ranges;
 
                nvic: interrupt-controller@e000e100  {
+                       #address-cells = <1>;
                        compatible = "arm,v7m-nvic";
                        reg = <0xe000e100 0xc00>;
                        interrupt-controller;
```

It looks like this will affect other ARM cores but again I'm far from an expert.

galak, 2021-06-23:
@ejohnso49 this fix looks good.  Can you send a PR with this and add fix to all the files in `dts/arm`

ejohnso49, 2021-06-23:
@galak Researching a little bit more and I think this applies to any node that has the `interrupt-controller` property. I also read up on `#address-cells`. I think my prior suggestion is incorrect.

I do not see any dts files defining nodes inside of interrupt-controller nodes as far as I can tell. I also found this example in some Linux dts files: https://lore.kernel.org/linux-devicetree/91e3405245c89f134676449cf3822285798d2ed2.1612189652.git.michal.simek@xilinx.com/

Since these nodes don't have any children I think setting `#address-cells` to `<0>` is correct.

ejohnso49, 2021-06-23:
Further research seems to say that `interrupt-controllers` will only have non-zero `#address-cells` if `interrupt-maps` are used. I didn't find any of this usage in the code base except for `edtlib` testing.

mbolivar-nordic, 2021-07-21:
@cfriedt I already approved the PR; we are waiting on @galak to approve it before merging. Why did you assign me and unassign him? Seems backwards...

PR #36499 (2021-08-02): dts: arm: Add #address-cells to nvic nodes
dtc v1.6.1 produces an error without specifying #address-cells on all interrupt-controller nodes. This commit only fixes this for `nvic` nodes in ARM architectures.

Fixes #36495
commit 7147e68f (2021-08-02): dts: arm: Add #address-cells to nvic nodes

dtc v1.6.1 produces an error without specifying #address-cells
on an interrupt-controller node.

Fixes #36495
closed 2021-08-02 by cfriedt