zephyrproject-rtos/zephyr #36496: bluetooth: only the first Extended Advertising Report with data status "incomplete, more data to  come" is issued

ArneJoris, 2021-06-23:
**Describe the bug**
When receiving an AUX_ADV_IND advertisement with 60 bytes of extended advertising data, zephyr issues only a single LE Extended Advertising Report meta event, with the first 42 bytes of the data and a data status of 'Incomplete, more data to come'.  

No further LE Extended Advertising Report meta events are sent and the last 18 bytes of advertising data are never communicated to the host.

**To Reproduce**
Both peripheral and central are using nRF52840 usb dongles (Board: nrf52840dongle_nrf52840) with zephyr 2.6 firmware, running the hci_usb sample.

Peripheral:
   ```
# LE Set Extended Advertising Parameters: handle 01, Connectable, Primary and secondary PHY is Coded 
hcitool cmd 0x08 0x0036 0x01 0x01 0x00 0x00 0x01 0x00  0x00 0x08 0x00 0x07 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x7f 0x03 0x04 0x03 0x0A 0x01 

#  LE Set Extended Advertising Data, operation is Complete extended advertising set with data length 0x3C
#  data type 0x01 Flags 06 
#  data type 0x03 16-bit service UUIDs 0x180a, 0x180f, 0x3010 
#  data type 0x07 128-bit service UUID b4703001-eb6b-4b82-ae21-cf035a429a94 
#  data type 0xFF Manufacturer specific data: manufacturer 0x9999, data 12 0x31 characters
#  data type 0x09 Complete local name: W123456789010 
hcitool cmd 0x08 0x0037 0x01 0x03 0x00 0x3C 0x02 0x01 0x06 0x07 0x03 0x0A 0x18 0x0F 0x18 0x10 0x30 0x11 0x07 0x94 0x9A 0x42 0x5A 0x03 0xCF 0x21 0xAE 0x82 0x4B 0x6B 0xEB 0x01 0x30 0x70 0xB4 0x0f 0xff 0x99 0x99 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x31 0x0e 0x09 0x57 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38 0x39 0x30 0x31 0x32 

# LE Set Extended Advertising Enable
hcitool cmd 0x08 0x0039 0x01 0x01 0x01 0x00 0x00 0x00 
```

Central:

```
# set default phy to encoded 
hcitool cmd 08 31 03 04 04 

# LE Set Extended Scan Parameters: active scan on coded PHY
hcitool cmd 0x08 0x0041 0x01 0x00 0x04 0x01 0xFF 0x01 0xFF 0x00 

#LE Set Event Mask to get extended advertising events
hcitool cmd 0x08 0x0001 0x00 0x18 0x00 0x00 0x00 0x00 0x00 0x00 

# LE Set Extended Scan Enable: scan continuously
hcitool cmd 0x08 0x0042 0x01 0x00 0x00 0x00 0x00 0x00
``` 

**Expected behavior**
Bluetooth core specification v5.2 says in 7.7.65.13 LE Extended Advertising Report event (page 2402):

> The Controller may split the data from a single advertisement (whether one 
> PDU or several) into several reports. If so, each report except the last shall 
> have an Event_Type with a data status field of "incomplete, more data to 
> come", while the last shall have the value "complete";

The expected behavior is for another LE Extended Advertising Report meta event to be issued with the remaining 18 bytes of advertising data and a data status of 'complete'.

**Impact**
All advertising payload is needed to communicate the peripheral's state to the central; not having the last 18 bytes is a showstopper for my project.

**Logs and console output**

On central, run 'btmon -T' to see LE meta events:

```
> HCI Event: LE Meta Event (0x3e) plen 68                                                                                             #46009 [hci0] 2021-06-23 14:55:48.691151
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0021
            Props: 0x0001
              Connectable
            Data status: Incomplete, more data to come
          Address type: Random (0x01)
          Address: E0:CF:78:95:AC:AD (Static)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x0a
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x2a
        02 01 06 07 03 0a 18 0f 18 10 30 11 07 94 9a 42  ..........0....B
        5a 03 cf 21 ae 82 4b 6b eb 01 30 70 b4 0f ff 99  Z..!..Kk..0p....
        99 31 31 31 31 31 31 31 31 31                    .111111111
> HCI Event: LE Meta Event (0x3e) plen 68                                                                                             #46010 [hci0] 2021-06-23 14:55:49.029170
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0021
            Props: 0x0001
              Connectable
            Data status: Incomplete, more data to come
          Address type: Random (0x01)
          Address: E0:CF:78:95:AC:AD (Static)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x0a
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x2a
        02 01 06 07 03 0a 18 0f 18 10 30 11 07 94 9a 42  ..........0....B
        5a 03 cf 21 ae 82 4b 6b eb 01 30 70 b4 0f ff 99  Z..!..Kk..0p....
        99 31 31 31 31 31 31 31 31 31                    .111111111
```

A bluetooth sniffer sees the entire advertising payload:

```
================================================================================
AdvA: E0:CF:78:95:AC:AD (Static) RSSI: -41 Hits: 4

Advertisement:
Timestamp: 0.956975     Length: 72      RSSI: -41       Channel: 0      PHY: Coded
Ad Type: AUX_ADV_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 70
AdvMode: Connectable
AdvA: E0:CF:78:95:AC:AD (Static) AdvDataInfo: 01 A0
47 46 49 09 AD AC 95 78 CF E0 01 A0 02 01 06 07 03 0A 18 0F 18 10 30 11 07 94 9A 42 5A 03 CF 21 AE 82 4B 6B EB 01 30 70 B4 0E 09 57 32 36 34 38 38 38 39 36 30 39 39 38 11 FF 99 99 31 31 31 31 31 31 31 31 31 31 31 31
G  F  I              x                                            0              B  Z        !        K  k        0  p           W  2  6  4  8  8  8  9  6  0  9  9  8
     1  1  1  1  1  1  1  1  1  1  1  1

Scan Response: None
================================================================================
```

**Environment (please complete the following information):**
- OS: linux debian 10 with kernel 5.10.33-v7l+
-  Toolchain: zephyr-sdk-0.12.2
-  Commit SHA or Version used: zephyr tag 2.6

cvinayak, 2021-06-24:
@ArneJoris Please provide any Kconfig changes in your hci_usb sample's prj.conf file.

ArneJoris, 2021-06-25:
The KConfig for the central is:
```CONFIG_GPIO=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_SERIAL=y
CONFIG_TINYCRYPT=y
CONFIG_TINYCRYPT_SHA256=y
CONFIG_TINYCRYPT_AES=y
CONFIG_MCUBOOT_BOOTUTIL_LIB=y
CONFIG_SOC_SERIES_NRF52X=y
CONFIG_SOC_NRF52840_QIAA=y
CONFIG_NFCT_PINS_AS_GPIOS=y
CONFIG_ARM_MPU=y
CONFIG_HW_STACK_PROTECTION=y
CONFIG_CONSOLE=y
CONFIG_ENTROPY_NRF5_THR_THRESHOLD=4
CONFIG_ENTROPY_NRF5_ISR_THRESHOLD=12
CONFIG_USB=y
# CONFIG_BT_PERIPHERAL is not set
CONFIG_BT_BROADCASTER=y
CONFIG_BT_EXT_ADV=y
CONFIG_BT_EXT_ADV_MAX_ADV_SET=2
CONFIG_BT_BUF_EVT_RX_SIZE=68
CONFIG_BT_BUF_EVT_DISCARDABLE_SIZE=43
CONFIG_BT_BUF_CMD_TX_SIZE=65
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y
# CONFIG_BT_CTLR_PRIVACY is not set
# CONFIG_BT_CTLR_PHY_2M is not set
CONFIG_BT_CTLR_ADV_EXT=y
CONFIG_BT_CTLR_SET_HOST_FEATURE=y
CONFIG_BT_CTLR_SETTINGS=y
CONFIG_BT_CTLR_VERSION_SETTINGS=y
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_ADV_DATA_BUF_MAX=2
CONFIG_BT_CTRL_ADV_ADI_IN_SCAN_RSP=y
CONFIG_BT_CTLR_SCAN_AUX_SET=2
CONFIG_BT_CTLR_CONN_RSSI_EVENT=y
CONFIG_BT_CTLR_ADV_INDICATION=y
CONFIG_BT_CTLR_SCAN_REQ_RSSI=y
CONFIG_USB_DEVICE_BLUETOOTH=y
CONFIG_SETTINGS=y

ArneJoris, 2021-06-25:
I can work around the problem by setting CONFIG_BT_BUF_EVT_RX_SIZE=255.

cvinayak, 2021-06-27:
HCI fragmentation of advertising reports is not implemented. This is something to be consider after chaining support is completed.

carlescufi, 2021-07-15:
Closing since fragmentation of advertising reports is not supported. Please open an enhancement instead if you'd like that implemented.

closed 2021-07-15 by carlescufi