zephyrproject-rtos/zephyr #36517: RFC: API Change: Flash map

de-nordic, 2021-06-24:
## Introduction

Proposing change in how flash partition identifiers are generated, from DTS, to generate IDs for enabled partitions only.

### Problem description

Currently IDs for all partitions are generated regardless whether the partition is enabled or not;
whether partition is enabled or not does not effect other partitions as they are given fixed positions.
The ID's are, in most cases, not used directly but rather obtained at compilation time with macros that dig
them out of DTS, by label or node.
This means that it is possible to use in code ID of partition that does not even exist or is available for
access, which will only be detected when access to such partition is performed at run-time.
There is additional macro that allows to check if partition is enabled, but it has been rather designed to include/exclude some code from compilation, basing on whether partition exists; the reference to not-enabled partition should itself cause error during compilation as it would allow to detect simple mistake.

The current implementation of flash map API uses ID to figure out description of partition, by searching array of flash_area objects and comparing the ID, with sequential IDs the objects that represent the partitions could be just indexed.

### Proposed change

Change to gen_defines.py that will generate IDs (_PARTITION_ID) for existing and enabled partitions only.
PR with proposed changes is here: https://github.com/zephyrproject-rtos/zephyr/pull/36394

## Detailed RFC

### Dependencies

Flash map, everyone that uses IDs.

### Concerns and Unresolved Questions

Is somebody using IDs directly, as hardcoded literals, rather than picking them from DTS?
The IDs that the user displays, or passes somewhere may get different after the change.

## Alternatives

Leaving the thing as it is.

galak, 2021-07-13:
Can we use the already generated ORDS value?

de-nordic, 2021-07-13:
The problem with ORDS is that they are unique identification of nodes across the entire DTS definition,, which means that the partition IDs generated from ORDs will depend on the node they are taken from, and can have non-consecutive values.
The proposal is intended to bring 0 based, consecutive ID values to PARTITION_ID, so that it would be possible to optimize access by ID, for example by collecting all defined partitions into array and simply indexing the array by ID.

carlescufi, 2021-07-13:
API meeting

- `struct device *partitions[NUM_PARTITIONS]` to translate from Partition ID to `struct device *` by using IDs as indices directly
- Would enforce partition IDs to be 0 .. N-1.
- @galak suggests using the preprocessor instead with `FOREACH` to generate the macros, instead of using `gen_defines.py`.
- Document the assumptions/constraints of Partition IDs (linearly ascending)

galak, 2021-07-13:
It looks like we do most of what I'm thinking with in `subsys/storage/flash_map/flash_map_default.c` already.  And it would just be using the index of default_flash_map[] for the ID.

de-nordic, 2021-07-13:
> 
> 
> It looks like we do most of what I'm thinking with in `subsys/storage/flash_map/flash_map_default.c` already. And it would just be using the index of default_flash_map[] for the ID.

The problem is that you do something lie FLASH_AREA_ID(storage) and this will give you 6, and you can not index with that number, because you have only, for example, 4 partitions enabled, but all are counted when ID is generated.

closed 2023-06-07 by de-nordic