zephyrproject-rtos/zephyr #36524: HSE clock doesn't initialize and blinky doesn't run on custom board when moving from zephyr v2.3.0 to v2.6.0

vandita5, 2021-06-24:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Our applications and custom boards are currently based off of zephyr-v2.3.0 (tag). We are looking to upgrade to zephyr-v2.6.0 to avail some improvements made recently and since 2.3.0. The custom board was based off of [nucleo_f429zi board from zephyr-v2.3.0](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.3.0/boards/arm/nucleo_f429zi) with changes to pins used, additional peripherals defined and used. I made some minimal changes needed to be able to build when I moved to zephyr-v2.6.0. Copying changes in [nucleo_f429zi board from zephyr-v2.6.0](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.6.0/boards/arm/nucleo_f429zi/nucleo_f429zi.dts). The clocks are now initialized in the dts but I didn't use pinctrl dtsi and gpios are still defined as with the older versions from the pinmux.c and within peripheral definition as below. Sharing snippets of these:
--- clock init in dts files ----
```
&clk_hse {
	hse-bypass;
	clock-frequency = <DT_FREQ_M(8)>; /* STLink 8MHz clock */
	status = "okay";
};

&pll {
	div-m = <8>;
	mul-n = <336>;
	div-p = <2>;
	div-q = <7>;
	clocks = <&clk_hse>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(168)>; /* highest value to get a precise USB clock */
	ahb-prescaler = <1>;
	apb1-prescaler = <4>;
	apb2-prescaler = <2>;
};
```
---- periph init with gpios defined old way ----
```
&i2c1 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

	bhi160@28 {
		compatible = "bosch,bhi160";
		reg = <0x28>;
		int-gpios = <&gpiod 11 GPIO_ACTIVE_HIGH>;
		label = "BHI160";
	};
};
```

Pinmux file looks similar to the pinmux.c for nucleo_f429zi board in zephyr-v2.3.0

I am able to build basic/blinky app in zephyr_v2.6.0 for this custom baord after these changes and am able to flash it. Based on debugging, it seems it never moves past checking if HSE oscillation is ready (LL_RCC_HSE_IsReady function in stm32f4xx_ll_rc.h).

Are there additional changes that I'm missing here?

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. git checkout zephyr-v2.6.0
3. west update
4. build samples/basic/blinky for custom board and flash it on

**Expected behavior**
A clear and concise description of what you expected to happen.
leds to blink

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
showstopper. Can't use zephyr v2.6.0

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) mac big sur (11) and windows 10
 - Toolchain (e.g Zephyr SDK, ...) zephyr sdk with gnuarmembedded
 - Commit SHA or Version used tags zephyr-v2.6.0 and zephyr-v2.3.0

**Additional context**
Add any other context about the problem here.

vandita5, 2021-06-24:
additional comment - I'm able to build and run the samples and some of our custom apps for the nucleo_f429zi board and run them as well without issues. It's jsut the cusotm board which isn't working so my guess is that I've missed some essential changes to upgrade the board files.

erwango, 2021-06-24:
@vandita5 Can you provide the working Kconfig configuration you were using for clock in V2.3.0?

vandita5, 2021-06-24:
Hey Erwan, I looked through the original config and realized that we didn't want to bypass the hse clock anymore (as it basically bypasses and uses the clock from the stlink on the nucleo baord). I removed that part from the dts definition and the code runs now. Thanks for your question resulting in that finding. :)

I am still missing some config on my board as I was trying to run echo_server and I get these logs when I try to debug it
```
00> [00:00:00.000,000] <inf> net_config: Waiting interface -1 ((nil)) to be up...
00> [00:00:00.000,000] <err> net_if: There is no network interface to work with!
00> [00:00:00.000,000] <inf> net_config: Initializing network
00> [00:00:00.000,000] <inf> net_config: Waiting interface -1 ((nil)) to be up...
00> [00:01:33.003,000] <err> net_config: Timeout while waiting network interface
00> [00:01:33.003,000] <err> net_config: Network initialization failed (-115)
00> [00:01:33.004,000] <inf> net_echo_server_sample: Run echo server
```
So, do you have any ideas on what other config I could be missing? Or do you know of any good documentation that could help setup the baord properly? I have not yet changed how pins are defined for the baord and am still using pinmux vs pinctrl.

vandita5, 2021-06-24:
I also get this exception whenever I try to debug any app with vscode
[image]

vandita5, 2021-06-24:
I made two changes that helped run echo_server sample.
1. I added these to the dts file 
```
&mac {
	status = "okay";
	pinctrl-0 = <&eth_mdc_pc1
		     &eth_rxd0_pc4
		     &eth_rxd1_pc5
		     &eth_ref_clk_pa1
		     &eth_mdio_pa2
		     &eth_crs_dv_pa7
		     &eth_tx_en_pg11
		     &eth_txd0_pb12
		     &eth_txd1_pb13>;
};
```
2. I added `CONFIG_CUSTOM_BOARD=y` to the board's defconfig file. This basically hacks `CONFIG_ETH_STM32_HAL` to y similar to [here](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.3.0/boards/arm/nucleo_f429zi/Kconfig.defconfig#L16). 
this config has been changed to be enabled only if a compatible device is enabled. I don't see such a device being enabled in nucleo_f429zi but I'm able to use it as it is and get data over a socket.
snippet from Kconfig.stm32hal for zephyr v2.6.0
```
menuconfig ETH_STM32_HAL
	bool "STM32 HAL Ethernet driver"
	default y if $(dt_compat_enabled,$(DT_COMPAT_ST_STM32_ETHERNET))
	select USE_STM32_HAL_ETH
	select NOCACHE_MEMORY if SOC_SERIES_STM32H7X && CPU_CORTEX_M7
	help
	  Enable STM32 HAL based Ethernet driver. It is available for
	  all Ethernet enabled variants of the F2, F4, F7 and H7 series.
```
Again points to possible missign configs.

erwango, 2021-06-25:
> this config has been changed to be enabled only if a compatible device is enabled. I don't see such a device being enabled in nucleo_f429zi but I'm able to use it as it is and get data over a socket.

The compatible device is actually 'mac'.
So that's what you're actually doing here:
```
&mac {
	status = "okay";
```

So this should enable `CONFIG_ETH_STM32_HAL=y` w/o requiring additional modification.
Can you confirm this is the case ?

vandita5, 2021-06-25:
@erwango That is correct. I believe there's something up with my debug setup. I wouldn't receive data if a debug session was runnign on vscode. Flashing the board and then closing the debug session helped the app run fine and data was available at the client.
The board also doesn't get flashed with west flash and I am being forced to use vscode right now. This is a new setup on a mac so it's possible something is missing.

erwango, 2021-06-25:
@vandita5 Thanks for the heads up.
It seems that initial point is closed and we're into something new now.
Would you mind closing the point and open a new point for clarfity (if required) ?

vandita5, 2021-06-25:
@erwango Sure thing. I'll find more info on my debug setup and create an issue if neeed. Thanks for your help with this!

vandita5, 2021-06-25:
Resolved by changing clock config to amtch the board and defining ethernet mac device in dts.

closed 2021-06-25 by vandita5