zephyrproject-rtos/zephyr #36531: msgq: support overwriting oldest element on put() or at least discard it

protobits, 2021-06-24:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to use a message queue to pass time-dependant data in a publish-subscribe framework
between threads. This would involve support for overwriting the oldest message in the
queue when a newer one arrives and the queue is full (since keeping the oldest element is less important than 
receiving the newer one).

**Describe the solution you'd like**
Since the underlying representation is a ring buffer it should be possible to simply overwrite the oldest element
by advancing the tail pointer and writing the data there. To support this usage an attribute could be introduced
to indicate this behavior w.r.t. to the default blocking one.

**Describe alternatives you've considered**
The alternative would be to do this manually by attempting to put() with K_NO_WAIT and when this fails
discard the data with a get(). However, get() currently requires supplying a data pointer. In this scenario,
it should be possible to pass NULL and have get() simply not copy the data.
This solution is sub-optimal though since it would be prone to race conditions (the queue could be filled
again by other thread right after an element was freed, or I could be removing a different element after space
was already freed in the meantime).

**Additional context**

peter-mitsis, 2021-12-16:
So I've been pondering this for a while. There are several methods available on how to do this.

1. Create a new type of kernel object to cover this use case.
2. Change the existing API *init* APIs so that they accept an *options* parameter.
3. Create a variant of the k_msgq_put() API that accepts an *options* parameter.
4. Create a new API through which its *flags* field can be modified post initialization.

Between the four options, I'm currently leaning towards options 3 and/or 4.

If option 3 is pursued, the current k_msgq_put() could simply become a wrapper for the one that accepts an *options* parameter, thereby maintaining backwards compatibility.

If option 4 is pursued, then a message queue with the desired options can not be created in one step, but would require modification post-creation.

Both options, would also introduce additional complexity as the rules for both how to behave when there are already threads pending on that message queue, as well as a decision on what impact the existing *timeout* parameter should have.

This bears more pondering.

danielstuart14, 2023-03-23:
I think something like proposed here https://github.com/zephyrproject-rtos/zephyr/issues/46339 would also be possible for this case. Adding a new method to msgq that overwrites the oldest element if full. An example: k_msgq_put_overwrite()