zephyrproject-rtos/zephyr #36543: LoRaWAN - Provide method to register a battery level callback

stoberblog, 2021-06-25:
**Is your enhancement proposal related to a problem? Please describe.**
The LoRaWAN standard allows the battery state of charge (0-100% represented by 1-255) to be sent in the MAC section of the transmission. The library used by Zephyr supports this, however the sub-system setup within Zephyr has the callback set to NULL (see $(ZEPHYR_BASE)/subsys/lorawan/lorawan.c line 480
`macCallbacks.GetBatteryLevel = NULL;
`
Ideally there should be a way of setting this and allowing it's use in applications.

**Describe the solution you'd like**
I would tend to have a __weak dummy callback in $(ZEPHYR_BASE)/subsys/lorawan/lorawan.c and allow the application writer to implement a proper version if they desire.
There may want to be a CONFIG_ to enable this as it may use extra bytes in the LoRaWAN payload.

**Describe alternatives you've considered**
Alternatively, a register callback function could be used.

**Additional context**
For the pull request writer, you may find it easier to test with chripstack rather than TTN. I've found that when in an application on ChirpStack, it clearly shows the the battery status for each device connected.

andysan, 2021-07-07:
This is already possible using the `lorawan_set_battery_level_callback` api.

kartben, 2023-12-14:
I believe this can be closed as it's been implemented /Cc @JordanYates

closed 2023-12-14 by kartben