zephyrproject-rtos/zephyr #36552: Bluetooth v2.6.0 connectable advertising leak/loss

nwsetec, 2021-06-25:
**Describe the bug**
While resolving the causes of two Bluetooth asserts described in issue: https://github.com/zephyrproject-rtos/zephyr/issues/36381 we uncovered another issue.  We have up to 4 mobile/phone devices connected and bonded to our Zephyr device which is acting as a Bluetooth peripheral.  After connecting all four mobile devices and then disconnected and reconnected each device a few times, gradually the total number of devices that would successfully re-connect would drop to one or sometimes zero.

**To Reproduce**
See config used in issue: https://github.com/zephyrproject-rtos/zephyr/issues/36381

**Expected behaviour**
Up to four central Bluetooth devices can connect to our peripheral in any sequence and with up to four at a time indefinitely.

**Impact**
Showstopper - end user will need to reset device to continue using it.

**Logs and console output**
Supplied to @cvinayak 

**Environment (please complete the following information):**
OS: Windows
Toolchain C compiler identification is GNU 9 Q4
Commit SHA c9fd7a6
+ cherry pick commits from https://github.com/zephyrproject-rtos/zephyr/pull/36515
+ EVENT_DONE_MAX to 4 recommendation from comment https://github.com/zephyrproject-rtos/zephyr/issues/36381#issuecomment-868254000

cvinayak, 2021-06-25:
@nwsetec Do you use `bt_conn_ref` and `bt_conn_unref` in your application? 

Also the logs you gave me has lot of dropped messages, you will need to increase the logging buffer size and provide me something where HCI commands issued by host is present.

joerchan, 2021-06-25:
The HCI_CORE debug log might be a bit verbose. Since we are only interested in advertising debug logs we could limit the debug log by commenting out  `#define BT_DBG_ENABLED IS_ENABLED(CONFIG_BT_DEBUG_HCI_CORE)` at the top of `subsys/bluetooth/host/adv.c`.

@jhedberg Should we create additional debug configurations for adv.c/scan.c/id.c (possibly others) after we split up hci_core.c?

cvinayak, 2021-06-25:
@nwsetec yes, if you can manage to get only the debug logs from host adv.c, that should suffice.

nwsetec, 2021-06-27:
@cvinayak no use of `bt_conn_ref` and `bt_conn_unref` in our application code.

We did recently upstream this commit into Zephyr: https://github.com/zephyrproject-rtos/zephyr/commit/c9fd7a6099c4fad73014aaf0ce364996afbba9e6 but `bt_conn_ref` and `bt_conn_unref` should be handled correctly inside of `bt_conn_foreach`.

nwsetec, 2021-06-27:
[host_adv.log](https://github.com/zephyrproject-rtos/zephyr/files/6721421/host_adv.log)
So In this log CONFIG_BT_DEBUG_LOG=n and CONFIG_BT_DEBUG_HCI_CORE=n and I modified host/adv.c like this: ` #define BT_DBG_ENABLED true//IS_ENABLED(CONFIG_BT_DEBUG_HCI_CORE)`
Mid way through the log I could only connect any 2 of the 4 mobile devices but not the full 4 that I could at the beginning.

nwsetec, 2021-06-27:
[RTT_Terminal_202106272133.zip](https://github.com/zephyrproject-rtos/zephyr/files/6721500/RTT_Terminal_202106272133.zip)
[RTT_Terminal_202106272138.zip](https://github.com/zephyrproject-rtos/zephyr/files/6721506/RTT_Terminal_202106272138.zip)
These were the two best logs I could get by the end of the each log only 1 device could connect instead of 4.

cvinayak, 2021-06-28:
> [host_adv.log](https://github.com/zephyrproject-rtos/zephyr/files/6721421/host_adv.log)
> So In this log CONFIG_BT_DEBUG_LOG=n and CONFIG_BT_DEBUG_HCI_CORE=n and I modified host/adv.c like this: ` [...]

I see `<dbg> bt_adv.bt_le_adv_resume: Host cannot resume connectable advertising (-12)`  which indicates host exhausted connection contexts.

cvinayak, 2021-06-28:
> [RTT_Terminal_202106272133.zip](https://github.com/zephyrproject-rtos/zephyr/files/6721500/RTT_Terminal_202106272133.zip) [...]

In these logs I see `<dbg> bt_hci_core.hci_cmd_done: opcode 0x200a status 0x00 buf 0x2000da70` on all attempts to use the HCI LE Advertising Enable command. There is no error from the Controller, and suspect the Host to have not attempted to resume advertising. Refer to my previous comment.

nwsetec, 2021-06-30:
Thanks @nashif

Yesterday I attempted a work around by using the `bt_le_adv_start()` API option `BT_LE_ADV_OPT_ONE_TIME` when advertising. Advertising was manually restarted in the connection and disconnection callbacks but this did not succeed as a work around and actually failed quicker.  After having 4 devices connected and at the disconnection of each device the advertising would fail to restart when calling `bt_le_adv_start()` (returning `-ENOMEM`).  After all devices were disconnected, advertising was not present and attempts to restart it failed by  calling `bt_le_adv_start()` would fail also returning `-ENOMEM`.

jhedberg, 2021-06-30:
Sounds like the host has a bt_conn reference leak (assuming that the.application isn't using gone of the host APIs that gives the application a reference and the application fails to unref it at the appropriate time, however from the description it sounds like this is not the case).

nwsetec, 2021-07-01:
That's right @jhedberg we're not using `bt_conn_ref` or `bt_conn_unref` in our application code.

Another issue we're experiencing in the project that I'd missed up until today is that if the Zephyr v2.6.0 based device is reset and an a mobile device connects (note it has previously been BLE bonded) the mobile app can read data and subscribe to notifications.  If the mobile app is closed and reopened (Bluetooth is disconnect and reconnected) then the mobile app can now read but can no longer receive notifications (even when toggling the notification subscribe/unsubscribe button in the nRFConnect app).  The the Zephyr device is reset and the mobile app reconnects again then notifications will work.

joerchan, 2021-07-01:
@nwsetec It would be best to create a separate issue for that.

nwsetec, 2021-07-01:
@joerchan ok no problem I'll create a separate issue

carlescufi, 2021-07-15:
@nwsetec just to clarify, you are not using directed advertising, right?

nwsetec, 2021-07-16:
@carlescufi no, we're not using directed advertising.

carlescufi, 2021-07-29:
@nwsetec is there a chance you can provide a small sample to reproduce this problem?

carlescufi, 2021-08-05:
@nwsetec ping

cvinayak, 2021-08-12:
@nwsetec Please verify if cherry-pick of #37561 and #37629 fixes this issue

nwsetec, 2021-08-17:
@cvinayak thanks for the updates, I'll try to find some time to test before the end of the week.

cvinayak, 2021-08-18:
@nwsetec latest upstream main now has all the fixes related to recently analyzed issues, should make it easy for you to test on the upstream main branch.

cfriedt, 2021-09-07:
@nwsetec - please reopen if the issue persists.

closed 2021-09-07 by cfriedt

nwsetec, 2021-09-11:
@cvinayak @cfriedt @carlescufi I've tested the same application code with Zephyr SHA 30db452 and the issue is no longer present, so this can stay closed.