zephyrproject-rtos/zephyr #36553: LoRaWAN Sample RX 2 Timeout

gabbas1, 2021-06-25:
Hey Guys, @Mani-Sadhasivam 

I am using the lorawan example (OTAA) and successfully can send a `join request` from the transceiver (Semtech SX1276 controlled by a nRF52840-dk). My LoRaWAN-GW receives the request and responds with a `join accept` (See pictures).

[image]
[image]

Somehow my transceiver doesn't seam to receive (or accept?!) the `join accept`:

```
2021-06-25 14:10:09,415 # *** Booting Zephyr OS build zephyr-v2.6.0-404-g13416c584ba8  ***
2021-06-25 14:10:10,388 # [00:00:00.452,209] <inf> sx127x: SX127x version 0x12 found
2021-06-25 14:10:10,395 # [00:00:00.550,445] <dbg> lorawan.lorawan_init: LoRaMAC Initialized
2021-06-25 14:10:15,488 # [00:00:05.550,537] <inf> lorawan_class_a: Joining network over OTAA
2021-06-25 14:10:15,495 # [00:00:05.560,455] <dbg> lorawan.lorawan_join: Network join request sent!
2021-06-25 14:10:23,822 # [00:00:13.882,568] <dbg> lorawan.MlmeConfirm: Received MlmeConfirm (for MlmeRequest 1)
2021-06-25 14:10:23,829 # [00:00:13.882,598] <err> lorawan: MlmeConfirm failed : Join failed
2021-06-25 14:10:23,837 # [00:00:13.883,789] <err> lorawan_class_a: lorawan_join_network failed: -22
```

Also my (Chirpstack)-LoRaWAN-GW says that it received a packet (the request) and emitted another one (i guess the join accept):

```
Jun 25 12:28:25 raspberrypi3 user.info chirpstack-concentratord-sx1301[5496]: Frame received, uplink_id: 3e3626a4-7eb5-4e8a-8f4d-51885fe24482, count_us: 25135716, freq: 868500000, bw: 125000, mod: LoRa, dr: SF12
Jun 25 12:28:25 raspberrypi3 user.info chirpstack-concentratord-sx1301[5496]: Enqueueing timestamped packet, downlink_id: e15bd81f-0034-46c9-87c2-e8bd22945e15, counter_us: 30135716, current_counter_us: 25420284
Jun 25 12:28:30 raspberrypi3 user.info chirpstack-concentratord-sx1301[5496]: Scheduled packet for TX, downlink_id: e15bd81f-0034-46c9-87c2-e8bd22945e15, count_us: 30135716, freq: 868500000, bw: 125000, mod: LoRa, dr: SF12
Jun 25 12:28:32 raspberrypi3 user.info chirpstack-concentratord-sx1301[5496]: Publishing stats event, stats_id: 7d710b38-ab7d-40b8-904a-8d3958e22f27, rx_received: 1, rx_received_ok: 1, tx_received: 1, tx_emitted: 1
```

Does anyone has an idea what could be wrong anyways?

PS: I have some other different LoRaWAN-Nodes (Raspi LoRa Node, RIOT LoRaWAN) up and running with the same setup and there are just working fine, so I dont think the gateway or Network-/Application-Server-Part is causing the problem.

# Setup:
## LoRaWAN Node
- Zephyr 2.6 (latest)
- nRF52840 Development Kit
- Semtech SX1276

## LoRaWAN GW
- Raspberry Pi
- Pi Supply LoRaWAN HAT (RAK2247)
- Chirpstack Gateway OS (Base)

## LoRaWAN Application/Network-Sevrer
- Chirpstack

GABS9718, 2021-06-30:
Hi @gabbas1 and @Mani-Sadhasivam,

I'm facing something similar. I can do the join in the network, but I dont receive the message back from the network server

[image]

I'm also using the nrf52840 and SX1276 with the zephyr 5.0. and my location uses the settings from australia 915MHz. Below is my prj.conf file

#GENERAL SETTINGS
CONFIG_CPLUSPLUS=y
CONFIG_REBOOT=y
CONFIG_NEWLIB_LIBC=y

#LOG
CONFIG_LOG=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n
CONFIG_PRINTK=y

#LORAWAN
CONFIG_SPI=y
CONFIG_SPI_NRFX=y
CONFIG_SPI_1=y
CONFIG_GPIO=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
CONFIG_LORAWAN=y
CONFIG_LORAMAC_REGION_AU915=y
CONFIG_SHELL=y
CONFIG_LORA_SHELL=y

Does someone has some tip?

Thanks

gabbas1, 2021-07-06:
Hi @GABS9718,

can you show me your configuration for the lora-transceiver on spi in the "nrf52840dk-nrf52840.dts"-file. Mine is in "../zephyr/boards/arm/nrf52840dongle_nrf52840/" and it looks like this:

```
&spi1 {
    compatible = "nordic,nrf-spi";
    status = "okay";
	sck-pin = <30>;
	mosi-pin = <31>;
	miso-pin = <29>;
	cs-gpios = <&gpio0 28 GPIO_ACTIVE_LOW >;
    lora: sx1276@0 {
		compatible = "semtech,sx1276";
		reg = <0>;
		label = "sx1276";
		reset-gpios = <&gpio0 4 GPIO_ACTIVE_LOW >;
		dio-gpios = <&gpio0 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>,
			<&gpio1 1 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
			<&gpio1 11 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
			<&gpio1 13 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
			<&gpio1 5 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >,
			<&gpio1 3 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH) >;
		rfi-enable-gpios = <&gpio1 7 GPIO_ACTIVE_HIGH >;
		rfo-enable-gpios = <&gpio1 2 GPIO_ACTIVE_HIGH >;
		pa-boost-enable-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH >;
		tcxo-power-gpios = <&gpio1 12 GPIO_ACTIVE_HIGH >;
		tcxo-power-startup-delay-ms = <5>;
		spi-max-frequency = <1000000>;
	};
};
```

Have you wired all of the DIO Pins from the SX1276 to the DK?

Thanks

gabbas1, 2021-07-15:
Hi everyone,

i managed to make it work.
On the Server-side (Chirpstack in my case) I had to change the Device-Profile of the node from LoRaWAN version 1.1.0 (latest) to 1.0.3. Now the joining process via OTAA works for me and data can be send/received.

@Mani-Sadhasivam can you say something about the implemented LoRaWAN version in Zepyhr?

tagunil, 2021-07-15:
LoRaWAN 1.1.x is not supported yet. I think we will wait for stable upstream support.

closed 2021-07-16 by gabbas1

tcpipchip, 2022-11-07:
Hi, i am getting the same issue!

Did you find it ?

flavioafferreira, 2023-05-01:
I am having the same problem. I am using nRF5340DK. On Helium Console appears Join Request and after Join Accepted. But the return of lorawan_join(&join_cfg) is -116 and sometimes -111. Should be 0. 

Join Request:
short-taupe-mole | -107 | -12.50 | 868.30 | SF12BW125 | May 1, 2023 9:07:36.895 PM

Join  Accepted:
short-taupe-mole | 16 | 0.00 | 868.30 | SF12BW125 | May 1, 2023 9:07:38.897 PM

This is my console:
[00:43:19.412,933] <27>[0m<inf> lorawan_class_a: Joining network over OTAA<27>[0m<\r><\n>
[00:43:19.416,320] <27>[0m<dbg> lorawan: lorawan_join: Network join request sent!<27>[0m<\r><\n>
[00:43:27.133,209] <27>[0m<dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)<27>[0m<\r><\n>
[00:43:27.133,239] <27>[1;31m<err> lorawan: MlmeConfirm failed : Rx 2 timeout<27>[0m<\r><\n>
[00:43:27.134,490] <27>[1;31m<err> lorawan_class_a: lorawan_join_network failed: **-116**<27>[0m<\r><\n>

It's being a little difficult because I am far from the gateway and sometimes it works and join, sometimes can't join.

this is my prj.conf

&spi4 {
    cs-gpios = <&gpio0 28 GPIO_ACTIVE_LOW>;
    lora0: sx1276@0 {
		        compatible = "semtech,sx1276";
		        reset-gpios = <&gpio1 10 GPIO_ACTIVE_LOW>;
		        dio-gpios = 
                 <&gpio1 11 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>, //DIO-0
                 <&gpio1 12 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>; //DIO-1
                 //<&gpio1 7  (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;  //DIO-2
		        power-amplifier-output = "pa-boost";
		        reg = <0>;
		        spi-max-frequency = <1000000>;
		        label = "sx1276";
          };

};

Does anyone have done any progress?

flavioafferreira, 2023-05-02:
I took a look on this example you send me and it worked with nRF5340dk and Zephyr.

Here is the full code:
https://github.com/flavioafferreira/lorawan

Thanks,

Flavio.

tcpipchip, 2023-05-02:
I am glad that I help you! Are from Brazil ?

flavioafferreira, 2023-05-02:
I am brazilian but I live in Italy, how about you?

tcpipchip, 2023-05-02:
I am Brazilian too! Live in Blumenau S.C

flavioafferreira, 2023-05-02:
Que legal, moro em Veneza, bom ter alguém para conversar. Aqui estou
conectando esse kit de desenvolvimento em algum gateway da rede Helium.
 Funcionou agora mas creio que exista uma problema secundário de reset.
Pois so' inicia a transmissão quando desconecto a energia do
módulo SX1276.

flavioafferreira, 2023-05-02:
▒[00:00:00.556,732] <inf> sx127x: SX127x version 0x12 found
*** Booting Zephyr OS build v3.2.99-ncs2-rc1 ***
[00:00:00.580,902] <dbg> lorawan: lorawan_set_region: Selected region 5
[00:00:00.636,810] <dbg> lorawan: lorawan_start: LoRaMAC Initialized
[00:00:01.144,317] <inf> lorawan_node: Joining network over OTAA
[00:00:01.154,632] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:05.176,422] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:05.185,882] <err> lorawan: MlmeConfirm failed : Tx timeout
lorawan_join_network failed: -116

Sleeping for 10s to try again to join network.

[00:00:15.205,169] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:22.983,489] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:22.992,980] <inf> lorawan: Joined network! DevAddr: 48000036
[00:00:23.001,098] <inf> lorawan_node: New Datarate: DR_0, Max Payload 51
[00:00:23.023,406] <inf> lorawan: Datarate changed: DR_0
[00:00:23.144,287] <inf> lorawan_node: Sending data...
[00:00:37.173,461] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:00:37.197,814] <err> lorawan: McpsRequest failed : Rx 2 timeout
[00:00:37.221,099] <err> lorawan_node: lorawan_send failed: -116
[00:01:00.859,527] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:01:00.882,843] <inf> lorawan_node: Port 0, Pending 0, RSSI -131dB, SNR -12dBm
[00:01:00.905,914] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:01:00.930,236] <dbg> lorawan: mcps_confirm_handler: McpsRequest success!
[00:01:00.954,040] <inf> lorawan_node: Data sent!
[00:01:25.046,813] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:01:25.071,136] <err> lorawan: McpsRequest failed : Rx 2 timeout
[00:01:25.094,390] <err> lorawan_node: lorawan_send failed: -116

from Helium's Console:
{
  "category": "uplink",
  "description": "Confirmed data up received",
  "device_id": "8d30b4b8-1bb1-4a10-932f-7a135da79243",
  "device_name": "Teste1",
  "id": "f2b3b7bb-5750-40ca-833d-440d4d03e884",
  "inserted_at": "2023-05-02T10:22:31",
  "organization_id": "371c630a-e4ac-4f65-a73c-3ed856d67b27",
  "reported_at": "1683022951844",
  "reported_at_naive": "2023-05-02T10:22:31",
  "router_uuid": "d5cc31a2-de6a-4fa6-8448-8e2b87fb488c",
  "sub_category": "uplink_confirmed",
  "updated_at": "2023-05-02T10:22:31",
  "fcnt_up": 3,
  "raw_packet": "gDYAAEiAAwAC+APziXATAYMWCouDdlk=",
  "payload": "aGVsbG93b3JsZA==",
  "payload_size": 10,
  "port": 2
}

tcpipchip, 2023-05-02:
Is the sx1276 connected to the correct gpio ?

flavioafferreira, 2023-05-02:
Yes, it is.
The data is being transmitted to Helium perfectly, but sometimes join's
command return -116(fail). When I turn off and turn on the module, works
once.
There is other thing, the gateway is 3km from here. Maybe could be the
signal be weak.

flavioafferreira, 2023-05-03:
I discovered why sometimes it work and stop. It's not a module problem or antenna. It's because the duty cycle.

[00:23:18.820,251] <err> lorawan_node: lorawan_send failed: -111
[00:23:28.842,163] <err> lorawan: LoRaWAN Send failed: Duty-cycle restricted

This specific duty cycle requirement of <1 % translates to a maximum of 36 seconds of TX
on-time within one hour and a maximum length of 3.6 seconds for any single transmission.

There is an article, they describe how it works. 
https://www.sensorsportal.com/HTML/DIGEST/december_2013/PDF_vol_161/P_1743.pdf

 If you go beyond 1% the system will give you this message:

  I adjusted the delay between lorawan_send() and now it's perfect.

tcpipchip, 2023-05-03:
yes!
I dont have problems because i installed the CHIRPSTACK