zephyrproject-rtos/zephyr #36568: net: lib: sockets: Assertion fails when zsock_close()

chyu313, 2021-06-26:
**Describe the bug**
When testing TCP stack with TCP client connecting to TCP server, assertion fails as following:
```
ASSERTION FAIL [!(mutex->owner != _kernel.cpus[0].current)] @ /zephyr/zephyrmaster/zephyr/kernel/mutex.c:211
@ /zephyr/zephyrmaster/zephyr/lib/os/assert.c:45
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2e127c (tcps_1)
E: Halting system
Exiting due to fatal error
```
Here is how it goes:
1. ```tcpclient thread``` socket connects to tcpserver socket with accepted fd=2
2. ```tcpserver thread``` calls zsock_close(fd=2)->z_impl_zsock_close()->k_mutex_lock(&fdtable[2].lock), then calls z_free_fd(2), and then context switch
3. ```tcpclient thread``` calls zsock_socket()->z_impl_zsock_socket()->zsock_socket_internal()->z_reserve_fd()->_find_fd_entry()=2 (which was freed in step 2), and then calls ```k_mutex_init(&fdtable[2].lock)```(but this mutex was locked by tcpserver thread in step 2).
4. context switch to ```tcpserver thread``` to continue execution of z_impl_zsock_close(), calls k_mutex_unlock() in which CHECKIF(mutex->owner != _current) causes assertion failure.

**To Solve**
The reason is that z_free_fd(sock) is called before k_mutex_unlock(lock) in z_impl_zsock_close(), so I move z_free_fd(sock) to be after k_mutex_unlock(lock).
```
int z_impl_zsock_close(int sock)
{
	...
	ret = vtable->fd_vtable.close(ctx);

	k_mutex_unlock(lock);

	z_free_fd(sock);

	return ret;
}
``` 

**To Reproduce**
Run the following application with native_posix.
```
#include <zephyr.h>
#include <stdio.h>
#include <net/socket.h>

#define NUM_THREAD  2
#define THREAD_STACK_SIZE 8192
#define THREAD_PRIORITY K_PRIO_COOP(8)
K_THREAD_STACK_ARRAY_DEFINE(tcpserver_stack, NUM_THREAD,
			    THREAD_STACK_SIZE);
struct k_thread tcpserver_thread[NUM_THREAD];
K_THREAD_STACK_ARRAY_DEFINE(tcpclient_stack, NUM_THREAD,
			    THREAD_STACK_SIZE);
struct k_thread tcpclient_thread[NUM_THREAD];

void tcpserver(void *ptr1, void *ptr2, void *ptr3)
{
    int tcpport = (int)ptr1;
    int listen_fd, data_fd;
    struct sockaddr_in sin;
    int n;

    listen_fd = zsock_socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    if (listen_fd <= 0) {
        return;
    }
    sin.sin_family = AF_INET;
    sin.sin_port = htons(tcpport);
    sin.sin_addr.s_addr = INADDR_ANY;
    zsock_bind(listen_fd, (const struct sockaddr *)&sin, sizeof(sin));
    zsock_listen(listen_fd, 1);
    while (1) {
        n = sizeof(struct sockaddr_in);
        data_fd = zsock_accept(listen_fd, (struct sockaddr *)&sin, &n);
        zsock_close(data_fd);
    }
}

void tcpclient(void *ptr1, void *ptr2, void *ptr3)
{
    int tcpport = (int)ptr1;
    int fd;
    struct sockaddr_in sin;

    while (1) {
        fd = zsock_socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
        if (fd <= 0) {
            return;
        }
        sin.sin_family = AF_INET;
        sin.sin_port = htons(tcpport);
        sin.sin_addr.s_addr = 0x0100007F; /* 127.0.0.1 */
        zsock_connect(fd, (struct sockaddr *)&sin, sizeof(sin));
        zsock_close(fd);
    }
}

void main(void)
{
    int i;
    char name[64];

    for (i = 0; i < NUM_THREAD; i++) {
    	k_thread_create(&tcpserver_thread[i], tcpserver_stack[i],
    	    K_THREAD_STACK_SIZEOF(tcpserver_stack[i]), tcpserver,
    	    (void *)(4001 + i), NULL, NULL, THREAD_PRIORITY, 0, K_NO_WAIT);
        snprintf(name, sizeof(name), "tcps_%d", i);
    	k_thread_name_set(&tcpserver_thread[i], name);
    	k_thread_start(&tcpserver_thread[i]);

    	k_thread_create(&tcpclient_thread[i], tcpclient_stack[i],
    	    K_THREAD_STACK_SIZEOF(tcpclient_stack[i]), tcpclient,
    	    (void *)(4001 + i), NULL, NULL, THREAD_PRIORITY, 0, K_NO_WAIT);
        snprintf(name, sizeof(name), "tcpc_%d", i);
    	k_thread_name_set(&tcpclient_thread[i], name);
    	k_thread_start(&tcpclient_thread[i]);
    }
}
```
**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit 89212a7fbf5fcd8e4d661c016344ae4bf2d46f53

cfriedt, 2021-07-09:
Thanks for finding that @chyu313. Will you also make a PR to fix?

chyu313, 2021-07-10:
> Thanks for finding that @chyu313. Will you also make a PR to fix?

Thanks for reviewing. I have made a PR.

PR #36861 (2021-07-13): net: lib: sockets: Fix assertion failure when zsock_close()
When zsock_close() is called, socket is freed before the mutex for the
socket is unlocked. If the freed socket is given to another thread
immediately, the mutex for the socket will be initialized by the new
socket owner, while the mutex is still locked by the thread calling
zosck_close().

Fixes #36568
commit a927ae39 (2021-07-13): net: lib: sockets: Fix assertion failure when zsock_close()

When zsock_close() is called, socket is freed before the mutex for the
socket is unlocked. If the freed socket is given to another thread
immediately, the mutex for the socket will be initialized by the new
socket owner, while the mutex is still locked by the thread calling
zosck_close().

Fixes #36568
closed 2021-07-13 by cfriedt