zephyrproject-rtos/zephyr #36569: '.. only:' is not working as expected in documnetaiton

nashif, 2021-06-26:
**Describe the bug**

We do have this role based content in our documentation that stopped working:

```
Zephyr Project Documentation
############################

.. only:: release

   Welcome to the Zephyr Project's documentation for version |version|.

   Documentation for the latest (main) development branch of Zephyr
   can be found at https://docs.zephyrproject.org/

.. only:: (development or daily)

   **Welcome to the Zephyr Project's documentation
   for the main tree under development** (version |version|).

```

The published documentation does not have any of the variants.

**To Reproduce**

check published docs and compare with source

**Expected behavior**

Need one variant at lease published

**Impact**

There is no way for us to create content based on the desired output, for example with

```
.. only:: html

   Use the version selection menu on the left to view
   documentation for a specific version of Zephyr.
```

nashif, 2021-06-26:
removing eager_only extension makes this work again, not sure what did break. Maybe `only` implementation in sphinx has changed and the extension is not compatible anymore? 

@pfalcon  FYI

gmarull, 2021-06-28:
Why do we need the `eager_only` extension? Does documentation build fine without?

nashif, 2021-06-28:
> Why do we need the `eager_only` extension? Does documentation build fine without?

builds fine, I think as a first step we should just remove it.

pfalcon, 2021-06-29:
> removing eager_only extension makes this work again, not sure what did break. Maybe only implementation in sphinx has changed and the extension is not compatible anymore?
> @pfalcon FYI

I wrote that extension quite some time ago for a particular project, which no longer uses it, and I didn't use it for a bunch of time either. So, I'm not sure about status of it. Of course, if needed, I can look into it. Otherwise, if there's an idea that it might not be needed any longer, that may be a good way too.

PR #36597 (2021-06-30): doc: extensions: remove eager_only
The eager_only is currently breaking the usage of the `.. only`
directive, so remove it. Documentation seems to build fine without it.

Fixes #36569
closed 2021-06-30 by nashif