zephyrproject-rtos/zephyr #36571: LoRa support for random DevNonce and NVS stack state storage

fabiobaltieri, 2021-06-26:
Hi, picking up from https://github.com/zephyrproject-rtos/zephyr/pull/35892#issuecomment-868849871, with the recent LoRaWAN stack update, the DevNonce default went from random to sequential (https://github.com/zephyrproject-rtos/loramac-node/commit/ff05e9b6a16ee9aebbcee613532e1eb2f0df6716). That is fine on production application, where the application also implements non volatile state storage, but it makes the out-of-the-box setup harder to use for prototypes and dev setup, which may always start from clear state. This is because normally DevNonce can only be used once in the network.

For example, this makes the `samples/subsys/lorawan/class_a` fairly inconvenient to run against The Things Network (and probably other LoRaWAN networks as well).

A useful feature for development would be to have a Kconfig option for going back to random DevNonce (https://github.com/zephyrproject-rtos/loramac-node/blob/zephyr/src/mac/LoRaMacCrypto.c#L967-L971), maybe using the Zephyr random number API.

For production application, it would be nice to have a subsystem level stack Nvs implementation, maybe with options of using either the settings subsystem or eeprom directly.

andysan, 2021-07-07:
Being able to persist the state of the stack is tracked in #31210. I have an implementation that uses the normal settings API that you might be able to use if you rebase it.

fabiobaltieri, 2021-07-08:
Hey that's great, I imagine every LoRa + Zephyr based project in the field had to implement that at application level, fantastic to get a common one.

However, I'm more concerned about the usability of the `lorawan/class_a` sample. With this change, out of the box the sample is only going to be able to perform OTAA at the first try once. Then it would only succeed on the second try, then third etc... and this is with few minutes between tries at SF12 (which AFAIK should not even be used without ADR).

Having NVS for the stack in the sample would not help much either, since the sample is probably used for tinkering on board that gets reflashed and erased with random stuff all the time.

The use case I have in mind is developers using that sample for prototyping or testing the radio drivers, and it would be nice to have a default configuration that is more likely to work consistently even if the board has been erased or does not support flash/eeprom, like it does with random DevNonce. Probably a good idea to add a warning for that setup, similarly to the default USB descriptors or the non secure random stuff. Right now that's not possible without modifying the stack code.

schneider42, 2021-08-17:
In its current state the sample actually only works once. After reboot it will try again with the nonce set to 0, fail to join and then stop (return from `main()`).

Mani-Sadhasivam, 2021-08-20:
> The use case I have in mind is developers using that sample for prototyping or testing the radio drivers, and it would be nice to have a default configuration that is more likely to work [...]

@fabiobaltieri Even though your point of getting the lorawan sample to work reliably sounds reasonable, I'm not happy with hacking the loramac-node for that purpose. The only way forward should be to add support for non-volatile (like eeprom) based context storage in zephyr lorawan subsystem.

Currently, I'm not getting enough bandwidth to work on this feature :/

schneider42, 2021-11-29:
Related PR: https://github.com/zephyrproject-rtos/zephyr/pull/40193

fabiobaltieri, 2022-05-26:
Hey I think we can call this done.

closed 2022-05-26 by fabiobaltieri