zephyrproject-rtos/zephyr #36572: kernel: Negtive mutex lock_count value

chyu313, 2021-06-26:
**Describe the bug**
In z_impl_k_mutex_unlock(), ```mutex->lock_count``` value can becomes ```-1``` when executing the following code piece if mutex->lock_count is zero before decrement. 
```
	/*
	 * If we are the owner and count is greater than 1, then decrement
	 * the count and return and keep current thread as the owner.
	 */
	if (mutex->lock_count - 1U != 0U) {
		mutex->lock_count--;
		goto k_mutex_unlock_return;
	}
```
This code segment should be changed to
```
	if (mutex->lock_count > 1U) {
		mutex->lock_count--;
		goto k_mutex_unlock_return;
	}
```
**To Reproduce**
Run the following application with native_posix, with options: -DCONFIG_ASSERT=n -DCONFIG_NO_RUNTIME_CHECKS=y -DCONFIG_NET_LOOPBACK=y -DCONFIG_NET_CONFIG_MY_IPV4_ADDR=\""127.0.0.1"\"

The following flow is how ``mutex->lock_count`` becomes ``-1``
1. ```tcpclient thread``` socket connects to tcpserver socket with accepted fd=2
2. ```tcpserver thread``` calls zsock_close(fd=2)->z_impl_zsock_close()->k_mutex_lock(&fdtable[2].lock), then calls z_free_fd(2), and then context switch.
3. ```tcpclient thread``` calls zsock_socket()->z_impl_zsock_socket()->zsock_socket_internal()->z_reserve_fd()->_find_fd_entry()=2 (which was freed in step 2), and then calls ```k_mutex_init(&fdtable[2].lock)```(but this mutex was locked by tcpserver thread in step 2).
4. ```tcpclient thread``` calls zsock_connect(fd=2)->z_impl_zsock_connect()->k_mutex_lock(&fdtable[2].lock), and then context switch occurs.
5. ```tcpserver thread``` continues execution of z_impl_zsock_close(), calls k_mutex_unlock(&fdtable[2].lock) (but this mutex was owned by tcpclient thread in step 4). k_mutex_unlock()  tries to find new_owner, but there is no thread waiting to lock this mutex, and finally set ```mutex->owner = NULL``` and ```mutex->lock_count = 0U```.
6. ```tcpclient thread``` continues execution of zsock_connect(fd=2), calls k_mutex_unlock(&fdtable[2].lock) with ```mutex->lock_count = 0``` which was set in step 5. And mutex->lock_count is decreased by 1 because ```(mutex->lock_count - 1U != 0U)```. As a result, ```mutex->lock_count = -1```
```
#include <zephyr.h>
#include <stdio.h>
#include <net/socket.h>

#define NUM_THREAD  2
#define THREAD_STACK_SIZE 8192
#define THREAD_PRIORITY K_PRIO_COOP(8)
K_THREAD_STACK_ARRAY_DEFINE(tcpserver_stack, NUM_THREAD,
			    THREAD_STACK_SIZE);
struct k_thread tcpserver_thread[NUM_THREAD];
K_THREAD_STACK_ARRAY_DEFINE(tcpclient_stack, NUM_THREAD,
			    THREAD_STACK_SIZE);
struct k_thread tcpclient_thread[NUM_THREAD];

void tcpserver(void *ptr1, void *ptr2, void *ptr3)
{
    int tcpport = (int)ptr1;
    int listen_fd, data_fd;
    struct sockaddr_in sin;
    int n;

    listen_fd = zsock_socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    if (listen_fd <= 0) {
        return;
    }
    sin.sin_family = AF_INET;
    sin.sin_port = htons(tcpport);
    sin.sin_addr.s_addr = INADDR_ANY;
    zsock_bind(listen_fd, (const struct sockaddr *)&sin, sizeof(sin));
    zsock_listen(listen_fd, 1);
    while (1) {
        n = sizeof(struct sockaddr_in);
        data_fd = zsock_accept(listen_fd, (struct sockaddr *)&sin, &n);
        zsock_close(data_fd);
    }
}

void tcpclient(void *ptr1, void *ptr2, void *ptr3)
{
    int tcpport = (int)ptr1;
    int fd;
    struct sockaddr_in sin;

    while (1) {
        fd = zsock_socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
        if (fd <= 0) {
            return;
        }
        sin.sin_family = AF_INET;
        sin.sin_port = htons(tcpport);
        sin.sin_addr.s_addr = 0x0100007F; /* 127.0.0.1 */
        zsock_connect(fd, (struct sockaddr *)&sin, sizeof(sin));
        zsock_close(fd);
    }
}

void main(void)
{
    int i;
    char name[64];

    for (i = 0; i < NUM_THREAD; i++) {
    	k_thread_create(&tcpserver_thread[i], tcpserver_stack[i],
    	    K_THREAD_STACK_SIZEOF(tcpserver_stack[i]), tcpserver,
    	    (void *)(4001 + i), NULL, NULL, THREAD_PRIORITY, 0, K_NO_WAIT);
        snprintf(name, sizeof(name), "tcps_%d", i);
    	k_thread_name_set(&tcpserver_thread[i], name);
    	k_thread_start(&tcpserver_thread[i]);

    	k_thread_create(&tcpclient_thread[i], tcpclient_stack[i],
    	    K_THREAD_STACK_SIZEOF(tcpclient_stack[i]), tcpclient,
    	    (void *)(4001 + i), NULL, NULL, THREAD_PRIORITY, 0, K_NO_WAIT);
        snprintf(name, sizeof(name), "tcpc_%d", i);
    	k_thread_name_set(&tcpclient_thread[i], name);
    	k_thread_start(&tcpclient_thread[i]);
    }
}
```

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit 89212a7fbf5fcd8e4d661c016344ae4bf2d46f53

andyross, 2021-06-30:
Analysis looks correct to me, though it seems like the exerciser is needlessly complicated.  How about "If you try to unlock an unlocked mutex, it will incorrectly succeed and decrement the count to -1" as a description instead?

Care to submit the patch?  You already have the code above.

chyu313, 2021-06-30:
> Care to submit the patch? You already have the code above.

OK, I will submit a pull request later.

PR #36634 (2021-07-06): kernel: Fix negative mutex lock_count value
If you try to unlock an unlocked mutex, it will incorrectly
succeeds and decreases the lock count to -1.

Fixes #36572
commit 0ef77d4e (2021-07-06): kernel: Fix negative mutex lock_count value

If you try to unlock an unlocked mutex, it will incorrectly
succeeds and decreases the lock count to -1.

Fixes #36572
closed 2021-07-06 by nashif