zephyrproject-rtos/zephyr #36575: Modbus RTU Client on ESP32

maweicoltsmart, 2021-06-27:
**Describe the bug**
Make the following modifications to the esp32.dts file:
```
&uart2 {
	status = "okay";
	current-speed = <115200>;
	tx-pin = <17>;
	rx-pin = <16>;
	/* rts-pin = <11>; */
	/* cts-pin = <6>; */
	/* hw-flow-control; */
	modbus0 {
		compatible = "zephyr,modbus-serial";
		label = "MODBUS0";
		status = "okay";
		de-gpios = <&gpio0 4 GPIO_ACTIVE_LOW>;	/* D9 */
	};
};
```
make and flash：west build -b esp32 samples\subsys\modbus\rtu_client

After the program runs, it has been in the callback function of sending interrupt.
File: modbus_serial.c
Function: uart_cb_handler
Endless loop: 

```
if (uart_irq_tx_ready(cfg->dev)) {
cb_handler_tx(ctx);
}

```

sylvioalves, 2021-06-29:
@maweicoltsmart Can you inform which esp32 board you have been working with?

closed 2021-09-12 by github-actions[bot]

ramb0t, 2022-01-01:
I'm still getting to the bottom of this issue, but I have found there are some weaknesses in both the ESP32 uart driver implementation (#41526) and in the general modbus subsystem implementation (see the fix here #40578) 
For anyone with the same problem please see my commits in this branch https://github.com/ramb0t/zephyr/tree/esp32_modbus_driver_fix
This is currently working for my ESP32 modbus system, but I think either the ESP32 uart driver or the modbus_serial uart_cb_handler() while loop needs some improvement to prevent infinite loops