zephyrproject-rtos/zephyr #36590: NVS sector size above 65535 not supported

lord-potato, 2021-06-28:
I'm running the NVS sample with an SPI NOR flash (Cypress S25FL512S), and the flash device returns 65536 as page size. The NVS api requires the sector size to be a multiple of page size, but the NVS sector size is stored in uint16_t. I could get it to work by changing the nvs_fs struct, so that the sector_size member is uint32_t.

Without the change, NVS sector size would overflow, and eventually lead to a sector size of 0.

The modified part of nvs.h:
```c
struct nvs_fs {
	off_t offset;		/* filesystem offset in flash */
	uint32_t ate_wra;		/* next alloc table entry write address */
	uint32_t data_wra;		/* next data write address */
	uint32_t sector_size;	/* filesystem is divided into sectors,
				 * sector size should be multiple of pagesize
				 */
	uint16_t sector_count;	/* amount of sectors in the filesystem */
	bool ready;		/* is the filesystem initialized ? */

	struct k_mutex nvs_lock;
	const struct device *flash_device;
	const struct flash_parameters *flash_parameters;
};
```

Laczen, 2021-06-28:
Hi @lord-potato, thank you for reporting this. A sector size of 2^16 is indeed not supported by nvs, increasing the sector-size to a uint32_t is however not a good idea.

On flash the adres of a data item is limited to a uint16_t, supporting a uint32_t would increase the storage requirements. Your case is exactly on the border that could be supported, but it would require a different approach: instead of a sector_size we could use a max sector_index (this would fit in uint16_t). Then we could replace the sector_size where it is used by sector_index + 1 (after casting to a uint32_t).

With so big a flash device probably a real fs (like little_fs) might be more appropiate. NVS is really meant for small devices.

lord-potato, 2021-06-29:
> instead of a sector_size we could use a max sector_index (this would fit in uint16_t). Then we could replace the sector_size where it is used by sector_index + 1 (after casting to a uint32_t).
That would indeed solve this exact problem, but I think you might be right saying this big of a flash deserves a proper file system.

I have potentially found another problem, since according to the [datasheet](https://www.cypress.com/file/177971/download), the flash should have an erase sector size of 256KB, but flash info shows 64KB sectors.

Laczen, 2021-06-29:
@lord-potato, there seems to be different versions, some of them have a erase sector of 64kB. But there seems to be some inconsistency in the datasheet (talking about FL-S with 128Mb/256Mb).

If you really intent to use nvs a quick fix would be to make sector-size uint32_t and then to limit it in init() to 2^16.

lord-potato, 2021-06-29:
Yeah, the datasheet is a bit confusing. I think I'm going to go forth using a proper file system, and leave NVS alone until I'm working with a much smaller flash.

Thanks for your time!

closed 2021-06-29 by lord-potato