zephyrproject-rtos/zephyr #36593: Failing IPv6 Ready compliance (RFC 2460)

vaussard, 2021-06-28:
**Describe the bug**
Our product is required to comply with a subset of the IPv6 Ready program (only part of Section 1 for RFC 2460) due to some regulatory constraints. I saw some old closed tickets related to that topic (for example #2333 and #3316), and I know that some compliance testing was performed in the past using Maxwell Pro, but could not find any meaningful public information regarding the current state of the IPv6 stack.

To move forward, we prepared a certification firmware based on Zephyr 2.5.0 and submitted it to the certification lab. Unfortunately results were not great: we failed 16 of the 29 test cases.

**Next Steps**
I will look into addressing the failing cases. To that end, I have a couple of questions regarding that topic:

- Did I missed any public source of information regarding the current conformance state of the IPv6 stack?
- I am a bit surprised by the high number of failed cases and wonder if this could be due to some specifics of our certification firmware or the testing setup. But looking at #3316 they reported that only 13 of the 54 tests passed (in 2017). Are these number something to be expected?
- Is someone else also working on the IPv6 conformance topic ? If yes, it could be beneficial to collaborate together in order to speed up the progress, especially to setup the test environement.
- The old tickets refer to https://wiki.zephyrproject.org/view/Zephyr_IPv6_testing_with_TAHI which is no longer available. I could not find any snapshot in the Wayback Machine. Does someone have a way to retrieve its content, since I assume that it could be of interest for my own testing ?

I will look into setting up TAHI to replicate the tests done by the laboratory.

Any information / help / feedback would be appreciated.

jukkar, 2021-06-29:
The TAHI excercice was long time ago, and if I recall correctly, there was lot of work to make it work properly and it was abandoned eventually. Then we tried the Maxwell Pro and got better progress with it. Last Maxwell Pro tests were run something like 2 years go so my memory might be wrong here but there ware some tests that we were not able to run successfully because of the test setup etc. But overall most of the Maxwell Pro conformance tests were passing ok. Note that Maxwell Pro is checking many IP based protocols, and I am not sure how comprehensive their IPv6 test suite is.

jukkar, 2021-06-29:
BTW, for any certification work etc, I would always recommend latest release, in this case 2.6. Bugs etc are fixed in newer releases and you hopefully would be able to avoid extra work backporting patches etc.

vaussard, 2021-06-29:
Thank you for your feedback @jukkar 

I went through the test cases and the wireshark captures that we received from the testing facility. I did a quick run-down of the 16 failures as follow:

- 4 test cases expected a ICMPv6 Parameter Problem as answer to a bogus IPv6 packet, but none sent by the target
- 8 test cases expected a ICMPv6 Parameter Problem as answer to a bogus IPv6 packet, and a ICMPv6 Parameter Problem was sent by the target but with incorrect `pointer` field
- 4 test cases expected a ICMPv6 Reply, but got no answer (including 3 cases related to fragmentation, which might indicate a configuration issue in my firmware)

There looks to be an issue with the computation of the `pointer` field since it was never correct, or at least not the one expected by the test suite. I will probably look into that first, since this could solve 50% of the failures.

Regarding 2.5 versus 2.6, I based the firmware on the latest version that was available at that time (which was one month after the release of 2.5). I did a quick diff of the IPv6 stack before posting my question, but found no meaningful fix for my failures. Of course I will rebase my firmware on top of the latest release.

jukkar, 2021-06-29:
> I did a quick run-down of the 16 failures as follow:
> 
>     * 4 test cases expected a ICMPv6 Parameter Problem as answer to a bogus IPv6 packet, but none sent by the target

Some of the bogus packets are dropped quite early without any icmpv6 reply. These should be quite trivial to fix.

> * 8 test cases expected a ICMPv6 Parameter Problem as answer to a bogus IPv6 packet, and a ICMPv6 Parameter Problem was sent by the target but with incorrect `pointer` field
>
>     * 4 test cases [...]

Currently IPv6 fragmentation is off by default, just set `CONFIG_NET_IPV6_FRAGMENT` to enable it.

vaussard, 2021-06-29:
> Some of the bogus packets are dropped quite early without any icmpv6 reply. These should be quite trivial to fix.

Thank you for the pointers.

I was able to quickly reproduce the first failing case (multiple hop-by-hop options) using `samples/net/eth_native_posix/` and a bit of Python script (since running TAHI is another story). This case is handled in `net_ipv6_input()`, but replacing the `goto drop;` by `goto bad_hdr;` does not seem sufficient. But this could be due to my hurried testing, I will investigate a bit later.

> Currently IPv6 fragmentation is off by default, just set CONFIG_NET_IPV6_FRAGMENT to enable it.

It was already enabled in the firmware using
```
CONFIG_NET_IPV6_FRAGMENT=y
CONFIG_NET_BUF_RX_COUNT=100
```

But I see that:
```
$ cat  zephyr/.config | grep FRAG
CONFIG_NET_IPV6_FRAGMENT=y
CONFIG_NET_IPV6_FRAGMENT_MAX_COUNT=1
CONFIG_NET_IPV6_FRAGMENT_TIMEOUT=5
```
I am not sure to understand `CONFIG_NET_IPV6_FRAGMENT_MAX_COUNT=1`, will it work if the packet has 3 fragments like in the test case ? And isn't the timeout supposed to be 60 seconds IIRC?

jukkar, 2021-06-30:
> I am not sure to understand `CONFIG_NET_IPV6_FRAGMENT_MAX_COUNT=1`, will it work if the packet has 3 fragments like in the test case ? And isn't the timeout supposed to be 60 seconds IIRC?

The count tells how many pending packets, waiting reassembly, can exists at one time. The amount of individual fragments is only determined how much you have allocated in the `CONFIG_NET_BUF_RX_COUNT`. The stack will not release the received net_bufs (fragments) until it is able to do the reassembly of the whole IPv6 packet or a timeout happens. Typically Zephyr based devices do not have much memory, so the timeout by default is only 5 sec even if RFC says otherwise. If you have lot of memory, then you can increase the values considerably.

vaussard, 2021-07-01:
Thank you for the explanations.
So apart the timeout, I think that I configured correctly the firmware. I will test using the IPv6Ready packets to see what is going on.

vaussard, 2021-07-02:
I did some good progress on the fragmentation issues, I will probably send a PR next week to address the small issues that I uncovered in the stack.

I had an issue with `net_icmpv6_send_error()` where I am not sure about the fix, @jukkar can probably help me since you wrote this piece of code (see commit 899228bb7f8e6a3925f49594185f796942e1e26c). This function was not sending the ICMPv6 error message in some cases. I discovered that it was erroring out after the first call to `net_pkt_write()`. To me, this makes sense, because we are trying to write in overwrite mode in a packet that was just allocated, thus without any valid data. So I did a stupid fix by removing the overwrite mode:

```
--- a/subsys/net/ip/icmpv6.c
+++ b/subsys/net/ip/icmpv6.c
@@ -254,7 +254,7 @@ int net_icmpv6_send_error(struct net_pkt *orig, uint8_t type, uint8_t code,
         * extra space for link addresses so that we can set the lladdr
         * pointers in net_pkt.
         */
-       net_pkt_set_overwrite(pkt, true);
 
        ret = net_pkt_write(pkt, net_pkt_lladdr_src(orig)->addr,
                            net_pkt_lladdr_src(orig)->len);
@@ -281,8 +281,8 @@ int net_icmpv6_send_error(struct net_pkt *orig, uint8_t type, uint8_t code,
        net_pkt_lladdr_src(pkt)->len = net_pkt_lladdr_dst(orig)->len;
        net_pkt_lladdr_dst(pkt)->len = net_pkt_lladdr_src(orig)->len;
 
-       net_pkt_set_overwrite(pkt, false);
-       net_pkt_cursor_init(pkt);
```

and now I am getting the expected ICMPv6 error message in all the cases that I tested. But I am probably missing something, since I do not understand why this overwrite was set. And apparently it worked before in some other cases, since we got some ICMPv6 error messages during the certification attempt (but not all).

jukkar, 2021-07-05:
There was a reason for that overwrite call, as described in the commit message of 899228bb7f8e6a3925f49594185f796942e1e26c, related PR is also here #29398. So it looks like there is still some things missing in the error function if things do not work properly.

jukkar, 2021-07-05:
Note that the fix introduced by commit 899228bb7f8e6a3925f49594185f796942e1e26c is a bit hackish and perhaps it is causing issues still. The commit basically allocates some extra space at the beginning of net_buf in order to store link layer addresses there so that we can point link layer pointer values in net_pkt there.

vaussard, 2021-07-09:
I read again the commit message of 899228b but I do not see where the overwrite mode is mentioned. It mentions that the original packet could be overwritten by fresh data (once unrefed), but says nothing about the newly allocated packet:
> When we are sending ICMPv6 error message, we need to store the
link local addresses of the received packet somewhere in order
to know where to send the new error message.
Easiest is to store the ll addresses is to the error message
itself, just before where the sent packet will start in memory.
We cannot use the original pkt to store the ll addresses
as that packet might get overwritten if we receive lot of packets.

I do not see how a packet that was just allocated using `net_pkt_alloc_with_buffer()` could be used in overwriting mode, since it is empty (unless `net_pkt_alloc_with_buffer()` writes some data in the buffers?).

I opened a PR to fix these issues, including the proposal to remove the overwrite mode from `net_icmpv6_send_error()`. Let's move the discussion there.

closed 2021-09-23 by cfriedt