zephyrproject-rtos/zephyr #36598: Lora driver TX done wait/synchronous call

fcgdam, 2021-06-28:
**Description of the problem:**

Hi: This problem/issue is related to the Lora radio driver, not Lorawan support. 

So, one issue with the sx127x radios is they can not be programmed while a RX or TX operation is in progress. While for RX the issue does not exist since the driver thread will block until the RXDone callback is called, the same is not true for TX.  This means that sending data returns immediately to the caller. If the caller sends again more data when the TX is still in-progress, the current TX will stop and a new one will start with the new data, effectively not sending the previous data to the air. (the receiver will not receive the complete packet).

This implies two things, either the TX API call also blocks like the RX API call, or there is the need to somehow to detect that the radio is in use/not finished the previous call/TX, something like bool isRadioBusy();

The TX callback is already in place, and I can confirm that (at least with my hardware) it works, all is missing is the supporting code for a blocking API call, much like the RX API call.

A major difference on the existing code is that the RX API call requires a timeout value for waiting for the RX, and will release the thread either when RX is done or the timeout is reached.  

Still doesn't make much sense to pass a timeout as a parameter to the send API call, but internally, if mimicking the RX code, a timeout value is necessary. Such timeout value can be calculated based on the data payload length plus the spread factor and bandwidth to calculate the air-time (according to Semtech formulas) and add some safety factor to it, or just plain define the timeout as the maximum time the largest possible payload will take at SF12/BW125.

**Possible solutions:**
1st) Add a api call to check if the radio is busy. -> Not ideal, but used on some other systems/APIs
2nd) Add a blocking send API call -> Ideal

**Final notes: Sample code for send blocking API call**
File: ...//zephyr/drivers/lora/sx12xx_common.c

```
static struct sx12xx_data {
    struct k_sem RX_sem;
    struct k_sem TX_sem;
    RadioEvents_t events;
    uint8_t *rx_buf;
    uint8_t rx_len;
    int8_t snr;
    int16_t rssi;
} dev_data;
.....
....
static void sx12xx_ev_tx_done(void)
{
    Radio.Sleep();

    k_sem_give(&dev_data.TX_sem);
    LOG_INF("Lora Radio: TX DONE!");
}

int sx12xx_lora_send(const struct device *dev, uint8_t *data,
             uint32_t data_len)
{
    int ret;

    Radio.SetMaxPayloadLength(MODEM_LORA, data_len);
    LOG_INF("Sending data...");
    Radio.Send(data, data_len);

    ret = k_sem_take(&dev_data.TX_sem, K_MSEC(15000) );   // Timeout sample value. To be calculated...
    if ( ret < 0 ) {
        LOG_ERR("Radio TX Done not triggered.");
        k_sem_give(&dev_data.TX_sem);
        return ret;
    }
    return 0;
}
....
....
```

aurel32, 2021-06-29:
You probably want to have a look at this PR, which solves that: https://github.com/zephyrproject-rtos/zephyr/pull/36577

fcgdam, 2021-06-30:
> You probably want to have a look at this PR, which solves that: #36577

Thanks, missed that on my search, but basically #36557 is spot on.

closed 2021-07-09 by fcgdam