zephyrproject-rtos/zephyr #36600: Bluetooth: Peripheral: Bond issue when using secure connection

XavierChapron, 2021-06-28:
**Describe the bug**
When using `peripheral_sc_only` sample on `nrf52832_mdk`, the first connection from Android `nRF Connect` application work, but the second always fails. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash `peripheral_sc_only` sample on `nrf52832_mdk`
2. Connect with `nRF Connect` application
3. Retrieve all characteristics using `Read characteristics` button
4. Disconnect
5. Reconnect to the device
6. Try to retrieve all characteristics again
4. See error

**Expected behavior**
Re connection should work and allow to retrieve device characteristics.

**Impact**
This really affect the secure BLE connection UX. the feature can still be used, but bonding information must be manually deleted before each re-connection.

**Logs and console output**
Device logs:
```
*** Booting Zephyr OS build zephyr-v2.6.0  ***
[00:00:00.398,193] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
[00:00:00.398,193] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
[00:00:00.398,223] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 0[0m
[00:00:00.398,864] [0m<inf> bt_hci_core: Identity: F2:06:B2:D5:EB:EE (random)[0m
[00:00:00.398,864] [0m<inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1[0m
[00:00:00.398,864] [0m<inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff[0m
Bluetooth initialized
Advertising successfully started
Connected 6C:16:3C:EF:27:FC (random)
Passkey for 6C:16:3C:EF:27:FC (random): 560967
Security changed: 6C:16:3C:EF:27:FC (random) level 4
Identity resolved 6C:16:3C:EF:27:FC (random) -> 5C:51:81:E8:2B:DC (public)
Pairing Complete
Disconnected from 5C:51:81:E8:2B:DC (public) (reason 0x13)
Connected 5C:51:81:E8:2B:DC (public)
Security failed: 5C:51:81:E8:2B:DC (public) level 1 err 8
Pairing Failed (8). Disconnecting.
Disconnected from 5C:51:81:E8:2B:DC (public) (reason 0x3d)
```

nRF Connect logs:
```
nRF Connect, 2021-06-28
SC only peripheral (F2:06:B2:D5:EB:EE)
D 17:03:42.003 gatt.close()
D 17:03:42.006 wait(200)
V 17:03:42.208 Connecting to F2:06:B2:D5:EB:EE...
D 17:03:42.209 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D 17:03:42.539 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D 17:03:42.557 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 17:03:42.558 Connected to F2:06:B2:D5:EB:EE
D 17:03:42.623 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D 17:03:42.623 [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3)
D 17:03:50.881 [Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_PIN (0)
D 17:04:04.644 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I 17:04:04.644 Device bonded
D 17:04:04.702 wait(1600ms)
V 17:04:06.304 Discovering services...
D 17:04:06.304 gatt.discoverServices()
D 17:04:06.323 [Callback] Services discovered with status: 0
I 17:04:06.323 Services discovered
V 17:04:06.353 Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
D 17:04:06.354 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
V 17:04:11.614 Reading all characteristics...
V 17:04:11.614 Reading characteristic 00002b29-0000-1000-8000-00805f9b34fb
D 17:04:11.614 gatt.readCharacteristic(00002b29-0000-1000-8000-00805f9b34fb)
I 17:04:11.697 Read Response received from 00002b29-0000-1000-8000-00805f9b34fb, value: (0x) 00
A 17:04:11.697 "No features" received
V 17:04:11.713 Reading characteristic 00002b2a-0000-1000-8000-00805f9b34fb
D 17:04:11.713 gatt.readCharacteristic(00002b2a-0000-1000-8000-00805f9b34fb)
I 17:04:11.773 Read Response received from 00002b2a-0000-1000-8000-00805f9b34fb, value: (0x) FB-77-E2-2D-82-8A-4B-4D-45-C6-C0-32-1E-CF-57-AF
A 17:04:11.773 "(0x) FB-77-E2-2D-82-8A-4B-4D-45-C6-C0-32-1E-CF-57-AF" received
V 17:04:11.787 Reading characteristic 00002a00-0000-1000-8000-00805f9b34fb
D 17:04:11.787 gatt.readCharacteristic(00002a00-0000-1000-8000-00805f9b34fb)
I 17:04:11.887 Read Response received from 00002a00-0000-1000-8000-00805f9b34fb, value: (0x) 53-43-20-6F-6E-6C-79-20-70-65-72-69-70-68-65-72-61-6C, "SC only peripheral"
A 17:04:11.887 "SC only peripheral" received
V 17:04:11.905 Reading characteristic 00002a01-0000-1000-8000-00805f9b34fb
D 17:04:11.905 gatt.readCharacteristic(00002a01-0000-1000-8000-00805f9b34fb)
I 17:04:11.959 Read Response received from 00002a01-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
A 17:04:11.959 "[0] Unknown" received
V 17:04:11.971 Reading characteristic 00002a04-0000-1000-8000-00805f9b34fb
D 17:04:11.971 gatt.readCharacteristic(00002a04-0000-1000-8000-00805f9b34fb)
I 17:04:12.040 Read Response received from 00002a04-0000-1000-8000-00805f9b34fb, value: (0x) 18-00-28-00-00-00-2A-00
A 17:04:12.040 "Connection Interval: 30.00ms - 50.00ms,
Slave Latency: 0,
Supervision Timeout Multiplier: 42" received
V 17:04:12.060 5 characteristics read
V 17:04:15.906 Disconnecting...
D 17:04:15.906 gatt.disconnect()
D 17:04:15.964 [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I 17:04:15.964 Disconnected
D 17:04:16.964 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D 17:04:26.273 gatt.close()
D 17:04:26.276 wait(200)
V 17:04:26.478 Connecting to F2:06:B2:D5:EB:EE...
D 17:04:26.478 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D 17:04:26.842 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D 17:04:26.905 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 17:04:26.905 Connected to F2:06:B2:D5:EB:EE
D 17:04:26.932 wait(1600ms)
V 17:04:28.534 Discovering services...
D 17:04:28.534 gatt.discoverServices()
D 17:04:28.566 [Callback] Services discovered with status: 0
I 17:04:28.566 Services discovered
V 17:04:28.590 Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
D 17:04:28.590 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
V 17:04:36.008 Reading all characteristics...
V 17:04:36.008 Reading characteristic 00002b29-0000-1000-8000-00805f9b34fb
D 17:04:36.008 gatt.readCharacteristic(00002b29-0000-1000-8000-00805f9b34fb)
D 17:04:56.841 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I 17:04:56.841 Bond information deleted
V 17:04:56.891 Reading characteristic 00002b2a-0000-1000-8000-00805f9b34fb
D 17:04:56.891 gatt.readCharacteristic(00002b2a-0000-1000-8000-00805f9b34fb)
E 17:04:56.891 Exception occurred (Reading characteristic failed)
V 17:04:56.892 Reading characteristic 00002a00-0000-1000-8000-00805f9b34fb
D 17:04:56.892 gatt.readCharacteristic(00002a00-0000-1000-8000-00805f9b34fb)
E 17:04:56.892 Exception occurred (Reading characteristic failed)
V 17:04:56.892 Reading characteristic 00002a01-0000-1000-8000-00805f9b34fb
D 17:04:56.892 gatt.readCharacteristic(00002a01-0000-1000-8000-00805f9b34fb)
E 17:04:56.892 Exception occurred (Reading characteristic failed)
V 17:04:56.892 Reading characteristic 00002a04-0000-1000-8000-00805f9b34fb
D 17:04:56.892 gatt.readCharacteristic(00002a04-0000-1000-8000-00805f9b34fb)
E 17:04:56.892 Exception occurred (Reading characteristic failed)
V 17:04:56.892 5 characteristics read
E 17:04:58.955 Error 133 (0x85): GATT ERROR
D 17:04:58.956 [Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E 17:04:58.956 Error 8 (0x8): GATT CONN TIMEOUT
I 17:04:58.956 Disconnected
D 17:04:59.016 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

**Environment (please complete the following information):**
 - Zephyr v2.6.0 (also reproduced on v2.4.0 and v2.5.0)
 - nRF Connect version 4.23.3
 - Android version 7.1.2

XavierChapron, 2021-06-28:
I've also try to disable bonding by setting `CONFIG_BT_BONDABLE=n` but this doesn't change anything.

XavierChapron, 2021-07-08:
Hi @joerchan 
Were you able to have a first look at this issue?
Does it fells like a mis-usage from my side, maybe a badly configured Kconfig ? Or do you think that's a real bug in Zephyr implementation ?
Thanks for your help.

carlescufi, 2021-07-29:
@XavierChapron, @joerchan is currently on holiday. We will update as soon as he's back.

joerchan, 2021-08-05:
@XavierChapron The disconnect reason is MIC failure, which means that the two devices are not using the same LTK. Can you please check with another phone (or peer). It could be a problem with your phone.
Also try to go through you existing bonds in the android bluetooth settings and delete possible old bonds for the device.

XavierChapron, 2021-08-05:
> @XavierChapron The disconnect reason is MIC failure, which means that the two devices are not using the same LTK. Can you please check with another phone (or peer). It could be a problem with your [...]

@joerchan Thanks for your answer.
I've already tested with one other peer (a tablet which I currently don't have access at the moment) and to delete all existing bonds, however this doesn't change anything.
Is there anything else I can test to help understand my issue?

joerchan, 2021-08-05:
@XavierChapron Ok, I've tried to follow your steps to reproduce it but I've not seen the same thing you are seeing.

```
*** Booting Zephyr OS build zephyr-v2.6.0  ***
Bluetooth initialized
Advertising successfully started
[00:00:00.367,156] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.367,156] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.367,187] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 0
[00:00:00.367,858] <inf> bt_hci_core: Identity: E2:94:C2:37:BF:F3 (random)
[00:00:00.367,858] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.367,889] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
Connected 5D:26:A8:94:94:E5 (random)
Passkey for 5D:26:A8:94:94:E5 (random): 104033
Security changed: 5D:26:A8:94:94:E5 (random) level 4
Identity resolved 5D:26:A8:94:94:E5 (random) -> 00:BE:3B:48:F8:72 (public)
Pairing Complete
Disconnected from 00:BE:3B:48:F8:72 (public) (reason 0x13)
Connected 00:BE:3B:48:F8:72 (public)
Security changed: 00:BE:3B:48:F8:72 (public) level 4
Disconnected from 00:BE:3B:48:F8:72 (public) (reason 0x08
```

We have seen issues with peripheral sending security requests to android. Can you tell me if you get the same issue by delaying the call to bt_conn_set_security in main.c. Use k_delayed_work_submit and something like 10 seconds.
Also remove the call to bt_conn_set_security from bt_gatt_connected in gatt.c

I can't see where it goes wrong from the logs, can you please enable CONFIG_BT_DEBUG_SMP=y and see if there for some strange reason ends up with different. Maybe also add logging of the LTK used in le_ltk_reply in hci_core.c.

If you have a spare nrf devkit could you set up the nRF Sniffer for Bluetooth LE and get a capture of the traffic.
In order to decrypt the traffic when encrypted we need to use the debug keys so enable CONFIG_BT_USE_DEBUG_KEYS=y and CONFIG_BT_STORE_DEBUG_KEYS=y, this could potentially alter the behavior of the android device but worth a try.

I used the nrf52840dk_nrf52840 board, you are saying nrf52832_mdk, which is not any board I'm familiar with. Is this something you have added locally?

XavierChapron, 2021-08-06:
@joerchan Thanks for your help.

> I used the nrf52840dk_nrf52840 board, you are saying nrf52832_mdk, which is not any board I'm familiar with. Is this something you have added locally?
It's a simple nrf52832 dev-kit : https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52832_mdk/doc/index.rst

I made new tests with the same board and the same phone but using as you recommend a little delay between connection and the call to `bt_conn_set_security()` using a delayed k_work..
With a delay of 3 seconds it seem to work as expected:
```
*** Booting Zephyr OS build zephyr-v2.6.0-1-gf7448aeb66b7  ***
[00:00:00.542,083] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.542,083] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.542,114] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 0
[00:00:00.542,755] <inf> bt_hci_core: Identity: F2:06:B2:D5:EB:EE (random)
[00:00:00.542,755] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.542,755] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
[00:00:00.542,785] <inf> main: Bluetooth initialized
[00:00:00.543,243] <inf> main: Advertising successfully started
[00:00:13.106,597] <inf> main: Connected 4C:73:60:06:8B:30 (random)
[00:00:16.106,628] <inf> main: delayed work executed
[00:00:18.066,162] <inf> main: Passkey for 4C:73:60:06:8B:30 (random): 784951
[00:00:29.930,938] <inf> main: Security changed: 4C:73:60:06:8B:30 (random) level 4
[00:00:29.983,612] <inf> main: Identity resolved 4C:73:60:06:8B:30 (random) -> 5C:51:81:E8:2B:DC (public)
[00:00:29.983,917] <inf> main: Pairing Complete
[00:00:58.324,829] <inf> main: Disconnected from 5C:51:81:E8:2B:DC (public) (reason 0x13)
[00:01:06.312,805] <inf> main: Connected 5C:51:81:E8:2B:DC (public)
[00:01:06.665,130] <inf> main: Security changed: 5C:51:81:E8:2B:DC (public) level 4
[00:01:09.312,866] <inf> main: delayed work executed
[00:01:36.370,178] <inf> main: Disconnected from 5C:51:81:E8:2B:DC (public) (reason 0x13)
[00:01:44.576,202] <inf> main: Connected 5C:51:81:E8:2B:DC (public)
[00:01:44.965,423] <inf> main: Security changed: 5C:51:81:E8:2B:DC (public) level 4
[00:01:47.576,232] <inf> main: delayed work executed
[00:02:10.358,154] <inf> main: Disconnected from 5C:51:81:E8:2B:DC (public) (reason 0x13) 
```
Note that the first time, the delayed work triggers a security level change, the other time, the stack handles it itself with a little delay of ~500ms and the k_work probably do nothing.

This seems like a solution for me.
However I've no idea if this solution is hacky or just the expected implementation?

I also wonder what are the implication in regard to security?
Does that means that there is a small moment after connection where characteristics can be read without knowing the device pin?

Let me know if y should run the additional tests with CONFIG_BT_DEBUG_SMP or something else.

joerchan, 2021-08-06:
> This seems like a solution for me.
However I've no idea if this solution is hacky or just the expected implementation?

So this is about who initiates the security, and it appears that the phones expect it to be them. The apple developer guide states this:
"The accessory should not request pairing until an ATT request is rejected using the Insufficient
Authentication error code. See the Bluetooth 4.0 specification, Volume 3, Part F, Section 4 for details"

So what you have run into is a known issue  where android phones when receiving a security request ends up deleting the Bond.

> I also wonder what are the implication in regard to security?
Does that means that there is a small moment after connection where characteristics can be read without knowing the device pin?

No, anything requiring security will not be accessible until the security requirement has been met.
But you have to state the security requirement for each characteristic individually.
The peripheral_sc_only sample does not add any characteristics with security requirement, have a look at the peripheral sample instead for some examples of this.

joerchan, 2021-08-06:
This is not an issue with zephyr, but with the android Bluetooth LE stack not handling security requests.
Workaround for the application is:
- Set `CONFIG_BT_GATT_AUTO_SEC_REQ=n`.
- Delay or don't initiate security request (calling bt_conn_set_security in the peripheral role).

closed 2021-08-06 by joerchan

XavierChapron, 2021-08-06:
@joerchan Thank to your help I was able to make it work as expected!

Pmr-precure, 2023-10-10:
Long shot here, I think I encountered same issue. CONFIG_BT_GATT_AUTO_SEC_REQ=n doesn't seems to work. Even though i set it. It says in the build that it isn't set. Removing bt_conn_set_security doesn't seems to work either. Because then it won't ask for a bond/pair.  And putting the delay doesn't do anything

joerchan, 2023-10-11:
@Pmr-precure By not set do you mean that it is marked as `CONFIG_BT_GATT_AUT_SEC_REQ is not set`, because that is the same as being =n.

This issue is a long time since I dealt with, I don't remember all the details here.
If you have an issue I suggest you either go to the discord channel to discuss your problem there, or open a new github issue and describe your problem there.

xchapron-ledger, 2023-10-11:
@Pmr-precure and @joerchan Hello,
That was a long time ago, and I don't have access to this code anymore, but as I remember it, the fix on my side was:
- use a delayed kwork to call `bt_conn_set_security()` so that it is called in another thread (mainly it needs to be done after the return of `connected` callback)

Pmr-precure, 2023-10-11:
> [@Pmr-precure](https://github.com/Pmr-precure) and [@joerchan](https://github.com/joerchan) Hello, That was a long time ago, and I don't have access to this code anymore, but as I remember it, the [...]

Thanks for even responding!  I knew it was a long shot! I havnt really been able to make a k_work_delayed_submit implementation correct.