zephyrproject-rtos/zephyr #36601: Add input support to audio_codec

nickstoughton, 2021-06-28:
**Is your enhancement proposal related to a problem? Please describe.**
The current APIs for audio codecs contain only support for output, e.g. `audio_code_start_output` and `audio_codec_stop_output`.

**Describe the solution you'd like**
Add APIs to support input through audio codecs. These can largely mirror the output functions, enums etc. Additional channels for input will need to be defined

**Describe alternatives you've considered**
I am developing a codec driver at present and have overloaded the output functions so that they also start/stop input as well, but this does not give any fine control to the user. Also, I want to be able to mute the input while playing output, and overloading does not allow this.

**Additional context**
Existing audio codecs will need updating to add these functions. Certainly, the TLV320DAC 310 supports input.

lenghonglin, 2021-11-30:
need support too !

nashif, 2022-11-20:
please submit a pull request with the changes you are looking for.

closed 2022-11-20 by nashif